pbkhrv/rtl_433-hass-addons #124: Partially detected Govee water leak sensors

Vlad-Star, 2022-11-11:
I am running [HaOS OVA](https://github.com/home-assistant/operating-system/releases/download/9.3/haos_ova-9.3.qcow2.xz) VM image with Home Assistant 2022.11.2 on top and NooElec NESDR Mini 2 USB Stick (RTL2832 + R820T2) connected to the host machine and exposed to the VM. I've installed rtl_433 + this discovery addon to it.

When I try to add [Govee H5054](https://www.amazon.com/Govee-Detectors-Wireless-Detector-Sensitive/dp/B07QSFRSJX) sensors through MQTT auto-discovery, I am getting the following notifications in the log, but only battery sensor gets added, no matter what I do, water leak sensor doesn't show up.

```
# Button press
DEBUG:paho.mqtt.client:Received PUBLISH (d0, q0, r0, m0), 'rtl_433/9b13b3f4-rtl433/events', ...  (123 bytes)
DEBUG:root:MQTT message: "{\"time\":\"2022-11-10 16:16:36\",\"model\":\"Govee-Water\",\"id\":25744,\"event\":\"Button Press\",\"code\":\"9b6f05050746\",\"mic\":\"PARITY\"}"
DEBUG:root:Discovery timeout in the future for: homeassistant/sensor/Govee-Water-25744/Govee-Water-25744-UTC/config

# Battery removal and re-insert
DEBUG:paho.mqtt.client:Received PUBLISH (d0, q0, r0, m0), 'rtl_433/9b13b3f4-rtl433/events', ...  (160 bytes)
DEBUG:root:MQTT message: "{\"time\":\"2022-11-10 16:16:39\",\"model\":\"Govee-Water\",\"id\":25744,\"battery_ok\":1.0,\"battery_mV\":3000,\"event\":\"Battery Report\",\"code\":\"9b6f039b034c\",\"mic\":\"PARITY\"}"
DEBUG:root:Discovery timeout in the future for: homeassistant/sensor/Govee-Water-25744/Govee-Water-25744-UTC/config

# Water Leak (shortened contacts)
DEBUG:paho.mqtt.client:Received PUBLISH (d0, q0, r0, m0), 'rtl_433/9b13b3f4-rtl433/events', ...  (121 bytes)
DEBUG:root:MQTT message: "{\"time\":\"2022-11-10 16:17:02\",\"model\":\"Govee-Water\",\"id\":25744,\"event\":\"Water Leak\",\"code\":\"9b6f04040746\",\"mic\":\"PARITY\"}"
DEBUG:root:Discovery timeout in the future for: homeassistant/sensor/Govee-Water-25744/Govee-Water-25744-UTC/config
DEBUG:paho.mqtt.client:Received PUBLISH (d0, q0, r0, m0), 'rtl_433/9b13b3f4-rtl433/events', ...  (121 bytes)
DEBUG:root:MQTT message: "{\"time\":\"2022-11-10 16:17:04\",\"model\":\"Govee-Water\",\"id\":25744,\"event\":\"Water Leak\",\"code\":\"9b6f04040746\",\"mic\":\"PARITY\"}"
DEBUG:root:Discovery timeout in the future for: homeassistant/sensor/Govee-Water-25744/Govee-Water-25744-UTC/config

```
[image]

**rtl_433.conf.template**:
```
report_meta level
report_meta noise
report_meta stats
report_meta time:iso:usec:utc:tz
report_meta protocol
# verbose

device      0
output mqtt://localhost:1883,user=<USER>,pass=<PASS>
frequency   433.92M

protocol 73  # LaCrosse TX141-Bv2, TX141TH-Bv2, TX141-Bv3, TX141W, TX145wsdth, (TFA, ORIA) sensor
protocol 192 # Govee Water Leak Detector H5054, Door Contact Sensor B5023

```

I had to add the following lines to mqtt.yaml to get both battery and water leak sensor recognized (which I'd expect to be done by auto-discovery addon).
```
binary_sensor:
  - name: Govee Water Leak Detector 27544
    state_topic: rtl_433/9b13b3f4-rtl433/devices/Govee-Water/25744/event
    json_attributes_topic: rtl_433/9b13b3f4-rtl433/devices/Govee-Water/25744
    unique_id: govee_water_25744
    device_class: moisture
    payload_on: Water Leak
    payload_off: Button Press

sensor:
  - name: Govee Water Leak Detector Battery 27544
    state_topic: rtl_433/9b13b3f4-rtl433/devices/Govee-Water/25744/battery_ok
    json_attributes_topic: rtl_433/9b13b3f4-rtl433/devices/Govee-Water/25744
    device_class: battery
    value_template: "{{ (value_json | float * 100) | int }}"
    unique_id: govee_water_25744_b
    unit_of_measurement: "%"
```

After adding them to mqtt.yaml I can see both sensors and their status:
[image]
[image]
[image]

I don't know what's wrong. Let me know if you need any further information, I have 4 more of those sensors to add :)

bombledmonk, 2022-12-13:
If these are new sensors, it seems there is a difference in what they are transmitting.  Here's more info on the topic.  https://github.com/merbanan/rtl_433/issues/2265

deviantintegral, 2023-02-11:
I think this is fixed in the latest rtl_433 master. Now that we have the "next" addon, I'm calling this fixed. Thank you!

Vlad-Star, 2023-02-12:
@deviantintegral Sorry, didn't have time to test it until last night. I've installed the "next" version of both (rtl_433 and discovery add-on), but still getting the same result. Activated 6 out of 10 sensors. Sensor's battery is discovered, but the sensor itself is not. No issues with discoveries for LaCrosse and Acurite that also use rtl_433.

`DEBUG:paho.mqtt.client:Received PUBLISH (d0, q0, r0, m0), 'rtl_433/9b13b3f4-rtl433/events', ...  (160 bytes)
DEBUG:root:MQTT message: "{\"time\":\"2023-02-12 14:18:07\",\"model\":\"Govee-Water\",\"id\":56948,\"battery_ok\":1.0,\"battery_mV\":3000,\"event\":\"Battery Report\",\"code\":\"218b039b035a\",\"mic\":\"PARITY\"}"
DEBUG:root:Discovery timeout in the future for: homeassistant/sensor/Govee-Water-56948/Govee-Water-56948-UTC/config
DEBUG:root:Discovery timeout in the future for: homeassistant/sensor/Govee-Water-56948/Govee-Water-56948-B/config
DEBUG:paho.mqtt.client:Received PUBLISH (d0, q0, r0, m0), 'rtl_433/9b13b3f4-rtl433/events', ...  (160 bytes)
DEBUG:root:MQTT message: "{\"time\":\"2023-02-12 14:18:08\",\"model\":\"Govee-Water\",\"id\":56948,\"battery_ok\":1.0,\"battery_mV\":3000,\"event\":\"Battery Report\",\"code\":\"218b039b035a\",\"mic\":\"PARITY\"}"
DEBUG:root:Discovery timeout in the future for: homeassistant/sensor/Govee-Water-56948/Govee-Water-56948-UTC/config
DEBUG:root:Discovery timeout in the future for: homeassistant/sensor/Govee-Water-56948/Govee-Water-56948-B/config
DEBUG:paho.mqtt.client:Received PUBLISH (d0, q0, r0, m0), 'rtl_433/9b13b3f4-rtl433/events', ...  (160 bytes)
DEBUG:root:MQTT message: "{\"time\":\"2023-02-12 14:18:09\",\"model\":\"Govee-Water\",\"id\":56948,\"battery_ok\":1.0,\"battery_mV\":3000,\"event\":\"Battery Report\",\"code\":\"218b039b035a\",\"mic\":\"PARITY\"}"
DEBUG:root:Discovery timeout in the future for: homeassistant/sensor/Govee-Water-56948/Govee-Water-56948-UTC/config
DEBUG:root:Discovery timeout in the future for: homeassistant/sensor/Govee-Water-56948/Govee-Water-56948-B/config
DEBUG:paho.mqtt.client:Received PUBLISH (d0, q0, r0, m0), 'rtl_433/9b13b3f4-rtl433/events', ...  (154 bytes)`

Still have to stick the following into the mqtt configuration.
`binary_sensor:
  - name: Govee Water Leak Detector 56948
    state_topic: rtl_433/9b13b3f4-rtl433/devices/Govee-Water/56948/event
    json_attributes_topic: rtl_433/9b13b3f4-rtl433/devices/Govee-Water/56948
    unique_id: govee_water_56948
    device_class: moisture
    payload_on: Water Leak
    payload_off: Button Press
`

Vlad-Star, 2023-02-12:
Also tried to leave only one of 191 or 231 protocols. 231 didn;t work at all (no messages in discovery log on button press).

deviantintegral, 2023-02-15:
The above message doesn't show `leak_num` field: https://github.com/merbanan/rtl_433/pull/2273/files#diff-a7909b4d30881faf8a9d0f5572de0ce8408b0a7440522a6ac9afb84c2359ae25R387

It must be coming from the 22.11 release? Can you post output from the next addon? https://github.com/merbanan/rtl_433/pull/2273 was merged in January so it's not in the last stable release.

Vlad-Star, 2023-02-17:
Thanks for reopening it!I am currently on 2023.2.3. I am away from home until Feb 27, can remotely pull the existing logs, but won't be able to physically reach out those sensors to press the buttons, shorten contacts etc. Please keep this issue open until then.

Just to check on the list of my actions: 
1) Deinstall/reinstall both "next" versions retaining  configs, then restart core (just in case). 
2) make sure log_level = debug in the add-on
3) get a new sensor, pull the battery tab to activate
4) press the button, enjoy beeps
5) shorten the contacts (to simulate water leak), enjoy the siren
6) post logs from the add-on UI here

Anything else that may help? Any other logs?

deviantintegral, 2023-02-18:
> I am currently on 2023.2.3

That's fine. There's no changes in the past several months at least that have anything do to with Home Assistant itself. You won't need to restart core. I don't even do that while developing.

Do you have to get a new sensor? I mean, you can if you want, but I would have thought for testing you could just short one you already have set up.

Vlad-Star, 2023-02-28:
`
# Button press
DEBUG:paho.mqtt.client:Received PUBLISH (d0, q0, r0, m0), 'rtl_433/9b13b3f4-rtl433/events', ...  (123 bytes)
DEBUG:root:MQTT message: "{\"time\":\"2023-02-28 16:55:18\",\"model\":\"Govee-Water\",\"id\":56948,\"event\":\"Button Press\",\"code\":\"218b05050750\",\"mic\":\"PARITY\"}"
DEBUG:root:homeassistant/sensor/Govee-Water-56948/Govee-Water-56948-UTC/config:{"device_class": "timestamp", "name": "Govee-Water-56948-UTC", "entity_category": "diagnostic", "enabled_by_default": false, "icon": "mdi:clock-in", "state_topic": "rtl_433/9b13b3f4-rtl433/devices/Govee-Water/56948/time", "unique_id": "Govee-Water-56948-UTC", "device": {"identifiers": ["Govee-Water-56948"], "name": "Govee-Water-56948", "model": "Govee-Water", "manufacturer": "rtl_433"}}
DEBUG:paho.mqtt.client:Sending PUBLISH (d0, q0, r1, m54065), 'b'homeassistant/sensor/Govee-Water-56948/Govee-Water-56948-UTC/config'', ... (389 bytes)
INFO:root:Published Govee-Water/56948: time
INFO:root:Skipped Govee-Water/56948: event, code

# Shortened contacts:
DEBUG:root:MQTT message: "{\"time\":\"2023-02-28 16:56:32\",\"model\":\"Govee-Water\",\"id\":56948,\"event\":\"Water Leak\",\"code\":\"218b04040750\",\"mic\":\"PARITY\"}"
DEBUG:root:Discovery timeout in the future for: homeassistant/sensor/Govee-Water-56948/Govee-Water-56948-UTC/config
DEBUG:paho.mqtt.client:Received PUBLISH (d0, q0, r0, m0), 'rtl_433/9b13b3f4-rtl433/events', ...  (121 bytes)
DEBUG:root:MQTT message: "{\"time\":\"2023-02-28 16:56:34\",\"model\":\"Govee-Water\",\"id\":56948,\"event\":\"Water Leak\",\"code\":\"218b04040750\",\"mic\":\"PARITY\"}"
DEBUG:root:Discovery timeout in the future for: homeassistant/sensor/Govee-Water-56948/Govee-Water-56948-UTC/config

# Battery re-insert:
DEBUG:paho.mqtt.client:Received PUBLISH (d0, q0, r0, m0), 'rtl_433/9b13b3f4-rtl433/events', ...  (160 bytes)
DEBUG:root:MQTT message: "{\"time\":\"2023-02-28 16:59:34\",\"model\":\"Govee-Water\",\"id\":56948,\"battery_ok\":1.0,\"battery_mV\":3000,\"event\":\"Battery Report\",\"code\":\"218b039b035a\",\"mic\":\"PARITY\"}"
DEBUG:root:Discovery timeout in the future for: homeassistant/sensor/Govee-Water-56948/Govee-Water-56948-UTC/config
DEBUG:root:Discovery timeout in the future for: homeassistant/sensor/Govee-Water-56948/Govee-Water-56948-B/config
DEBUG:paho.mqtt.client:Received PUBLISH (d0, q0, r0, m0), 'rtl_433/9b13b3f4-rtl433/events', ...  (160 bytes)
DEBUG:root:MQTT message: "{\"time\":\"2023-02-28 16:59:36\",\"model\":\"Govee-Water\",\"id\":56948,\"battery_ok\":1.0,\"battery_mV\":3000,\"event\":\"Battery Report\",\"code\":\"218b039b035a\",\"mic\":\"PARITY\"}"
DEBUG:root:Discovery timeout in the future for: homeassistant/sensor/Govee-Water-56948/Govee-Water-56948-UTC/config
DEBUG:root:Discovery timeout in the future for: homeassistant/sensor/Govee-Water-56948/Govee-Water-56948-B/config`

Vlad-Star, 2023-02-28:
I didn't open the new one, but pulled one of the existing sensors, which still shows only one entity:
[image]

Vlad-Star, 2023-03-01:
@deviantintegral do you want me to submit the same data (from above) to https://github.com/merbanan/rtl_433/issues ?

deviantintegral, 2023-03-02:
Yes plase, because I would have expected `"code":"218b04040750"` would be broken down into other fields like "leak_num" or similar.

Vlad-Star, 2023-03-03:
Based on the conversation over there, is seems that **leak_num** field may or may be present, depending on the sensor's revision

https://github.com/merbanan/rtl_433/blob/master/src/devices/govee.c#L224
vs
https://github.com/merbanan/rtl_433/blob/master/src/devices/govee.c#L375

deviantintegral, 2023-03-08:
Ok, so does the event field give you anything useful with those older sensors?

Vlad-Star, 2023-03-08:
@deviantintegral looks like on the old sensors the event data field only used to report status of the battery; in case of pressing the button or water leak, this field is just repeating the previous byte (i.e. 0x0404 or 0x0505)

[This commit](https://github.com/merbanan/rtl_433/commit/5d16fb6a4c95485fbf94f332ba6072323cf694c1#diff-a7909b4d30881faf8a9d0f5572de0ce8408b0a7440522a6ac9afb84c2359ae25) seems to be describing behavior of the old sensors correctly

Vlad-Star, 2023-03-08:
Just to confirm, the following code, manually added to HA MQTT configuration, brings the water leak binary sensor to life. I'd certainly expect (and appreciate) the discovery plugin doing the same for me :) 
```
  - name: Govee Water Leak Detector 56948
    state_topic: rtl_433/9b13b3f4-rtl433/devices/Govee-Water/56948/event
    json_attributes_topic: rtl_433/9b13b3f4-rtl433/devices/Govee-Water/56948
    unique_id: govee_water_56948
    device_class: moisture
    payload_on: Water Leak
    payload_off: Button Press
```
[image]

deviantintegral, 2023-03-15:
Thanks for sharing that.

Given that you've got this working manually, we know there's no changes needed in rtl_433 itself. However, there's no support built-in for water leak (or even contact) sensors in the autodiscovery script. I do think it wouldn't be too much work to add in support for your water sensor. Here's a few examples:

https://github.com/merbanan/rtl_433/blob/master/examples/rtl_433_mqtt_hass.py#L392-L403

If you were to add those in (and test, as I can't!) and create a PR upstream, I'd consider patching it in to the next release.

Vlad-Star, 2023-03-27:
I am afraid I don't have enough knowledge to create a PR, though will be happy to help testing it, given the exact directions

jefmartel, 2023-07-29:
@Vlad-Star Sorry to necro bump this threat but I am having similatr issue

Can you tell me where you inserted those line?

```
binary_sensor:
  - name: Govee Water Leak Detector 56948
    state_topic: rtl_433/9b13b3f4-rtl433/devices/Govee-Water/56948/event
    json_attributes_topic: rtl_433/9b13b3f4-rtl433/devices/Govee-Water/56948
    unique_id: govee_water_56948
    device_class: moisture
    payload_on: Water Leak
    payload_off: Button Press
```

You said in the first post MQTT.yaml but I have no such file on my HA.

Vlad-Star, 2023-07-29:
(off-topic) @jefmartel I've got an e-mail notification with your question but don't see it here. I assume you already resolved it, but just in case answering it here. This is the line I have in my **configuration.yaml**
`mqtt: !include mgtt.yaml`

and binary sensor definition block goes to **mqtt.yaml**:
```
binary_sensor:
  - name: Govee Water Leak Detector 56948
 ...
```

jefmartel, 2023-07-29:
> (off-topic) @jefmartel I've got an e-mail notification with your question but don't see it here. I assume you already resolved it, but just in case answering it here. This is the line I have in my **configuration.yaml** `mqtt: !include mgtt.yaml`

Hi thanks for your reply. I have finally the whole set up working after too many hours to work on that. What I failed to see was the mqtt.yaml was an included file in configuration.yaml. I also failed to see that the new entity generated from the mqtt.yaml file would only appear in the entity pane, not the. mqtt device. Your comment on Feb 28 was the final light I needed to get everything working.

Thank you again.

kcpants, 2023-08-19:
FYI: I've just submitted a pull request to address this issue in the main rtl_433 repository https://github.com/merbanan/rtl_433/pull/2605 

Because these water sensors do not always have a unique identifying field, I had to add some new logic that checks the actual value of the event field so it wasn't quite as straightforward as adding a new entry to the mapping dictionary (hopefully this wont hold up the change from being integrated).

If the PR is accepted the sensors can be detected by running the auto-discovery addon and triggering a moisture event (by shorting the contacts on the sensor).

kcpants, 2023-09-27:
@deviantintegral The first part of the fix for this issue has been merged to `rtl_433`. It changes the behavior of `rtl_433` (no changes to the auto-discovery python script yet). Do you know if there is any manual work necessary to update `rtl_433-next` to include the latest commit so I can work on testing the changes to `rtl_433_mqtt_hass.py` or will it happen automatically?

I uninstalled and reinstalled `rtl_433-next` and based on the log output it looks like it's running yesterday's code (2023-09-26-1506) so perhaps I just need to wait a day.
```
Starting rtl_433 with rtl_433.conf...
[rtl_433] rtl_433 version nightly-17-g8e2be252 branch master at 202309261506 inputs file rtl_tcp RTL-SDR
```

deviantintegral, 2023-09-29:
The nightly job runs once a day, so that was probably it:

```
docker run --rm -it ghcr.io/pbkhrv/rtl_433-hass-addons-rtl_433-next-aarch64:next bash
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
bash-5.1# rtl_433 -v
rtl_433 version nightly branch master at 202309272030 inputs file rtl_tcp RTL-SDR
```

kcpants, 2023-09-29:
Thanks, I came to the same conclusion as well. with the new build I was able to test the change in PR 2605 and it works as expected. Once it is integrated, these sensors will be fully discoverable by running the MQTT auto-discovery add-on then triggering a `detect_wet` event (by pressing the button or shorting the contacts), as well as a battery event (by removing and reinserting the battery).

kcpants, 2023-10-15:
For any folks still invested in getting a solution to this problem, the [pull request containing the fix](https://github.com/merbanan/rtl_433/pull/2605) seems to have stalled but since there is no open feedback and it has been approved by one of the `rtl_433` collaborators it's a bit unclear why.

If you'd like to see this issue closed I recommend either leaving a comment on the open pull request stating that the fix would be helpful to you and asking if there's anything else needed in order to merge the change or opening a separate pull request that contains that same change and trying to merge it yourself (I've attached a diff below if you'd like to take that approach -- no need to credit me in the PR).

[govee.txt](https://github.com/pbkhrv/rtl_433-hass-addons/files/12908510/govee.txt)

I will leave one final comment on the PR myself in the hopes of getting some more attention on it but beyond that I don't believe there's much more I can do.

kcpants, 2023-10-17:
FYI, the pull request has been merged. This issue can likely now be closed.

closed 2023-10-19 by deviantintegral