pbkhrv/rtl_433-hass-addons #126: pull latest from RTL_433

tinstep, 2022-12-28:
### The problem

Hello,
Great addon 
I just got an Inkbird Pool temp controller and was hoping you could pull the latest from rtl_433 as it has a few updates to pull the temps - main change was here https://github.com/merbanan/rtl_433/pull/2220

### What addon are you reporting the bug for?

rtl_443

### What is the addon version?

latest

### What type of MQTT Broker are you using?

Home Assistant Mosquitto MQTT Broker

### Addon log messages

```txt
.
```

### Additional information

.

sombrero2003, 2023-01-16:
Hello,
thank you for this great add-on.
I am also curious about this. I understand that this add-on uses a pre-compiled image of rtl_433 but I am wondering whether it's possible to force the compilation of rtl_433. Just as tinstep, I would like to use a more recent version of rtl_433 that improves support for some of my sensors. Thanks again.

deviantintegral, 2023-01-18:
I'm working on automating this at https://github.com/pbkhrv/rtl_433-hass-addons/pull/129. The linked PR isn't merged yet, and it seems like there's work and complications there.

What I'd like to do is get the above in, and then default to using the latest tag with a config option to switch to a master build. It looks like a new tag came out recently at https://github.com/merbanan/rtl_433/releases/tag/22.11 so there's some good timing there.

deviantintegral, 2023-02-11:
This is done! You can get 22.11 with the normal upgrade. If you need master, you'll need to switch to the "next" addon and then reinstall it anytime you want a newer version.

closed 2023-02-11 by deviantintegral