pbkhrv/rtl_433-hass-addons #127: cant add repository

dartfrogdk, 2023-01-01:
### The problem

Cmd('git') failed due to: exit code(128) cmdline: git clone -v --recursive --depth=1 --shallow-submodules https://github.com/pbkhrv/rtl_433-hass-addons/tree/main/rtl_433 /data/addons/git/b05f010a stderr: 'Cloning into '/data/addons/git/b05f010a'... fatal: repository 'https://github.com/pbkhrv/rtl_433-hass-addons/tree/main/rtl_433/' not found '

### What addon are you reporting the bug for?

rtl_443

### What is the addon version?

0.2.3

### What type of MQTT Broker are you using?

Home Assistant Mosquitto MQTT Broker

### Addon log messages

```txt
dont have any
```

### Additional information

none

deviantintegral, 2023-01-02:
> fatal: repository 'https://github.com/pbkhrv/rtl_433-hass-addons/tree/main/rtl_433/' not found

I think that's the wrong URL. Can you try https://github.com/pbkhrv/rtl_433-hass-addons instead?

gon0, 2023-08-26:
I have the same issue with the most recent HA versions ( Home Assistant 2023.8.4 Supervisor 2023.08.1 Operating System 10.5 Frontend 20230802.1 - latest). After clicking "ADD", nothing happens:
[image]

deviantintegral, 2023-08-27:
When I last ran into this it was because of a bug with copy / paste in the macOS app. Using a web browser fixed it.

_but_ looking at the error you're getting, I think like above it's due to adding the path to an individual addon, whereas you need to add just the repository and then all available addons will show up. I will add a note to the readme.

closed 2023-08-27 by deviantintegral