pbkhrv/rtl_433-hass-addons #130: Documentation of directory storing configuration files

EPeterE, 2023-01-08:
### The problem

Hi
I think the documentation needs updating, however I am new to this and could be mistaken. The addon is not looking in the directory in which the manual says to create the *.conf.template file in.
However, if the *.conf.template file is placed in the directory /config/ it is found.

[image]

[image]

### What addon are you reporting the bug for?

rtl_443

### What is the addon version?

0.3.1

### What type of MQTT Broker are you using?

Home Assistant Mosquitto MQTT Broker

deviantintegral, 2023-02-11:
🤔 hm, the addon should only look for `/config/rtl_433.conf`, not `/config/rtl_433.conf.template`. That was for backwards compatibility with the addon from before we supported multiple radios.

Do you have suggestions for docs updates aimed at new users? Are you saying that if you move the file you have out of the way, that the addon _doesn't_ create a starter file at `/config/rtl_433/rtl_433.conf.template`?

Gliebig, 2023-03-26:
Same issue I have. A folder called \rtl_433 is created upon install under the \config folder containing the example config
\config\rtl_433\rtl_433.conf.template.example
I created one file \config\rtl_433\rtl_4331.conf.template for one radio
```
device :0000000433
frequency 433.88M
convert customary
output mqtt://${192.168.9.20}:${1883},user=${mqtt},pass=${idontcare},retain=${retain},devices=rtl_433/9b13b3f4-rtl433/devices[/type][/model][/subtype][/channel][/id],events=rtl_433/9b13b3f4-rtl433/events,states=rtl_433/9b13b3f4-rtl433/states
```
And I created a second file \config\rtl_433\rtl_4332.conf.template for a second radio
```
device :0000000915
frequency 915M
convert customary
output mqtt://${192.168.9.20}:${1883},user=${mqtt},pass=${idontcare},retain=${retain},devices=rtl_433/9b13b3f4-rtl433/devices[/type][/model][/subtype][/channel][/id],events=rtl_433/9b13b3f4-rtl433/events,states=rtl_433/9b13b3f4-rtl433/states
```

but it does not look in that folder for the configuration files. Here's the log (including the spelling error):
```
Starting rtl_433 -c /config/rtl_4331.conf.template
rtl_433 version 22.11 branch  at 202211191645 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Failed to stat "/config/rtl_4331.conf.template"
``` 
When I move both files to \config\ 
then the log shows this:
```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
Starting rtl_433 -c /config/rtl_4331.conf.template
[18:17:50] WARNING: rtl_433 now supports automatic configuration and multiple radios. The rtl_433_conf_file option is deprecated. See the documentation for migration instructions.
rtl_433 version 22.11 branch  at 202211191645 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Publishing MQTT data to ${192.168.9.20} port ${1883}
Publishing device info to MQTT topic "rtl_433/9b13b3f4-rtl433/devices[/type][/model][/subtype][/channel][/id]".
Publishing events info to MQTT topic "rtl_433/9b13b3f4-rtl433/events".
Publishing states info to MQTT topic "rtl_433/9b13b3f4-rtl433/states".
MQTT connect (${192.168.9.20}:${1883}) failed: cannot parse address
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
```
It will not start even though the MQTT address is correct and the second configuration file does not run.

I have 2 bash scripts I used on a different pi that work using the command line to start rtl_433 and not the config files that works great However, I needed to move the radios to my pi running Home Assistant Supervisor on my RV and this addon fails.

deviantintegral, 2024-02-17:
Ah! I missed the warning in the screenshot:

> [20:10:28] WARNING: rtl_433 now supports automatic configuration and multiple radios. The rtl_433_conf_file option is
deprecated. See the documentation for migration instructions.

Empty out that configuration option in the addon config and restart. The warning will go away and you'll get log messages with `Starting rtl_433 with <file>...` instead.

catduckgnaf, 2024-02-19:
If you want to use my newer add-on that is actively developed, 
https://github.com/catduckgnaf/rtl_433_ha

morganlnick, 2024-05-15:
I am at a complete loss with this, Im a novice but I am doing my best to carefully review the documentation and threads around this issue, but still end up completely lost. I need to enable "output kv' to identify some generic sensors I purchased, the addon is on and running, and publishing to MQTT, but I cannot find the config file anywhere! There is no rtl_433.conf file in the /config directory, it seems there is not .conf file anywhere. The only file Ive been able to find generated by the installation are in the /config/rtl_433/rtl_433.conf.temp but the add-on is still happily running without an issue reporting absolutely none of the information I need to see in the log, and rejecting every .conf file I try to create for it because of this deprecation. 

Is there something obvious that I'm missing? Is it just a hidden file? In an entirely different directory? Do I need to learn how to fork the project to make changes to the automatic configuration? 
[image]

AlejandroBalseiro, 2024-11-15:
> Ah! I missed the warning in the screenshot:
>
>
> Empty out that configuration option in the addon config and restart. The warning will go away and you'll get log messages with `Starting rtl_433 with [...]

Hey!

I'm quite new on this, but I'm facing some issues here. I got the same warning, and I can't find how to make the change.

Whenever I try to install the add-on, I'm requested to create the file, but I can't find the config folder to follow the steps, although: it got lost.

my server for HA is a HA green (in case that matters).

Thanks!