pbkhrv/rtl_433-hass-addons #131: rtf_433 and stick working on linux... but not with home assistant and this add on

MadMacMad, 2023-01-13:
### The problem

hello,
if i'll use the usb stick on my computer and use the following config:
`   gain          0
   frequency     868.30M
   hop_interval  600
   ppm_error     0
   frequency     868.30M
   sample_rate   1000000
   report_meta   time:utc
   protocol 172 # Bresser Weather Center 6-in-1 (newer 5-in-1)
   protocol 173 # Bresser Weather Center 7-in-1
   output mqtt://192.168.178.142:1883,user=xxx,pass=xxx,retain=0,devices=rtl_433/bresser51
   output kv`
it works perfect and sending the data to MQTT Broker ... can see all data in HASS ... 
but when i'll put the stick in the raspberry with HASS ... it works just a few seconds it looks like...
sometimes i'll see the results in Konsole but most of time it's stuck after connection successful to MQTT...
stick and config working well... but the plugin stop's working after a few seconds...

output on linux:
`Publishing MQTT data to 192.168.178.142 port 1883
Publishing device info to MQTT topic "rtl_433/bresser51".
Registered 2 out of 223 device decoding protocols [ 172-173 ]
Found Rafael Micro R820T tuner
Exact sample rate is: 1000000.026491 Hz
[R82XX] PLL not locked!
Sample rate set to 1000000 S/s.
Tuner gain set to Auto.
Tuned to 868.300MHz.
MQTT Connected...
baseband_demod_FM: low pass filter for 1000000 Hz at cutoff 200000 Hz, 5.0 us
MQTT Connection established.
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-01-13 17:09:28
model     : Bresser-6in1 id        : 55879a19
channel   : 0            Battery   : 1             Temperature: 8.0 C
Humidity  : 79           Sensor type: 1            Wind Gust : 0.9 m/s
Wind Speed: 0.9 m/s      Direction : 174           UV        : 0.0
Startup   : 1            Flags     : 0             Integrity : CRC
_`

### What addon are you reporting the bug for?

rtl_443

### What is the addon version?

0.3.1

### What type of MQTT Broker are you using?

Home Assistant Mosquitto MQTT Broker

### Addon log messages

```txt
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
Starting rtl_433 with rtl_433.conf...
[rtl_433] rtl_433 version 21.12-159-g5f76eae6 branch  at 202210212043 inputs file rtl_tcp RTL-SDR
[rtl_433] Use -h for usage help and see https://triq.org/ for documentation.
[rtl_433] 
[rtl_433] New defaults active, use "-Y classic -s 250k" for the old defaults!
[rtl_433] 
[rtl_433] Publishing MQTT data to 192.168.178.142 port 1883
[rtl_433] Publishing device info to MQTT topic "rtl_433/bresser51".
[rtl_433] Registered 2 out of 223 device decoding protocols [ 172-173 ]
[rtl_433] Detached kernel driver
[rtl_433] Found Rafael Micro R820T tuner
[rtl_433] Exact sample rate is: 1000000.026491 Hz
[rtl_433] [R82XX] PLL not locked!
[rtl_433] Sample rate set to 1000000 S/s.
[rtl_433] Tuner gain set to Auto.
[rtl_433] Tuned to 868.300MHz.
[rtl_433] Allocating 15 zero-copy buffers
[rtl_433] MQTT Connected...
[rtl_433] baseband_demod_FM: low pass filter for 1000000 Hz at cutoff 200000 Hz, 5.0 us
[rtl_433] MQTT Connection established.
```

### Additional information

[rtl_433] MQTT Connection established.

there it stop's....
checked the same now on my other Raspberry ... no problem... doing the same as on my linux machine...
plz plz fix
thx & respect
M.

deviantintegral, 2023-01-14:
Things to check:

1. Make sure you see the mqtt connection being established in the mqtt addon logs.
2. Connect to the mqtt broker using MQTT Explorer and see if your data is making it there.
3. Add verbose (and additional levels) to your config to see if there's anything useful. I would do it both on your computer where it works, and the Pi where it doesn't so you can compare: https://github.com/merbanan/rtl_433/blob/master/conf/rtl_433.example.conf#L21-L25
4. Check if there's any chance there's a physical reason why one connection works and the other doesn't. Is your Pi somewhere it couldn't see the signal, but your computer could? Are you plugging in other USB devices that are causing the Pi to be under power, depending on what it's power supply is?

MadMacMad, 2023-01-18:
hello,
well just for anyone who has this issue too...
check your config... no extra spaces in front and between name and value

`
frequency 868.30M
report_meta time:tz
protocol 172 # Bresser Weather Center 6-in-1 (newer 5-in-1)
output mqtt://192.168.178.142:1883,user=xxx,pass=xxx,retain=1,devices=rtl_433/bresser51
`

deviantintegral, 2023-01-18:
Nice! So you're saying that something like `frequency     868.30M` breaks? If so, I think being more flexible parsing that would be a good feature request to file against rtl_433.

MadMacMad, 2023-01-18:
So you're saying that something like frequency     868.30M breaks?

Yes ... tried several times here... depend on machine...
on Pi yes ... on Mac Os X not... 
Pi with latest OS and self compiled ... config with spaces don't work
same with HASSI as OS ...
loading config with spaces on Mac OS X ... got rte_433 via brew ... no issue...

config with spaces also in front of keyword and between -> not good

`   gain          0`

JJoaquinrl, 2023-04-23:
> ### El problema
> hola, si usaré la memoria USB en mi computadora y usaré la siguiente configuración: ganancia 0 frecuencia 868.30M hop_interval 600 ppm_error 0 frecuencia 868.30M sample_rate 1000000 report_meta time:utc protocol 172 # Bresser Weather Center 6-in-1 (más nuevo 5 en 1) protocolo 173 # Bresser Weather Center 7 en 1 salida mqtt://192.168.178.142:1883,user=xxx,pass=xxx,retain=0,devices=rtl_433/bresser51 salida kv
> 
> funciona perfecto y envía los datos a MQTT Broker... puedo ver todos los datos en HASS... pero cuando pongo el dispositivo en la frambuesa con HASS... funciona solo unos segundos parece... a veces veo los resultados en Konsole, pero la mayoría de las veces se atasca después de que la conexión con MQTT se haya realizado con éxito... la configuración y la configuración funcionan bien... pero el complemento deja de funcionar después de unos segundos...
> 
> Salida en Linux: Publicación de datos MQTT en el puerto 192.168.178.142 1883 Publicación de información del dispositivo en el tema MQTT "rtl_433/bresser51". Se registraron 2 de 223 protocolos de decodificación de dispositivos [ 172-173 ] Sintonizador Rafael Micro R820T encontrado La frecuencia de muestreo exacta es: 1000000.026491 Hz [R82XX] ¡PLL no bloqueado! Frecuencia de muestreo establecida en 1000000 S/s. Ganancia del sintonizador configurada en Auto. Sintonizado a 868.300MHz. MQTT conectado... baseband_demod_FM: filtro de paso bajo para 1000000 Hz en corte 200000 Hz, 5.0 us MQTT Conexión establecida.
> 
> hora: 2023-01-13 17:09:28 modelo: Bresser-6in1 id: 55879a19 canal: 0 Batería: 1 Temperatura: 8,0 C Humedad: 79 Tipo de sensor: 1 Ráfaga de viento: 0,9 m/s Velocidad del viento: 0,9 m/ s Dirección : 174 UV : 0.0 Arranque : 1 Indicadores : 0 Integridad : CRC _
> 
> ### ¿Para qué complemento estás reportando el error?
> rtl_443
> 
> ### ¿Cuál es la versión adicional?
> 0.3.1
> 
> ### ¿Qué tipo de agente MQTT está utilizando?
> Inicio Asistente Mosquitto MQTT Broker
> 
> ### Mensajes de registro de complementos
> ```
> s6-rc: info: service s6rc-oneshot-runner: starting
> s6-rc: info: service s6rc-oneshot-runner successfully started
> s6-rc: info: service fix-attrs: starting
> s6-rc: info: service fix-attrs successfully started
> s6-rc: info: service legacy-cont-init: starting
> s6-rc: info: service legacy-cont-init successfully started
> s6-rc: info: service legacy-services: starting
> s6-rc: info: service legacy-services successfully started
> Starting rtl_433 with rtl_433.conf...
> [rtl_433] rtl_433 version 21.12-159-g5f76eae6 branch  at 202210212043 inputs file rtl_tcp RTL-SDR
> [rtl_433] Use -h for usage help and see https://triq.org/ for documentation.
> [rtl_433] 
> [rtl_433] New defaults active, use "-Y classic -s 250k" for the old defaults!
> [rtl_433] 
> [rtl_433] Publishing MQTT data to 192.168.178.142 port 1883
> [rtl_433] Publishing device info to MQTT topic "rtl_433/bresser51".
> [rtl_433] Registered 2 out of 223 device decoding protocols [ 172-173 ]
> [rtl_433] Detached kernel driver
> [rtl_433] Found Rafael Micro R820T tuner
> [rtl_433] Exact sample rate is: 1000000.026491 Hz
> [rtl_433] [R82XX] PLL not locked!
> [rtl_433] Sample rate set to 1000000 S/s.
> [rtl_433] Tuner gain set to Auto.
> [rtl_433] Tuned to 868.300MHz.
> [rtl_433] Allocating 15 zero-copy buffers
> [rtl_433] MQTT Connected...
> [rtl_433] baseband_demod_FM: low pass filter for 1000000 Hz at cutoff 200000 Hz, 5.0 us
> [rtl_433] MQTT Connection established.
> ```
> 
> ### Información adicional
> [rtl_433] Conexión MQTT establecida.
> 
> ahí se detiene... verifiqué lo mismo ahora en mi otra Raspberry... no hay problema... haciendo lo mismo que en mi máquina linux... por favor, arreglen gracias y respeten a M.

hello I belong to a home automation group with home assistant and we also use the rtl repository. About 20 people have the same problem and it starts up fine but stays in mqtt established connection and does not receive anything. We give it to restart adlon and the third or fourth time it is already seen that it receives data. as a curiosity to say that if the config file is not there or is empty it is always started, but of course we need the mqtt. We have tried what you say about sun spaces, but without positive results. If you can give me any help I would appreciate it. all the best

JJoaquinrl, 2023-04-24:
I forgot to mention that I have 4 rtl usb devices and only one does what I said. a Fitipower FC0012 the others don't do that and it boots perfectly when restarting home assistant every night, then the issue of the config file I don't think it is, because the other rtl usb never give that boot problem. thank you

deviantintegral, 2023-05-01:
@JJoaquinrl You can more easily confirm the radio is working by adding an additional `output kv` line. If you see data printed out in the logs, but nothing in mqtt, then you know the problem is somewhere between rtl_433 and your mqtt broker.

JJoaquinrl, 2023-05-01:
> @JJoaquinrl You can more easily confirm the radio is working by adding an additional `output kv` line. If you see data printed out in the logs, but nothing in mqtt, then you know the problem is somewhere between rtl_433 and your mqtt broker.

Hello, thank you for your interest.
It does not show any data with that parameter when it does not start.
the problem is not in mqtt, it is when loading rtl433 and that model of usb.
I have 4 using them for years and I put this model because users of my home automation group reported this problem to me, I bought that same model and the aforementioned certainly happens.
sometimes it starts fine and sometimes it doesn't.
The only way I have been able to know if it starts well or badly is with the noise report, if current noise level is 42 db it means that it has started badly, if current noise level is 16 db it has started well, but I can't get it to restart automatically because this message is not transmitted to the mqtt

JJoaquinrl, 2023-05-01:
[image]

wackla33, 2023-05-02:
Good night.  I have carried out the checks that the user @JJoaquinrl comments and I modestly think that I may have detected the real problem that affects us.  Some time ago I commented on that same problem and the answer you gave me was that it seemed like a hardware problem related to the power supply of my raspi, but after many checks, the last one being the noise issue mentioned by @JJoaquinrl, the result is what He mentions that if the noise settles at 42.1 dB, the USB device is disconnected, which only makes reports every 10 minutes, but does not send the information to Mqtt or, at most, only does so with one device, the first it detects, but not the rest.  However, if the noise levels are around 15 or 16 decibels at start-up, the operation is perfect.  I have created automations to force the restart of MQTT and RTL 433 successively and the error, although it occurs, takes longer to do so.

deviantintegral, 2023-05-02:
Good details. I think based on your notes that this isn’t really a problem with the addon specifically, but something with the physical radio and antenna, or perhaps a bug in rtl_433.

For example, see this comment showing how changing antennas and using USB extension cables can change the noise floor: https://github.com/merbanan/rtl_433/issues/1780#issuecomment-907374325

JJoaquinrl, 2023-05-02:
> Buenos detalles. Creo que, según sus notas, esto no es realmente un problema con el complemento específicamente, sino algo con la radio física y la antena, o tal vez un error en rtl_433.
> 
> Por ejemplo, vea este comentario que muestra cómo el cambio de antenas y el uso de cables de extensión USB pueden cambiar el ruido de fondo: [merbanan/rtl_433#1780 (comentario)](https://github.com/merbanan/rtl_433/issues/1780#issuecomment-907374325)

It is not a real noise, it is just a piece of information in which we detect that it is not working.

deviantintegral, 2024-02-17:
I think this is solved, and not related to this addon so I'm closing. Feel free to open a new issue if it is. Thanks!

closed 2024-02-17 by deviantintegral