pbkhrv/rtl_433-hass-addons #132: Scan Multiple frequencys  433.92 and 915mhz

Grey-Lancaster, 2023-01-17:
### The problem

I have several diff sensors.  They show up fine with just one frequency.  My attempts to get both  have alluded me so far.

Google groups says it should work :-)
You can use the hop feature -H  I use it to scan 433 and 915 example rtl_433 -f 433920000 -f 915000000 -H 20
change the hop interval 20 (seconds) to what ever you wish,  I use 14 seconds. A little experimenting with interval is required. you may miss a few results here and there. I use rtl_433  to decode atlas weather station and Ecowitt soil moisture testers. They run automations for my awning wind alert automatic in and soil moisture and rain fall for garden irrigation system and etc..

With this config file it finds the 915, but not the 433?

# This is an empty template for configuring rtl_433. mqtt information will be
# automatically added. Create multiple files ending in '.conf.template' to
# manage multiple rtl_433 radios, being sure to set the 'device' setting. The
# device must be set before mqtt output lines.
# https://github.com/merbanan/rtl_433/blob/master/conf/rtl_433.example.conf
#-f 433920000 -f 915000000 -H 20
frequency     915M
frequency     433.92M
hop_interval  100
output mqtt://${host}:${port},user=${username},pass=${password},retain=${retain}

# Uncomment the following line to also enable the default "table" output to the
# addon logs.
output kv

### What addon are you reporting the bug for?

rtl_443

### What is the addon version?

0.3.1

### What type of MQTT Broker are you using?

Home Assistant Mosquitto MQTT Broker

### Addon log messages

```txt
[rtl_433] Temperature: 19.6 C
[rtl_433] Humidity  : 45 %
[rtl_433] Pressure  : 981.5 hPa
[rtl_433] Integrity : CRC
[rtl_433] Tuned to 433.920MHz.
[rtl_433] Allocating 15 zero-copy buffers
[rtl_433] _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
[rtl_433] center_frequency: 433920000                        frequencies: 915000000, 433920000                     hop_times : 100
[rtl_433] bitbuffer_add_bit: Warning: row count limit (50 rows) reached
[rtl_433] bitbuffer_add_bit: Warning: row count limit (50 rows) reached
[rtl_433] bitbuffer_add_bit: Warning: row count limit (50 rows) reached
[rtl_433] bitbuffer_add_bit: Warning: row count limit (50 rows) reached
[rtl_433] bitbuffer_add_bit: Warning: row count limit (50 rows) reached
[rtl_433] bitbuffer_add_bit: Warning: row count limit (50 rows) reached
[rtl_433] bitbuffer_add_bit: Warning: row count limit (50 rows) reached
[rtl_433] bitbuffer_add_bit: Warning: row count limit (50 rows) reached
[rtl_433] bitbuffer_add_bit: Warning: row count limit (50 rows) reached
[rtl_433] bitbuffer_add_bit: Warning: row count limit (50 rows) reached
[rtl_433] bitbuffer_add_bit: Warning: row count limit (50 rows) reached
[rtl_433] Tuned to 915.000MHz.
[rtl_433] _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
[rtl_433] center_frequency: 915000000                        frequencies: 915000000, 433920000                     hop_times : 100
[rtl_433] Allocating 15 zero-copy buffers
[rtl_433] bitbuffer_add_bit: Warning: row count limit (50 rows) reached
[rtl_433] bitbuffer_add_bit: Warning: row count limit (50 rows) reached
[rtl_433] _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
[rtl_433] 
[rtl_433] time      : 2023-01-16 19:56:52
[rtl_433] model     : Fineoffset-WH32B
[rtl_433] ID        : 229
[rtl_433] Battery   : 1
[rtl_433] Temperature: 19.6 C
[rtl_433] Humidity  : 45 %
[rtl_433] Pressure  : 981.4 hPa
[rtl_433] Integrity : CRC
[rtl_433] _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
[rtl_433] 
[rtl_433] time      : 2023-01-16 19:57:18
[rtl_433] model     : Fineoffset-WH32
[rtl_433] ID        : 89
[rtl_433] Battery   : 1
[rtl_433] Temperature: 7.6 C
[rtl_433] Humidity  : 64 %
[rtl_433] Integrity : CRC
[rtl_433] _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
[rtl_433] 
[rtl_433] time      : 2023-01-16 19:57:56
[rtl_433] model     : Fineoffset-WH32B
[rtl_433] ID        : 229
[rtl_433] Battery   : 1
[rtl_433] Temperature: 19.6 C
[rtl_433] Humidity  : 45 %
[rtl_433] Pressure  : 981.4 hPa
[rtl_433] Integrity : CRC
[rtl_433] Tuned to 433.920MHz.
[rtl_433] _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
[rtl_433] center_frequency: 433920000                        frequencies: 915000000, 433920000                     hop_times : 100
[rtl_433] Allocating 15 zero-copy buffers
[rtl_433] bitbuffer_add_bit: Warning: row count limit (50 rows) reached
[rtl_433] bitbuffer_add_bit: Warning: row count limit (50 rows) reached
[rtl_433] Tuned to 915.000MHz.
[rtl_433] Allocating 15 zero-copy buffers
[rtl_433] _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
[rtl_433] center_frequency: 915000000                        frequencies: 915000000, 433920000                     hop_times : 100
[rtl_433] _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
[rtl_433] 
[rtl_433] time      : 2023-01-16 20:01:08
[rtl_433] model     : Fineoffset-WH32B
[rtl_433] ID        : 229
[rtl_433] Battery   : 1
[rtl_433] Temperature: 19.6 C
[rtl_433] Humidity  : 45 %
[rtl_433] Pressure  : 981.6 hPa
[rtl_433] Integrity : CRC
[rtl_433] Tuned to 433.920MHz.
[rtl_433] _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
[rtl_433] center_frequency: 433920000                        frequencies: 915000000, 433920000                     hop_times : 100
[rtl_433] Allocating 15 zero-copy buffers
[rtl_433] bitbuffer_add_bit: Warning: row count limit (50 rows) reached
[rtl_433] bitbuffer_add_bit: Warning: row count limit (50 rows) reached
[rtl_433] bitbuffer_add_bit: Warning: row count limit (50 rows) reached
[rtl_433] bitbuffer_add_bit: Warning: row count limit (50 rows) reached
[rtl_433] bitbuffer_add_bit: Warning: row count limit (50 rows) reached
[rtl_433] bitbuffer_add_bit: Warning: row count limit (50 rows) reached
[rtl_433] bitbuffer_add_bit: Warning: row count limit (50 rows) reached
[rtl_433] bitbuffer_add_bit: Warning: row count limit (50 rows) reached
[rtl_433] bitbuffer_add_bit: Warning: row count limit (50 rows) reached
[rtl_433] bitbuffer_add_bit: Warning: row count limit (50 rows) reached
[rtl_433] bitbuffer_add_bit: Warning: row count limit (50 rows) reached
[rtl_433] bitbuffer_add_bit: Warning: row count limit (50 rows) reached
[rtl_433] bitbuffer_add_bit: Warning: row count limit (50 rows) reached
[rtl_433] bitbuffer_add_bit: Warning: row count limit (50 rows) reached
[rtl_433] bitbuffer_add_bit: Warning: row count limit (50 rows) reached
[rtl_433] bitbuffer_add_bit: Warning: row count limit (50 rows) reached
[rtl_433] bitbuffer_add_bit: Warning: row count limit (50 rows) reached
[rtl_433] bitbuffer_add_bit: Warning: row count limit (50 rows) reached
[rtl_433] Tuned to 915.000MHz.
[rtl_433] Allocating 15 zero-copy buffers
[rtl_433] _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
[rtl_433] center_frequency: 915000000                        frequencies: 915000000, 433920000                     hop_times : 100
```

Grey-Lancaster, 2023-01-18:
Thanks, I saw that yesterday a fingered with pulse_detect autoleveland it is also wonky :-)  I mean it would only show working at 915 and not 433

I think I have found the issue but can not seem to fix it.
If I run just 433 it sts the sampling rate to 
Sample rate set to 250000 S/s.
and my device shows up

but when I run both frequencies the sample rate is set to
Sample rate set to 1000000 S/s
and the 433mhs device does not show up, but the 915's do

I have tried adding
sample_rate   250k
but is still goes to 10k when both are enabled.

hop_interval  30
pulse_detect autolevel
sample_rate   250k
frequency     433.92M
pulse_detect autolevel
frequency     915.00M
pulse_detect autolevel

With just 433

[rtl_433] Found Rafael Micro R820T tuner
[rtl_433] Exact sample rate is: 250000.000414 Hz
[rtl_433] [R82XX] PLL not locked!
[rtl_433] Sample rate set to 250000 S/s.
[rtl_433] Tuner gain set to Auto.
[rtl_433] Tuned to 433.920MHz.
[rtl_433] Allocating 15 zero-copy buffers
[rtl_433] baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
[rtl_433] MQTT Connected...
[rtl_433] MQTT Connection established.

with both:
[rtl_433] Found Rafael Micro R820T tuner
[rtl_433] Exact sample rate is: 1000000.026491 Hz
[rtl_433] [R82XX] PLL not locked!
[rtl_433] Sample rate set to 1000000 S/s.
[rtl_433] Tuner gain set to Auto.
[rtl_433] Tuned to 433.920MHz.
[rtl_433] Allocating 15 zero-copy buffers
[rtl_433] baseband_demod_FM: low pass filter for 1000000 Hz at cutoff 100000 Hz, 10.0 us
[rtl_433] MQTT Connected...
[rtl_433] MQTT Connection established.

Thanks 
Grey

deviantintegral, 2024-02-17:
I actually ended up buying a second (well third 😅 ) myself after having long-term stability issues with frequency hopping. The stick would just hang and the whole thing would need to be rebooted. Since this is all in the control of the underlying rtl_433 binary and not the addon, I'm closing, but feel free to reopen if there's something I've missed.

closed 2024-02-17 by deviantintegral