pbkhrv/rtl_433-hass-addons #133: How set other frequency than default?

Hamilkar247, 2023-01-19:
### The problem

Well .. i have a problem i can't understant how i have set not default 433MHz frequency. 

I add to my hacs both of project

first i check `rtl_433` `Current version: 0.3.1`

but when I run program after created config file, program didn't set 868.8MHz and use 433MHz I saw

```The rtl_433_conf_file option is deprecated. See the documentation for migration instructions.```

But i am not sure where I find "documentation for migration instruction".

Next try - I installed `rtl_433 MQTT Auto Discovery`

[image]

[image]

but here i don't see fields to define frequency and devices ... I see only Acurate at 433MHz, when i know i have my klimalogg beside me and i can catch them at other device in rtl_433

[image]

### What addon are you reporting the bug for?

rtl_433_mqtt_autodiscover

### What is the addon version?

0.5.1

### What type of MQTT Broker are you using?

Home Assistant Mosquitto MQTT Broker

Hamilkar247, 2023-01-20:
Okay, solution is simple - I am idiot - in version 3.1
 i create rtl_433.conf file, and give path with rtl_433.config file.
 
In newer options still i don't know how i can change default frequency and devices.

deviantintegral, 2023-01-21:
The supported configuration method is to have configuration files in the `rtl_433` folder in your HA configuration directory. If you don't touch the addon config form, it will create an example one when the addon starts up that you can modify. This is how we can support multiple radios in one system.

This was documented for existing users in the CHANGELOG at https://github.com/pbkhrv/rtl_433-hass-addons/blob/main/rtl_433/CHANGELOG.md#020---2022-02-14, but looking at https://github.com/pbkhrv/rtl_433-hass-addons/blob/main/rtl_433/README.md I can see how it's not clear. I'm reopening this to get that done. Thank you!

Here's my current configs - hopefully this gives you an idea:

`/config/rtl_433/weather_station.conf.template`:
```
# This is an empty template for configuring rtl_433. mqtt information will be
# automatically added. Create multiple files ending in '.conf.template' to
# manage multiple rtl_433 radios.
# https://github.com/merbanan/rtl_433/blob/master/conf/rtl_433.example.conf

device 0
output mqtt://${host}:${port},user=${username},pass=${password},retain=${retain}
report_meta time:iso:tz:local
convert si
```

`/config/rtl_433/doorbell.conf.template`:
```
device 1
frequency 916.8M
output mqtt://${host}:${port},user=${username},pass=${password},retain=false
report_meta time:iso:tz:local
```

invisible999, 2023-02-25:
Does 'device 0' and 'device 1' refer to two different RT_SDR dongles or to the same dongle but operating on different frequencies?

deviantintegral, 2023-02-26:
Two different dongles. They can only listen to one frequency at a time, or they can hop over different frequencies. Check the rtl_433 examples docs for more details.

closed 2024-02-17 by deviantintegral