pbkhrv/rtl_433-hass-addons #143: Log.sh Error when Running Independently?

efaden, 2023-03-14:
### The problem

Using the following in docker-compose.

  ```
rtl_433_autodiscovery:
    container_name: rtl_433_autodiscovery
    image: ghcr.io/pbkhrv/rtl_433-hass-addons-rtl_433_mqtt_autodiscovery-amd64 # On Raspberry Pi replace `amd64` with the appropriate architecture.
    environment:
      - MQTT_HOST=ip
      - MQTT_USERNAME=user
      - MQTT_PASSWORD=password
      - MQTT_PORT=1883
      - RTL_TOPIC=rtl_433/+/events
      - DISCOVERY_PREFIX=homeassistant
      - DISCOVERY_INTERVAL=600
      - LOG_LEVEL=debug
      - OTHER_ARGS=--retain
```

### What addon are you reporting the bug for?

rtl_443

### What is the addon version?

Latest

### What type of MQTT Broker are you using?

Home Assistant Mosquitto MQTT Broker

### Addon log messages

```txt
I'm getting .... 

rtl_433_autodiscovery    | [s6-init] making user provided files available at /var/run/s6/etc...exited 0.
rtl_433_autodiscovery    | [s6-init] ensuring user provided files have correct perms...exited 0.
rtl_433_autodiscovery    | [fix-attrs.d] applying ownership & permissions fixes...
rtl_433_autodiscovery    | [fix-attrs.d] done.
rtl_433_autodiscovery    | [cont-init.d] executing container initialization scripts...
rtl_433_autodiscovery    | [cont-init.d] done.
rtl_433_autodiscovery    | [services.d] starting services
rtl_433_autodiscovery    | [services.d] done.
rtl_433_autodiscovery    | /usr/lib/bashio/log.sh: line 107: debug: unbound variable
rtl_433_autodiscovery    | [cmd] /run.sh exited 1
rtl_433_autodiscovery    | [cont-finish.d] executing container finish scripts...
rtl_433_autodiscovery    | [cont-finish.d] done.
rtl_433_autodiscovery    | [s6-finish] waiting for services.
rtl_433_autodiscovery    | [s6-finish] sending all processes the TERM signal.
rtl_433_autodiscovery    | [s6-finish] sending all processes the KILL signal and exiting.
^CGracefully stopping... (press Ctrl+C again to force)
```

deviantintegral, 2023-03-15:
Can you try specifying an image tag? Both `0.6.0` and `next` work for me. `latest` (the default) literally just means "last pushed to the repository" so it could be any PR's code too.

`image: ghcr.io/pbkhrv/rtl_433-hass-addons-rtl_433_mqtt_autodiscovery-amd64:0.6.0`

We should probably update the example to specify a version.

```yaml
version: '3'
services:
  rtl_433_autodiscovery:
    container_name: rtl_433_autodiscovery
    image: ghcr.io/pbkhrv/rtl_433-hass-addons-rtl_433_mqtt_autodiscovery-aarch64:0.6.0
    environment:
      - MQTT_HOST=ip
      - MQTT_USERNAME=user
      - MQTT_PASSWORD=password
      - MQTT_PORT=1883
      - RTL_TOPIC=rtl_433/+/events
      - DISCOVERY_PREFIX=homeassistant
      - DISCOVERY_INTERVAL=600
      - LOG_LEVEL=debug
      - OTHER_ARGS=--retain
```

On my mac (which doesn't even have a radio plugged in or access to any mqtt server right now) I get to the point where it's looping trying to connect, which I think is past your "usage" line:

```console
$ docker-compose up
[+] Running 1/0
 ⠿ Container rtl_433_autodiscovery  Recreated                                                                             0.0s
Attaching to rtl_433_autodiscovery
rtl_433_autodiscovery  | s6-rc: info: service s6rc-oneshot-runner: starting
rtl_433_autodiscovery  | s6-rc: info: service s6rc-oneshot-runner successfully started
rtl_433_autodiscovery  | s6-rc: info: service fix-attrs: starting
rtl_433_autodiscovery  | s6-rc: info: service fix-attrs successfully started
rtl_433_autodiscovery  | s6-rc: info: service legacy-cont-init: starting
rtl_433_autodiscovery  | s6-rc: info: service legacy-cont-init successfully started
rtl_433_autodiscovery  | s6-rc: info: service legacy-services: starting
rtl_433_autodiscovery  | s6-rc: info: service legacy-services successfully started
rtl_433_autodiscovery  | Running in stand-alone docker mode
rtl_433_autodiscovery  | Starting rtl_433_mqtt_hass.py...
rtl_433_autodiscovery  | INFO:root:Enabling debug logging
rtl_433_autodiscovery  | INFO:root:Discovering all devices
rtl_433_autodiscovery  | DEBUG:root:MQTT Client: Starting Loop
rtl_433_autodiscovery  | DEBUG:paho.mqtt.client:Connection failed, retrying
rtl_433_autodiscovery  | DEBUG:paho.mqtt.client:Connection failed, retrying
rtl_433_autodiscovery  | DEBUG:paho.mqtt.client:Connection failed, retrying
rtl_433_autodiscovery  | DEBUG:paho.mqtt.client:Connection failed, retrying
rtl_433_autodiscovery  | DEBUG:paho.mqtt.client:Connection failed, retrying
```

efaden, 2023-03-17:
That fixed it.... now I am getting an error about the discovery timeout being in the future.

deviantintegral, 2023-03-17:
> an error about the discovery timeout being in the future.

That's not an error! That means that it has already published a device, and is not going to re-send it based on `DISCOVERY_INTERVAL`.

efaden, 2023-03-17:
Yep. It's running great now. Thanks

closed 2023-03-17 by deviantintegral