pbkhrv/rtl_433-hass-addons #146: Addons RTL_433 stops working

TrueMB, 2023-05-01:
### The problem

Hello!
I got multiple Issues, where I am not sure, if they are somehow connected.
The important thing, it already worked once! And as far I know, I didn't change anything after that.

Problem 1: Each time I am restarting the Addon, I get an error. After the 3rd time it works.
Tested multiple times and seems to work always after the 3rd try.
```
[rtl_433] LIBUSB_ERROR_NOT_FOUND: Entity not found!
[rtl_433] Check your RTL-SDR dongle, USB cables, and power supply.```

Problem 2: Today I got in the System log following error:
```
Failed to to call /addons/9b13b3f4_rtl433/stats - Container addon_9b13b3f4_rtl433 is not running
```
After reinstalling the Addon, the error seems to have vanished.

Problem 3: RTL_433 isn't publishing data anymore. Nothing changed. Even before the reinstallation of the Addon. MQTT Server is also connted. 
The HomeAssistant did restart after the working/not working state.

Config (\config\rtl_433\rtl_433.conf.template):
```
# This is an empty template for configuring rtl_433. mqtt information will be
# automatically added. Create multiple files ending in '.conf.template' to
# manage multiple rtl_433 radios, being sure to set the 'device' setting. The
# device must be set before mqtt output lines.
# https://github.com/merbanan/rtl_433/blob/master/conf/rtl_433.example.conf

device 0

output mqtt://${host}:${port},user=${username},pass=${password},retain=${retain}
report_meta time:iso:usec:tz

frequency 868M

# To keep the same topics when switching between the normal and edge versions,
# use this output line instead.
# output mqtt://${host}:${port},user=${username},pass=${password},retain=${retain},devices=rtl_433/9b13b3f4-rtl433/devices[/type][/model][/subtype][/channel][/id],events=rtl_433/9b13b3f4-rtl433/events,states=rtl_433/9b13b3f4-rtl433/states

# Uncomment the following line to also enable the default "table" output to the
# addon logs.
output kv

# Disable TPMS sensors by default. These can cause an overwhelming number of
# devices and entities to show up in Home Assistant.
# This list is generated by running:
# rtl_433 -R help 2>&1 | grep -i tpms | sd '.*\[(\d+)\].*' 'protocol -/run.sh'
#    [59]  Steelmate TPMS
#    [60]  Schrader TPMS
#    [82]  Citroen TPMS
#    [88]  Toyota TPMS
#    [89]  Ford TPMS
#    [90]  Renault TPMS
#    [95]  Schrader TPMS EG53MA4, PA66GF35
#    [110]  PMV-107J (Toyota) TPMS
#    [123]* Jansite TPMS Model TY02S
#    [140]  Elantra2012 TPMS
#    [156]  Abarth 124 Spider TPMS
#    [168]  Schrader TPMS SMD3MA4 (Subaru)
#    [180]  Jansite TPMS Model Solar
#    [186]  Hyundai TPMS (VDO)
#    [201]  Unbranded SolarTPMS for trucks
#    [203]  Porsche Boxster/Cayman TPMS
protocol -59
protocol -60
protocol -82
protocol -88
protocol -89
protocol -90
protocol -95
protocol -110
protocol -123
protocol -140
protocol -156
protocol -168
protocol -180
protocol -186
protocol -201
protocol -203
protocol 70
#protocol 115
#protocol 116
#protocol 185

convert si

#decoder n=Honeywell_ActivLink,m=FSK_PWM,s=160,l=320,r=400,y=480,invert,bits=48```

I hope somebody got maybe an Idea, since I would love to use this for catching the doorbell Signal.
If more informations are needed, tell me and I will add it.

### What addon are you reporting the bug for?

rtl_443

### What is the addon version?

0.4.0

### What type of MQTT Broker are you using?

Home Assistant Mosquitto MQTT Broker

### Addon log messages

```txt
-------------------------------- Log Nr. 1 --------------------------------
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
Starting rtl_433 with rtl_433.conf...
[rtl_433] rtl_433 version 22.11 branch  at 202211191645 inputs file rtl_tcp RTL-SDR
[rtl_433] Use -h for usage help and see https://triq.org/ for documentation.
[rtl_433] Publishing MQTT data to core-mosquitto port 1883
[rtl_433] Publishing device info to MQTT topic "rtl_433/9b13b3f4-rtl433/devices[/type][/model][/subtype][/channel][/id]".
[rtl_433] Publishing events info to MQTT topic "rtl_433/9b13b3f4-rtl433/events".
[rtl_433] Publishing states info to MQTT topic "rtl_433/9b13b3f4-rtl433/states".
[rtl_433] 
[rtl_433] New defaults active, use "-Y classic -s 250k" for the old defaults!
[rtl_433] 
[rtl_433] Registered 177 out of 223 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36 38-58 63 67-71 73-81 83-87 91-94 96-100 102-105 108-109 111-116 119 121 124-128 130-139 141-149 151-155 157-161 163-167 170-175 177-179 181-185 187-197 199 202 204-215 217-223 70 ]
[rtl_433] Found Rafael Micro R820T tuner
[rtl_433] Exact sample rate is: 1000000.026491 Hz
[rtl_433] [R82XX] PLL not locked!
[rtl_433] Sample rate set to 1000000 S/s.
[rtl_433] Tuner gain set to Auto.
[rtl_433] Tuned to 868.000MHz.
[rtl_433] Allocating 15 zero-copy buffers
[rtl_433] Async read stalled, exiting!
[rtl_433] 
[rtl_433] LIBUSB_ERROR_NOT_FOUND: Entity not found!
[rtl_433] Check your RTL-SDR dongle, USB cables, and power supply.
[rtl_433] 
[rtl_433] WARNING: async read failed (-5).
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

-------------------------------- Log Nr. 2 --------------------------------
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
Starting rtl_433 with rtl_433.conf...
[rtl_433] rtl_433 version 22.11 branch  at 202211191645 inputs file rtl_tcp RTL-SDR
[rtl_433] Use -h for usage help and see https://triq.org/ for documentation.
[rtl_433] Publishing MQTT data to core-mosquitto port 1883
[rtl_433] Publishing device info to MQTT topic "rtl_433/9b13b3f4-rtl433/devices[/type][/model][/subtype][/channel][/id]".
[rtl_433] Publishing events info to MQTT topic "rtl_433/9b13b3f4-rtl433/events".
[rtl_433] Publishing states info to MQTT topic "rtl_433/9b13b3f4-rtl433/states".
[rtl_433] 
[rtl_433] New defaults active, use "-Y classic -s 250k" for the old defaults!
[rtl_433] 
[rtl_433] Registered 177 out of 223 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36 38-58 63 67-71 73-81 83-87 91-94 96-100 102-105 108-109 111-116 119 121 124-128 130-139 141-149 151-155 157-161 163-167 170-175 177-179 181-185 187-197 199 202 204-215 217-223 70 ]
[rtl_433] Found Rafael Micro R820T tuner
[rtl_433] Exact sample rate is: 1000000.026491 Hz
[rtl_433] [R82XX] PLL not locked!
[rtl_433] Sample rate set to 1000000 S/s.
[rtl_433] Tuner gain set to Auto.
[rtl_433] Tuned to 868.000MHz.
[rtl_433] Allocating 15 zero-copy buffers
[rtl_433] MQTT Connected...
[rtl_433] baseband_demod_FM: low pass filter for 1000000 Hz at cutoff 200000 Hz, 5.0 us
[rtl_433] MQTT Connection established.
```

deviantintegral, 2023-05-01:
I would assume the simplest explanation - there's a problem with the USB connection to the radio. Do you have a powered USB hub you can try? Does the radio work fine on another computer?

TrueMB, 2023-05-01:
I could try testing a bit with the dongle. Guess this makes sense. Thanks for the idea!

Wanted to make sure, that I didn't misconfigured the settings.

TrueMB, 2023-05-02:
So I did change the port, reconfigured it in VMWare and it seems to work for now. Might be until the next restart.

I think disconnecting the dongle once and then restarting the VM fixes it temporarily.
This is not solved for me, but not really something, where the RTL Hass module can help. 

Thanks for the idea devian! I will look deeper into that, if it happens again.

closed 2023-05-02 by TrueMB

deviantintegral, 2023-05-02:
Good to hear! I have one install where Home Assistant is running inside of a VM with KVM. USB was totally unreliable with my RTL dongles where they would reset or disconnect. However, my z-wave adapter on the same system works fine. I ended up moving rtl_433 out to the host.

In contrast, I’ve never one had a radio drop its USB connection on a my Raspberry PI.