pbkhrv/rtl_433-hass-addons #150: rtl_433_mqtt_hass.py: error: argument -p/--port: invalid int value: 'null'

M0LTE, 2023-05-29:
### The problem

I run Home Assistant and rtl_433 in Docker, and trying to run rtl_433_autodiscovery in another container. However something seems to be up with the parameters being passed to rtl_433_mqtt_hass.py:

```rtl_433_autodiscovery    | [s6-init] ensuring user provided files have correct perms...exited 0.
rtl_433_autodiscovery    | [fix-attrs.d] applying ownership & permissions fixes...
rtl_433_autodiscovery    | [fix-attrs.d] done.
rtl_433_autodiscovery    | [cont-init.d] executing container initialization scripts...
rtl_433_autodiscovery    | [cont-init.d] done.
rtl_433_autodiscovery    | [services.d] starting services
rtl_433_autodiscovery    | [services.d] done.
rtl_433_autodiscovery    | Using an external mqtt broker.
rtl_433_autodiscovery    | curl: (6) Could not resolve host: supervisor
rtl_433_autodiscovery    | [18:47:56] ERROR: Something went wrong contacting the API
rtl_433_autodiscovery    | Starting rtl_433_mqtt_hass.py...
rtl_433_autodiscovery    | usage: rtl_433_mqtt_hass.py [-h] [-d] [-q] [-u USER] [-P PASSWORD] [-H HOST]
rtl_433_autodiscovery    |                             [-p PORT] [-r] [-f] [-R RTL_TOPIC]
rtl_433_autodiscovery    |                             [-D DISCOVERY_PREFIX] [-i DISCOVERY_INTERVAL]
rtl_433_autodiscovery    | rtl_433_mqtt_hass.py: error: argument -p/--port: invalid int value: 'null'
rtl_433_autodiscovery    | [cmd] /run.sh exited 2
rtl_433_autodiscovery    | [cont-finish.d] executing container finish scripts...
rtl_433_autodiscovery    | [cont-finish.d] done.
rtl_433_autodiscovery    | [s6-finish] waiting for services.
rtl_433_autodiscovery    | [s6-finish] sending all processes the TERM signal.
rtl_433_autodiscovery    | [s6-finish] sending all processes the KILL signal and exiting.
rtl_433_autodiscovery exited with code 2```

My docker-compose.yml:

```
version: '3'
services:
  rtl_433_autodiscovery:
    container_name: rtl_433_autodiscovery
    image: ghcr.io/pbkhrv/rtl_433-hass-addons-rtl_433_mqtt_autodiscovery-aarch64
    environment:
      - MQTT_HOST=10.45.0.1
      - MQTT_PORT=1883
```

I have no supervisor in my setup, and I have my own standalone MQTT server, no auth. Is my setup supported?

### What addon are you reporting the bug for?

rtl_433_mqtt_autodiscover

### What is the addon version?

0.2.3

### What type of MQTT Broker are you using?

Other (details in the bug description)

Sparticuz, 2023-06-23:
I'm getting the same thing

varunsaravagi, 2023-07-04:
I am getting the same thing

varunsaravagi, 2023-07-04:
This is happening because the `run.sh` in the latest docker image doesn't contain the standalone docker code. The docker image needs to be updated to include the latest `run.sh`

deviantintegral, 2023-07-07:
Is the addon version `0.2.3` above correct? That was a version for `rtl_433`, not the autodiscovery addon.

`0.2.3` is now over a year old. Can you upgrade to the latest releases?

kristoficko, 2023-08-11:
Same situation. Any ETA for a fix? Many thanks for the efforts!

Mike

deviantintegral, 2023-08-14:
@kristoficko are you able to confirm the addon version?  The original report is confusing that way.

I am hoping to look at this in the next week or two. Otherwise, help from others submitting and reviewing fixes is always appreciated!

kristoficko, 2023-08-15:
I'm running HA as a docker container without Add-on support, so I'm also running autodiscovery directly as a container.
Container repository:
`ghcr.io/pbkhrv/rtl_433-hass-addons-rtl_433_mqtt_autodiscovery-amd64`
Tag:
`latest`
Image ID:
`734fa2d3616f`

Docker compose entry:
`rtl_433_autodiscovery:
    container_name: rtl_433_autodiscovery
    image: ghcr.io/pbkhrv/rtl_433-hass-addons-rtl_433_mqtt_autodiscovery-amd64
    environment:
      - MQTT_HOST=xxx
      - MQTT_USERNAME=xxx
      - MQTT_PASSWORD=xxx
      - MQTT_PORT=1883
      - RTL_TOPIC=rtl_433[/id]
`
Happy to help!

Mike

deviantintegral, 2023-08-15:
Ah! Can you instead set the image to run 0.6.0 with `ghcr.io/pbkhrv/rtl_433-hass-addons-rtl_433_mqtt_autodiscovery-amd64:0.6.0`. I think our automation isn't pushing `latest` images at all, which is why that date is 19 months old.

kristoficko, 2023-08-15:
Issue resolved here as well when specifying docker image version as 0.6.0.

Many thanks!

closed 2023-11-06 by deviantintegral