pbkhrv/rtl_433-hass-addons #151: Feature Request : Filter for auto-add

dlasher, 2023-06-07:
### The problem

Lots of cars in my neighborhood have 433mhz-based TPMS - At (4) sensors per vehicle, I'd love to have some sort of "auto-add" filter that does NOT add the TPMS/* devices automatically.

example:
TPMS-Citroen-30d7fe80
TPMS-Citroen-30f16191
TPMS-Citroen-30f2032f
TPMS-Citroen-30f2911d
TPMS-Citroen-8cad8262
TPMS-Citroen-8cad8bf0
TPMS-Citroen-d064c992
TPMS-Hyundai-VDO-01be7ff2
TPMS-Hyundai-VDO-01be81b0
TPMS-Hyundai-VDO-11d7d5ae
TPMS-Hyundai-VDO-11d7d5c9
TPMS-Hyundai-VDO-11d7d680
TPMS-Hyundai-VDO-211fcd22
TPMS-Schrader-2033B59
TPMS-Schrader-2033CC7
TPMS-Schrader-EG53MA4-148EF9

### What addon are you reporting the bug for?

rtl_433_mqtt_autodiscover

### What is the addon version?

0.6.0

### What type of MQTT Broker are you using?

Home Assistant Mosquitto MQTT Broker

### Addon log messages

```txt
INFO:root:Published TPMS/Citroen/8cad8bf0: time, pressure_kPa, temperature_C, rssi, snr, noise
INFO:root:Skipped TPMS/Citroen/8cad8bf0: protocol, state, flags, repeat, maybe_battery, freq1, freq2
INFO:root:Published Acurite-606TX/162: time, battery_ok, temperature_C, rssi, snr, noise
INFO:root:Skipped Acurite-606TX/162: protocol
<snip>
INFO:root:Published TPMS/Citroen/30f16191: time, pressure_kPa, temperature_C, rssi, snr, noise
INFO:root:Skipped TPMS/Citroen/30f16191: protocol, state, flags, repeat, maybe_battery, freq1, freq2
```

### Additional information

thanks

deviantintegral, 2023-06-09:
The default rtl_433 configuration we create excludes TPMS sensors. Then, the data is never sent to mqtt or the autodiscovery addon. Are you using this configuration? If so, I would guess there's additional protocols added since we should add.

https://github.com/pbkhrv/rtl_433-hass-addons/blob/main/rtl_433/run.sh#L55-L90

dlasher, 2023-06-09:
Thanks, didn't realize you could filter there.

I'm using the rtl_433 docker from here : https://github.com/hertzg/rtl_433_docker and then feeding into MQTT. That image doesn't appear to filter out anything, which in theory isn't bad, as long as the rtl_433_mqtt_autodiscover doesn't automagically add them to Home Assistant.

Chicken/Egg problem - either we filter them at the RTL_433 side, or we filter what to auto-add. It feels like a GUI-based filter in the rtl_433_mqtt_autodiscover portion would be a more easily accessible way to control what gets added into HA.

dlasher, 2023-06-14:
For now I figured out how to filter them out on the rtl_433 side - not easy when using the container - you have to map them out in the create, because trying to edit/keep any persistent config file is PITA.

`      - '-R-59,-60,-82,-88,-89,-90,-95,-110,-123,-140,-156,-168,-180,-186,-201,-203,-208,-212,-225,-226,-241'`

Takes care of all the TPMS for me.

FWIW, still think that filtering on the auto_discover is a more user-friendly fix.

deviantintegral, 2023-06-14:
Sounds good. The autodiscovery script is also maintained upstream. If it gets support for some sort of filtering, then once that's done we can figure out to expose it from the home assistant addon side.

dlasher, 2023-06-15:
Sadly, the docker container I'm using doesn't appear to pass R flags correctly, so I can't filter there, at least at the moment - bug opened : https://github.com/hertzg/rtl_433_docker/issues/81

catduckgnaf, 2023-09-30:
I implemented this with my auto discovery plugin. can select a filter from the config in the add on. I plan to keep maintaining it myself.

closed 2024-02-17 by deviantintegral