pbkhrv/rtl_433-hass-addons #152: Discovered entities with a name that starts with the device name

s992, 2023-08-04:
### The problem

I'm not entirely sure that this is an issue caused by this addon, so please let me know if I'm posting in the wrong place! :)

With a recent HA upgrade, I started receiving the following warning:

[image]

In case the image doesn't load or is otherwise not legible, the text is:

```
Discovered entities with a name that starts with the device name

This stops working in version 2024.2.0. Please address before upgrading.

Some MQTT entities have an entity name that starts with the device name. This is 
not expected. To avoid a duplicate name the device name prefix is stripped of the 
entity name as a work-a-round. Please inform the maintainer of the software 
application that supplies the affected entities to fix this issue.

List of affected entities:

- sensor.acurite_5n1_a_2468_f
- sensor.acurite_5n1_a_2468_ws
- sensor.acurite_5n1_a_2468_rt
- sensor.acurite_5n1_a_2468_wd
- sensor.acurite_5n1_a_2468_h
- sensor.acurite_5n1_a_2468_b
```

The "Learn More" link links out to https://developers.home-assistant.io/blog/2023-057-21-change-naming-mqtt-entities/.

I'm using [this weather station](https://www.acurite.com/5-in-1-pro-weather-sensor-temperature-humidity-rainfall-wind-speed-wind-direction-06014rm.html) and it's been working fine (and continues to work fine!) for about a year now. I have not integrated any other devices with rtl_433 or MQTT.

Should I be concerned by this warning? Is there anything I can do to resolve it?

### What addon are you reporting the bug for?

rtl_433_mqtt_autodiscover

### What is the addon version?

0.6.0

### What type of MQTT Broker are you using?

Home Assistant Mosquitto MQTT Broker

### Additional information

`rtl_433` addon version is 0.4.0
`rtl_433 MQTT Auto Discovery` addon version is 0.6.0

deviantintegral, 2023-08-05:
I don't think there's anything for you to worry about right now. The actual breaking change won't kick in until 2024.2.

I just upgraded myself, and it looks like this is triggering for all of my zigbee2mqtt devices too. I will probably wait to see how they solve this before tackling this here.

deviantintegral, 2023-08-05:
Actually, this will need to be fixed in the upstream rtl-433 repository since that's where the autodiscovery script itself lives.

BarryMar, 2023-08-11:
This just got fixed in https://github.com/merbanan/rtl_433/commit/9e0fb41e4c04fc73f5bfa2191f898f86a1e6a684

lindsand, 2023-11-06:
Good to know it's fixed for coming releases (as it's not urgent).

The warning comes from here, device name is repeated is the sensor names:
- and are now truncated by MQTT to -T (for Temperature) and -B (for Battery) and so on.

[image]

deviantintegral, 2023-11-06:
I'm leaving this open until https://github.com/merbanan/rtl_433/issues/2652 is resolved and we can point to a new release containing this fix for the stable branch.

closed 2025-03-22 by deviantintegral