pbkhrv/rtl_433-hass-addons #155: Auto Discover-Next Startup Error "rtl_433_mqtt_hass.py: error: unrecognized arguments: -T null"

gtmax500, 2023-09-02:
### The problem

I have the new version of the GoVee water detectors which were included in the latest version of rtl_433 and I can manually configure them when running the -Next version of rtl_433-hass (Next) and see them on MQTT Explorer. However, I cannot get rtl_433 MQTT Auto Discovery (next) to start on my HAOS running on bare metal. Whenever I try to start the service I get the following error:

rtl_433_mqtt_hass.py: error: unrecognized arguments: -T null

Full log details were added to this issue.

The regular version of rtl_433 MQTT Auto Discover (0.6.0) starts up fine and I can auto discover the device but it only configures an entity for the battery and time, nothing else.

I'm pretty new to Hass so it could be just a configuration issue but I cannot seem to get this service to start no matter what I do. Also, sorry if this is not the appropriate way to report this issue.

### What addon are you reporting the bug for?

rtl_433_mqtt_autodiscover

### What is the addon version?

Next

### What type of MQTT Broker are you using?

Home Assistant Mosquitto MQTT Broker

### Addon log messages

```txt
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
mqtt found in this Home Assistance instance.
Starting rtl_433_mqtt_hass.py...
usage: rtl_433_mqtt_hass.py [-h] [-d] [-q] [-u USER] [-P PASSWORD] [-H HOST]
                            [-p PORT] [-c CA_CERT] [-r] [-f] [-R RTL_TOPIC]
                            [-D DISCOVERY_PREFIX] [-i DISCOVERY_INTERVAL]
                            [-x EXPIRE_AFTER] [-I IDS [IDS ...]]
rtl_433_mqtt_hass.py: error: unrecognized arguments: -T null
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
```

deviantintegral, 2023-09-03:
How recently did you install the autodiscovery next addon? HA only downloads new versions of branch releases if you uninstall and reinstall the addon.

This does have me thinking that we should put some sort of output / date / version in the logs before starting up to help troubleshoot issues like this.

gtmax500, 2023-09-03:
> How recently did you install the autodiscovery next addon? HA only downloads new versions of branch releases if you uninstall and reinstall the addon.
>
> This does have me thinking that we should put [...]

I just started working with HA the last two weeks or so, so at least since them. I have also uninstalled and reinstalled a few times over the last few days so it should be the latest.

catduckgnaf, 2023-09-04:
I can report the same issue

lcoppenr, 2023-09-17:
Same issue here.

Jsalas424, 2023-09-17:
Reporting same issue here, just installed today:

```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
mqtt found in this Home Assistance instance.
Starting rtl_433_mqtt_hass.py...
usage: rtl_433_mqtt_hass.py [-h] [-d] [-q] [-u USER] [-P PASSWORD] [-H HOST]
                            [-p PORT] [-c CA_CERT] [-r] [-f] [-R RTL_TOPIC]
                            [-D DISCOVERY_PREFIX] [-i DISCOVERY_INTERVAL]
                            [-x EXPIRE_AFTER] [-I IDS [IDS ...]]
rtl_433_mqtt_hass.py: error: unrecognized arguments: -T null
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
```

ifuchs, 2023-09-18:
I uninstalled and installed the (next) version and it died with a Code 2 seconds after starting.

Chartreusito, 2023-09-25:
I can report the same issue with a fresh install of Next AddOn

catduckgnaf, 2023-09-30:
Hey guys, I fixed it in my new branch I am maintaining.  and contributing too. With a script owned by the community. Works with next and non next.

Also, much easier to add more sensors https://github.com/catduckgnaf/rtl_433_ha

deviantintegral, 2023-10-01:
@catduckgnaf since you've got a fix already, do you mind opening a PR for it? I took a quick look at your repo but I didn't see an obvious commit to cherry-pick.

catduckgnaf, 2023-10-01:
> @catduckgnaf since you've got a fix already, do you mind opening a PR for it? I took a quick look at your repo but I didn't see an obvious commit to cherry-pick.

I did already. Do you see it?

catduckgnaf, 2023-10-01:
@deviantintegral https://github.com/pbkhrv/rtl_433-hass-addons/pull/160

I plan to update the script as well.
The add-on I've hosted separately.
Currently my add-on will default to retain true, and it will only auto discover 1 id at a time, but that is configurable from the add on gui config.

I made the script and template on its own for this reason.

I also plan to do version numbers so updates will be rolled out without needing to uninstall and reinstall.

My main rtl_433 doesn't work yet, but I hope to correct that soon.

unverbraucht, 2023-10-24:
I fixed this in in PR #162. I'm not aware of any way how you can try this out in HA because I don't know how to specify branches when adding repos. If you want to feel free to download the version from my PR and install it locally as described in https://developers.home-assistant.io/docs/add-ons/tutorial

(sorry for copy'n'paste - there are multiple issues with the same fix).

deviantintegral, 2023-10-28:
This fix is released now. Thanks all!

closed 2023-10-28 by deviantintegral