pbkhrv/rtl_433-hass-addons #159: rtl_433 MQTT Auto Discovery (next) does not start

invisible999, 2023-09-26:
### The problem

RTL_433 Next is installed and started. I see messages in MQTT Explorer

When I install rtl_433 MQTT Auto Discovery (next), it does not start. Log output is shown below. Also, in the HAOS log there is following entry
`2023-09-25 23:01:35.209 ERROR (MainThread) [homeassistant.components.hassio] Failed to to call /addons/9b13b3f4_rtl433mqttautodiscovery-next/stats - Container addon_9b13b3f4_rtl433mqttautodiscovery-next is not running
`

### What addon are you reporting the bug for?

rtl_433_mqtt_autodiscover

### What is the addon version?

0.4.0

### What type of MQTT Broker are you using?

Home Assistant Mosquitto MQTT Broker

### Addon log messages

```txt
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
mqtt found in this Home Assistance instance.
Starting rtl_433_mqtt_hass.py...
usage: rtl_433_mqtt_hass.py [-h] [-d] [-q] [-u USER] [-P PASSWORD] [-H HOST]
                            [-p PORT] [-c CA_CERT] [-r] [-f] [-R RTL_TOPIC]
                            [-D DISCOVERY_PREFIX] [-i DISCOVERY_INTERVAL]
                            [-x EXPIRE_AFTER] [-I IDS [IDS ...]]
rtl_433_mqtt_hass.py: error: unrecognized arguments: -T null
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
```

catduckgnaf, 2023-09-28:
I am fixing things. Just made this

https://github.com/catduckgnaf/rtl_433_haos_autodiscovery_addon Please read the readme, and here as well.

https://github.com/catduckgnaf/rtl_433_ha

I am working on updating everything to be streamlined. This will work with next or not.

SunsetGolfer, 2023-10-16:
Same problem here 👎 
How can I help?

deviantintegral, 2023-10-16:
Thanks for offering!

I reviewed the above repo for changes that would clearly fix the argument parsing I could patch or cherry-pick in, but I didn't find one.

My time is limited at the moment (but will be opening up at the end of the month). I'm more than happy to merge a PR with 1-2 reviews, understanding I may not actually be able to test it myself before merging.

catduckgnaf, 2023-10-16:
> Thanks for offering!
>
> I reviewed the above repo for changes that would clearly fix the argument parsing I could patch or cherry-pick in, but I didn't find one.
>
> My time is limited at the moment (but [...]

Hello, I've made some changes since then. 

I am working on one or two things, and then really want to actively develop and work on things. 
I would be happy to take over maintaining the project, however I plan do make alot more changes, so I guess when I'm ready I'll continue with a Fork instead. I don't think that is the best solution for the community. I have the time, passion, and devices to pick up the torch, would much rather you spend your time helping me get up to speed. 

Let me know

fortyfortynine, 2023-10-22:
> I am fixing things. Just made this
> 
> https://github.com/catduckgnaf/rtl_433_ha
> 
> I am working on updating everything to be streamlined. This will work with next or not.

How do I use your script? I was able to use your template and now the log is showing my devices but a little lost on how to use the script.

catduckgnaf, 2023-10-23:
> How do I use your script? I was able to use your template and now the log is showing my devices but a little lost on how to use the script.

I have no idea why they don't want to merge my fix, at least for the time being. But I will be working on an intergration.
Quite honestly the script is mostly useless. It doesn't discover anything but weather sensors. Another reason I wanted to take ownership

fortyfortynine, 2023-10-24:
> I have no idea why they don't want to merge my fix, at least for the time being. But I will be working on an intergration. Quite honestly the script is mostly useless. It doesn't discover anything [...]

I’ll give a try adding my sensors in configuration.yaml. I was using another add-on (acurite2mqtt) but it doesn’t support adding any thing using protocol higher than 223. Trying to use rtl_433 so I can get a Govee water leak sensor using 231 added to Home Assistant.

catduckgnaf, 2023-10-24:
> I’ll give a try adding my sensors in configuration.yaml. I was using another add-on (acurite2mqtt) but it doesn’t support adding any thing using protocol higher than 223. Trying to use rtl_433 so I [...]

Its sad, but you are so right, most people seem to not use the script and so what you are saying. I am simply going to work on redoing this. I'd rather learn to code verse fight pointless gatekeeping

fortyfortynine, 2023-10-24:
> Its sad, but you are so right, most people seem to not use the script and so what you are saying. I am simply going to work on redoing this. I'd rather learn to code verse fight pointless gatekeeping

Thank you! I’ll keep my eye on your integration for when it’s published.

unverbraucht, 2023-10-27:
Probably caused by #164 and not #162 :)

deviantintegral, 2023-10-28:
I just finished rolling a new release, and now that config.json's fixes are on main this should be fixed. Thanks!

closed 2023-10-28 by deviantintegral