pbkhrv/rtl_433-hass-addons #163: How to install in HA Core?

haraldhh, 2023-10-23:
### The problem

This might be a totally newbie question and more about the documentation and not the code but here it goes: 

How do I install this rtl_433 MQTT Auto Discovery Add-on in HA Core?

### What addon are you reporting the bug for?

rtl_443

### What is the addon version?

0.2.3

### What type of MQTT Broker are you using?

Other (details in the bug description)

### Addon log messages

```txt
mosquitto
```

unverbraucht, 2023-10-24:
Hi @haraldhh, HA Core doesn't run add-ons since it's just the Python program itself. Add-Ons are in the end mostly Docker containers, which means you can run this one in any container environment. You'd have to run the image ghcr.io/pbkhrv/rtl_433-hass-addons-rtl_433-ARCH - replacing ARCH with amd64 for PCs or armv7/aarch64 for ARM-based boards like Raspis depending on your OS. Running `uname` in the terminal will tell you which arch you have.

You might find it easier to run generic rtl-433 in Docker instead since this repo is about making it work seamlessly as an Add-on. You'd probably still use rtl-433-mqtt autodiscovery from this repo. Have a look at https://github.com/hertzg/rtl_433_docker.

haraldhh, 2023-10-26:
Got it thanks. Running the python-script rtl_433_mqtt_hass.py  by itself  in CLI seems to work fine.

unverbraucht, 2023-10-27:
Awesome. Can you close the issue?

haraldhh, 2023-10-27:
Sorry, sure. Thanks

closed 2023-10-27 by haraldhh