pbkhrv/rtl_433-hass-addons #167: 0.4.1 Stopped receiving

sciurius, 2023-10-30:
### The problem

After the upgrade to 0.4.1 the addon stopped receiving/transmitting information.

I've been using the rtl_433 addon for more than a 18 months with a USB2.0 DAB FM Radio DVB-T RTL2832U R820T2 RTL SDR TV Stick Dongle Digital USB TV HDTV Tuner Receiver IR Remote with Antenna (sic). The dongle is in the T620 that runs HomeAssistant, with an external antenna.

After the latest upgrade (a couple of days ago) the addon doesn't receive/transmit information. RTL_433 starts up, configures itself to the desired protocols and frequency, connects to the MQTT broker, and then nothing.

When I plug the dongle in another (linux) system and run rtl_433 with the same config it works fine.

### What addon are you reporting the bug for?

rtl_443

### What is the addon version?

0.4.1

### What type of MQTT Broker are you using?

Other (details in the bug description)

### Addon log messages

```txt
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
[08:09:20] WARNING: rtl_433 now supports automatic configuration and multiple radios. The rtl_433_conf_file option is deprecated. See the documentation for migration instructions.
Starting rtl_433 -c /config/rtl_433/rtl_433.conf
rtl_433 version 22.11 branch  at 202211191645 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.

New defaults active, use "-Y classic -s 250k" for the old defaults!

Publishing MQTT data to mqtt.squirrel.nl port 1883
Publishing device info to MQTT topic "rtl_433/bresser51".
Registered 2 out of 223 device decoding protocols [ 172 119 ]
Found Rafael Micro R820T/2 tuner
Exact sample rate is: 250000.000414 Hz
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 868.300MHz.
Allocating 15 zero-copy buffers
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 50000 Hz, 20.0 us
MQTT Connected...
```

### Additional information

Mosquitto add-on runs as an independent broker.

rtl_433 config:
````
gain          0
frequency     868.30M
hop_interval  600
ppm_error     0
sample_rate   250k
report_meta   time:tz
protocol 172 # Bresser Weather Center 6-in-1 (newer 5-in-1)
protocol 119 # Bresser Weather Center 5-in-1 (old)
output kv
output mqtt=http://mqtt.squirrel.nl,user=****,pass=****,retain=1,devices=rtl_433/bresser51
````

Zbyszek1255, 2023-10-30:
I have that same problem

gitlaman, 2023-10-30:
I have the same problem after updating to 0.4.1, i stopped receiving from my AmbientWeather-WH31E. After restoring from partial backup to 0.4.0 the sensors started receiving the temperature and humidity values.

sciurius, 2023-10-30:
As a starter, it would be nice if the repo would archive previous versions. I can't seem to find the addon in my backups.

gitlaman, 2023-10-30:
> As a starter, it would be nice if the repo would archive previous versions. I can't seem to find the addon in my backups.

When i do addon update Homeassistant have option that is auto checked for auto backup, after that you can partial restore the addon through System/Backups.

sciurius, 2023-10-30:
True. I was looking in the wrong place ;).

deviantintegral, 2023-10-30:
My lacrosse weather stations are working for me. I haven’t upgraded my other location yet. What’s odd is that we upgraded the base docker image, but didn’t change the rtl_433 version at all.

Zbyszek1255, 2023-10-30:
For me, after returning to versions 0.4.0 and 0.6.0, the Bresser station still does not report temperatures. Currently I only have -42.1dB. Even if I disconnect the antenna, I have the same logs

sciurius, 2023-10-30:
After restoring the backup to 0.4.0 everything works fine again (Bresser 5-in-1).

Chartreusito, 2023-10-30:
What i really don't get here, is that i followed @sciurius blog to add a bresser 5in1 station, and ended up using next branch of the integration cause it was the one working for me. And it's still working ok for me now. Thanks again Johan !

sciurius, 2023-10-31:
@Chartreusito What version of the add-on and rtl_433 is working for you?

Chartreusito, 2023-10-31:
> @Chartreusito What version of the add-on and rtl_433 is working for you?

According to my add-on logs, my rtl_433 next in using this build : `rtl_433 version nightly-8-g76423bcc branch master at 202309201853 inputs file rtl_tcp RTL-SDR`
My rtl_433 MQTT Auto Discovery Add-On is up-to-date and using version: 0.7.0

Sensor wise i am NOT using your initial yaml though, i ended up keeping the entities populated by auto discovery which are named like this : `sensor.bresser_5in1_233_gs`

sciurius, 2023-10-31:
Interesting. I tried rtl433(next) and this is what I got in the log.

First the rtl_433 (0.4.0) log:
````
[11:17:09] WARNING: rtl_433 now supports automatic configuration and multiple radios. The rtl_433_conf_file option is deprecated. See the documentation for migration instructions.
Starting rtl_433 -c /config/rtl_433/rtl_433.conf
rtl_433 version 22.11 branch  at 202211191645 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.

New defaults active, use "-Y classic -s 250k" for the old defaults!

Publishing MQTT data to mqtt.squirrel.nl port 1883
Publishing device info to MQTT topic "rtl_433/bresser51".
Registered 2 out of 223 device decoding protocols [ 172 119 ]
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 868.300MHz.
Allocating 15 zero-copy buffers
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 50000 Hz, 20.0 us
MQTT Connected...
MQTT Connection established.
````

Now the (next) log, after stopping 0.4.0 and starting (next):
````
[11:14:59] WARNING: rtl_433 now supports automatic configuration and multiple radios. The rtl_433_conf_file option is deprecated. See the documentation for migration instructions.
Starting rtl_433 -c /config/rtl_433/rtl_433.conf
rtl_433 version 22.11-242-g3a9c61b1 branch master at 202310301815 inputs file rtl_tcp RTL-SDR

New defaults active, use "-Y classic -s 250k" if you need the old defaults

MQTT: Publishing MQTT data to mqtt.squirrel.nl port 1883
MQTT: Publishing device info to MQTT topic "rtl_433/bresser51".
Found Rafael Micro R820T/2 tuner
[SDR] Using device 0: Realtek, RTL2838UHIDIR, SN: 00000001, "Generic RTL2832U OEM"
Exact sample rate is: 250000.000414 Hz
Allocating 15 zero-copy buffers
````

It doesn't work. It seems as if it skips part of the config file (registered 2 protocols, tuned to 868MHz).
Also, there is no "MQTT connected".

sciurius, 2023-10-31:
Update: It does seem to register the two protocols (I can see that when I run with verbose 7), but I'm pretty sure it does not tune to the 868.30M frequency (so it receives no bresser data).

Does anyone knows what the bug fat (orange) warning means?
````
WARNING: rtl_433 now supports automatic configuration and multiple radios. The rtl_433_conf_file option is deprecated. See the documentation for migration instructions.
````
I could not find any migration instructions.

sciurius, 2023-10-31:
Ok, I think I found the culprit. For frequency 868.3M (in general, over 800M) a .0, rate of 1024k is required. This is handled in the code but can be overruled with a `sample_rate` line in the config. Older versions of rtl_433 seemed to cope with this, but newer do not. In other words, this works:
````
frequency 868.3M
````
because the sample_rate will be set to 1024k. This also works:
````
sample_rate 250k
frequency 868.3M
````
since the frequency setting overrules the sample rate. This does not work
````
frequency 868.3M
sample_rate 250k
````

Now I have 0.4.0, 0.4.1 and next all working properly.

pbabilas, 2023-10-31:
For me nothing works, not even 433. My last status when it was ok:
https://share.merce.com/files/7e469b33-a2fa-49d7-b57b-01889eeeaa17.png

Tried even run it on official docker image, and 433 some times works. I'm consider if my dongle is not broken :(

deviantintegral, 2023-10-31:
@sciurius re the warning, it was a change made back in 2021 so it’s deeper in the Changelog: https://github.com/pbkhrv/rtl_433-hass-addons/blob/main/rtl_433/CHANGELOG.md#013---2021-12-30

What I see here is some tricky behaviour. For some people the 0.4.0 release fixed things, for others it was just as broken, and for others 0.4.1 is working fine. I’m guessing those with issues have tried turning on verbose logging, but if not that may help give us a next step?

Or, if we conclusively knew that rtl_433 master is working well, we could ask upstream to tag a release or build from master in the main addon.

sciurius, 2023-11-01:
Thanks for your explanation of the warning and migration. It may be worth to add the migration info to the addon config screen. And emphasise that e.g. `rtl_433/bresser51.template` will not work, it needs to be `rtl_433/bresser51.conf.template`.

As for the versions, after removing the `sample_rate` line from my config, 0.4.0, 0.4.1 and next all work fine for me.
I'm not sure it solves Zbyszek1255's problem (the line `Sample rate set to 1000000 S/s.` indicates 伊 doesn't have a faulty sample rate). 

With the `sample_rate 250k` line following the frequency line 0.4.0 works but 0.4.1 and later break (silently).

ap20132022, 2023-11-02:
I have the same problem:
`s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
[10:17:43] WARNING: rtl_433 now supports automatic configuration and multiple radios. The rtl_433_conf_file option is deprecated. See the documentation for migration instructions.
Starting rtl_433 -c /config/rtl_433/rtl_433.conf.template
rtl_433 version 22.11 branch  at 202211191645 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Publishing MQTT data to 192.168.0.151 port 1883
Publishing device info to MQTT topic "rtl_433[/id]".
Registered 1 out of 223 device decoding protocols
Found 1 device(s)

trying device  0:  Realtek, RTL2838UHIDIR, SN: 00000001
Found Fitipower FC0012 tuner
Using device 0: Generic RTL2832U OEM
Sample rate set to 1024000 S/s.
Bit detection level set to 0.0 (Auto).
Tuner gain set to Auto.
Reading samples in async mode...
Tuned to 433.920MHz.
Allocating 15 zero-copy buffers
MQTT Connected...
MQTT Connection established.`

But nothing was sent over MQTT. 
My configuration:
```
frequency   433.920M
sample_rate   1024k

#protocol:
protocol 3   # Prologue, FreeTec NC-7104, NC-7159-675 temperature sensor
#protocol 25  # Globaltronics GT-WT-02 Sensor
#protocol 153 # Cotech 36-7959 wireless weather station with USB
#protocol 19 # Nexus
#protocol 173 # Bresser Weather Center 7-in-1

convert     si
verbose 8
#output json
#output kv
#output log
output      mqtt://192.168.0.151:1883,user=xxx,pass=xxx,retain=0,devices=rtl_433[/id]
#analyze false

```
I've tried with 1024k and 250k sample rate. But nothing changes.

sciurius, 2023-11-02:
@ap20132022 Assuming this is 0.4.1, did it work with 0.4.0?

sciurius, 2023-11-02:
> Thanks for your explanation of the warning and migration. It may be worth to add the migration info to the addon config screen. And emphasise that e.g. `rtl_433/bresser51.template` will not work, it needs to be `rtl_433/bresser51.conf.template`.

Actually it would be even better to remove the startup warning and add a note to the `rtl433_conf_file` config item in the UI similar to

Using the config file setting restricts rtl_433 to a single radio device only. If you have multiple devices create a directory `/config/rtl_433` and place a config template (a file with extension `.conf.template`) for each of the devices. The config template may use environment variables.

Just a suggestion.

deviantintegral, 2023-11-02:
> Actually it would be even better to remove the startup warning and add a note to the rtl433_conf_file config item in the UI similar to

I agree, but unfortunately there's no description key in the options schema: https://developers.home-assistant.io/docs/add-ons/configuration/#options--schema

Our addon isn't that far off from the docker containers at https://github.com/hertzg/rtl_433_docker. Can anyone replicate similar issues using it, specifically with the alpine images?

deviantintegral, 2023-11-02:
What didn't work? Some error, or did you get the new version somehow?

If your backup is corrupt or broken, you can create a new backup of the addon, download it, extract it, and then manually change the version (it may be listed multiple times). Re-compress, and restore from backup. I haven't actually tested this myself.

raffaelevall, 2023-11-02:
Hi, I have the same issue and the backup restore don't work for me.
`
Starting rtl_433 with rtl_433.conf...
[rtl_433] rtl_433 version 22.11 branch  at 202211191645 inputs file rtl_tcp RTL-SDR
[rtl_433] Use -h for usage help and see https://triq.org/ for documentation.
[rtl_433] Publishing MQTT data to core-mosquitto port 1883
[rtl_433] Publishing device info to MQTT topic "rtl_433/9b13b3f4-rtl433/devices[/type][/model][/subtype][/channel][/id]".
[rtl_433] Publishing events info to MQTT topic "rtl_433/9b13b3f4-rtl433/events".
[rtl_433] Publishing states info to MQTT topic "rtl_433/9b13b3f4-rtl433/states".
[rtl_433] 
[rtl_433] New defaults active, use "-Y classic -s 250k" for the old defaults!
[rtl_433] 
[rtl_433] Registering protocol [119] "Bresser Weather Center 5-in-1"
[rtl_433] Registering protocol [172] "Bresser Weather Center 6-in-1, 7-in-1 indoor, soil, new 5-in-1, 3-in-1 wind gauge, Froggit WH6000, Ventus C8488A"
[rtl_433] Registering protocol [173] "Bresser Weather Center 7-in-1"
[rtl_433] Publishing MQTT data to 192.168.1.120 port 1883
[rtl_433] Publishing device info to MQTT topic "rtl_433[/id]".
[rtl_433] Registered 3 out of 223 device decoding protocols
[rtl_433] Found 1 device(s)
[rtl_433] 
[rtl_433] trying device  0:  Realtek, RTL2838UHIDIR, SN: 00000001
[rtl_433] Found Fitipower FC0012 tuner
[rtl_433] Using device 0: Generic RTL2832U OEM
[rtl_433] Sample rate set to 1024000 S/s.
[rtl_433] Bit detection level set to 0.0 (Auto).
[rtl_433] Tuner gain set to Auto.
[rtl_433] Reading samples in async mode...
[rtl_433] Tuned to 868.300MHz.
[rtl_433] Allocating 15 zero-copy buffers
[rtl_433] MQTT Connected...
[rtl_433] baseband_demod_FM: low pass filter for 1024000 Hz at cutoff 204800 Hz, 4.9 us
[rtl_433] MQTT Connection established.
[rtl_433] MQTT Connected...
[rtl_433] MQTT Connection established.
[rtl_433] Current noise level -42.1 dB, estimated noise -42.1 dB
`

tks

Siwus90, 2023-11-02:
Same problem, after update it died.

`s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
[00:30:56] WARNING: rtl_433 now supports automatic configuration and multiple radios. The rtl_433_conf_file option is deprecated. See the documentation for migration instructions.
Starting rtl_433 -c /config//rtl_433/rtl_433.conf.template
rtl_433 version 22.11 branch  at 202211191645 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Publishing MQTT data to 192.168.68.108 port 1883
Publishing device info to MQTT topic "rtl_433/9b13b3f4-rtl433/devices[/type][/model][/subtype][/channel][/id]".
Publishing events info to MQTT topic "rtl_433/9b13b3f4-rtl433/events".
Publishing states info to MQTT topic "rtl_433/9b13b3f4-rtl433/states".

New defaults active, use "-Y classic -s 250k" for the old defaults!

Registered 1 out of 223 device decoding protocols [ 119 ]
Found Fitipower FC0012 tuner
Exact sample rate is: 1000000.026491 Hz
Sample rate set to 1000000 S/s.
Tuner gain set to Auto.
Tuned to 868.300MHz.
Allocating 15 zero-copy buffers
MQTT Connected...
baseband_demod_FM: low pass filter for 1000000 Hz at cutoff 200000 Hz, 5.0 us
MQTT Connection established.`

sciurius, 2023-11-02:
> Just a suggestion.

Also, that there's no such thing as "automatic configuration". You still need to set the frequency and protocols.

pbabilas, 2023-11-02:
ok, i've officially tried everything :(

My config file:
`frequency     868.3M
frequency     433.92M
hop_interval  30
protocol 172 # Bresser Weather Center 6-in-1 (newer 5-in-1)
protocol 119
protocol 8
protocol 73
protocol 75
protocol 76
protocol 124
protocol 166
protocol 170
protocol 171
output mqtt=http://core-mosquitto:1883,user=rtl,pass=***,retain=1,devices=rtl_433[/model]
`

had gain 0, sampla rate 250k, changed order, restored backup to v4.0, not getting any signal reads. Before when I started without protocols there was many devices shown, and now nothing. I've also ordered new usb stick and the same result.

Any idea what to do next?

sciurius, 2023-11-02:
`Gain 0` may be a good thing to add (keep).
You're tuned to a single frequency, so `hop_interval` makes no sense (remove it).
Adding `output kv` shows anything received (in case the MQTT communication is the problem).
What is in the addon log when you run the addon? Add `verbose 2` for more info.

My next step would be to plug the stick in another computer and run rtl_433 there to see if the hardware is ok. Also, AFAIK, these usb dongles usually won't work without an external antenna.

pbabilas, 2023-11-03:
This config is something I've ended up after many configuration: Now it looks like that with kv:

```gain          0
frequency     868.30M
frequency     433.92M
hop_interval  30
ppm_error     0
sample_rate   250k
report_meta   time:tz
protocol 172 # Bresser Weather Center 6-in-1 (newer 5-in-1)
protocol 119
protocol 8
protocol 73
protocol 75
protocol 76
protocol 124
protocol 166
protocol 170
protocol 171
output kv```

All that it can find is:

```s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
[06:59:03] WARNING: rtl_433 now supports automatic configuration and multiple radios. The rtl_433_conf_file option is deprecated. See the documentation for migration instructions.
Starting rtl_433 -c /config/rtl_433.conf
rtl_433 version 22.11 branch  at 202211191645 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.

New defaults active, use "-Y classic -s 250k" for the old defaults!

Registered 10 out of 223 device decoding protocols [ 172 119 8 73 75-76 124 166 170-171 ]
Found Fitipower FC0012 tuner
Exact sample rate is: 250000.000414 Hz
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 868.300MHz.
Allocating 15 zero-copy buffers
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 50000 Hz, 20.0 us
Tuned to 433.920MHz.
Allocating 15 zero-copy buffers
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
center_frequency: 433920000                        frequencies: 868300000, 433920000                     hop_times : 30
Tuned to 868.300MHz.
Allocating 15 zero-copy buffers
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
center_frequency: 868300000                        frequencies: 868300000, 433920000                     hop_times : 30```

Tried also stick on another pc and the same result - whats funny I've already bought new one stick to see if its broken but the same result :o

On both devices was able to recive signal from 433 once cant repeat it with the same configuration anymore:

```(☸ |pluto:default) b-code@bcode-XPS  ~  sudo rtl_433 -f 433M -F json
rtl_433 version 22.11 (2022-11-19) inputs file rtl_tcp RTL-SDR SoapySDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/root/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 191 out of 223 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-197 199 201-215 217-223 ]
Detached kernel driver
Found Fitipower FC0012 tuner
Exact sample rate is: 250000.000414 Hz
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.000MHz.
Allocating 15 zero-copy buffers
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
{"time" : "2023-11-02 21:23:50", "model" : "LaCrosse-TX141Bv3", "id" : 189, "channel" : 1, "battery_ok" : 0, "temperature_C" : 14.100, "test" : "No"}
^CSignal caught, exiting!
Reattached kernel driver
```

Thought that maybe something is blocking my radio signal but bresser reciver is working ok.
Don't know whats next :(

sciurius, 2023-11-03:
@pbabilas Removing the sample_rate from the config did the trick for me.

Apparently you are monitoring multiple devices similtaneously. You could try one at a time.

pbabilas, 2023-11-03:
As i wrote before, I've tried already everything ;(
With only one freq and no sample_rate:

```s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
Starting rtl_433 -c /config/rtl_433.conf
[07:55:18] WARNING: rtl_433 now supports automatic configuration and multiple radios. The rtl_433_conf_file option is deprecated. See the documentation for migration instructions.
rtl_433 version 22.11 branch  at 202211191645 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Registering protocol [172] "Bresser Weather Center 6-in-1, 7-in-1 indoor, soil, new 5-in-1, 3-in-1 wind gauge, Froggit WH6000, Ventus C8488A"
Registering protocol [119] "Bresser Weather Center 5-in-1"
Registering protocol [8] "LaCrosse TX Temperature / Humidity Sensor"
Registering protocol [73] "LaCrosse TX141-Bv2, TX141TH-Bv2, TX141-Bv3, TX141W, TX145wsdth, (TFA, ORIA) sensor"
Registering protocol [75] "LaCrosse TX35DTH-IT, TFA Dostmann 30.3155 Temperature/Humidity sensor"
Registering protocol [76] "LaCrosse TX29IT, TFA Dostmann 30.3159.IT Temperature sensor"
Registering protocol [124] "LaCrosse/ELV/Conrad WS7000/WS2500 weather sensors"
Registering protocol [166] "LaCrosse Technology View LTV-WSDTH01 Breeze Pro Wind Sensor"
Registering protocol [170] "LaCrosse Technology View LTV-WR1 Multi Sensor"
Registering protocol [171] "LaCrosse Technology View LTV-TH Thermo/Hygro Sensor"
Registered 10 out of 223 device decoding protocols
Found 1 device(s)

trying device  0:  Realtek, RTL2838UHIDIR, SN: 00000001
Found Fitipower FC0012 tuner
Using device 0: Generic RTL2832U OEM
Exact sample rate is: 250000.000414 Hz
Sample rate set to 250000 S/s.
Bit detection level set to 0.0 (Auto).
Tuner gain set to Auto.
Reading samples in async mode...
Tuned to 433.000MHz.
Allocating 15 zero-copy buffers
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
```

Nothing happen, it make no sense ;(

sciurius, 2023-11-03:
I see a frequency of 433.00MHz, shouldn't this be 433.92MHz?
I also see a sample rate of 250k, which seems not to work for 868.3MHz.

You could try the tool `gqrx` to see if signals are received.

pbabilas, 2023-11-03:
I will not even pretend that I know what i'm doing :P

https://share.merce.com/files/cb839cef-c0f8-422d-8685-c627979efad4.png

sciurius, 2023-11-03:
These are the Bresser signals.
[image]

sciurius, 2023-11-03:
When you run `rtl_433 -c /dev/null -f 868.3M` you should get some output.
````
% rtl_433 -c /dev/null -f 868.3M
rtl_433 version 22.11 (2022-11-19) inputs file rtl_tcp RTL-SDR SoapySDR
Use -h for usage help and see https://triq.org/ for documentation.

New defaults active, use "-Y classic -s 250k" if you need the old defaults

[Protocols] Registered 211 out of 245 device decoding protocols [ 1-4 8 10-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119-121 124-128 130-149 151-161 163-168 170-175 177-197 199 201-215 217-228 230-232 234-241 243-244 ]
Found Rafael Micro R820T tuner
[SDR] Using device 0: Realtek, RTL2838UHIDIR, SN: 00000001, "Generic RTL2832U OEM"
Exact sample rate is: 1000000.026491 Hz
[R82XX] PLL not locked!
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-11-03 10:41:25
model     : Bresser-6in1 id        : 14100077
channel   : 0            Sensor type: 1            Wind Gust : 1.2 m/s
Wind Speed: 1.2 m/s      Direction : 68            Rain      : 1350.8 mm
Startup   : 1            Flags     : 1             Integrity : CRC
````

pbabilas, 2023-11-03:
so it seams not ok:
```~ # rtl_433 -c /dev/null -f 868.3M
rtl_433 version 22.11 (2022-11-19) inputs file rtl_tcp RTL-SDR with TLS
Use -h for usage help and see https://triq.org/ for documentation.

New defaults active, use "-Y classic -s 250k" for the old defaults!

Registered 191 out of 223 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-197 199 201-215 217-223 ]
Found Fitipower FC0012 tuner
Exact sample rate is: 1000000.026491 Hz
Sample rate set to 1000000 S/s.
Tuner gain set to Auto.
Tuned to 868.300MHz.
Allocating 15 zero-copy buffers
baseband_demod_FM: low pass filter for 1000000 Hz at cutoff 200000 Hz, 5.0 us
```

Protocols count is not the same

pbabilas, 2023-11-03:
maybe there happened somethink impossible and my new dongle has the same error as the old one xD becouse its just ridiculous. I think that I will try to order new one and check if this helps :D

pbabilas, 2023-11-03:
Oo one diffrence, why You have:

`rtl_433 version 22.11 (2022-11-19) inputs file rtl_tcp RTL-SDR **SoapySDR**`

But I'm heaving:

`rtl_433 version 22.11 (2022-11-19) inputs file rtl_tcp RTL-SDR **with TLS**`

raffaelevall, 2023-11-03:
> forse è successo qualcosa di impossibile e il mio nuovo dongle ha lo stesso errore di quello vecchio xD perché è semplicemente ridicolo. Penso che proverò a ordinarne uno nuovo e controllerò se questo aiuta: D

It would be strange if all our dongles broke at the same time

sciurius, 2023-11-03:
Yes, I noticed. Unfortunately I really don't know what that implies :( .

pbabilas, 2023-11-03:
> > forse è successo qualcosa di impossibile e il mio nuovo dongle ha lo stesso errore di quello vecchio xD perché è semplicemente ridicolo. Penso che proverò a ordinarne uno nuovo e controllerò se questo aiuta: D
> 
> It would be strange if all our dongles broke at the same time

agreed, but has no other idea.

pbabilas, 2023-11-03:
What kind of hardware you own? Maybe i will buy something other than my acctual?

sciurius, 2023-11-03:
My dongle is https://www.amazon.nl/NESDR-Mini-ADS-B-ontvangerset-SDR-softwarepakketten-antenne-ingang/dp/B009U7WZCA .
I've been told that RTL2832U & R820T-tuner is a good combo.

deviantintegral, 2023-11-04:
> Oo one diffrence, why You have:
> 
> rtl_433 version 22.11 (2022-11-19) inputs file rtl_tcp RTL-SDR **SoapySDR**
> 
> But I'm heaving:
> 
> rtl_433 version 22.11 (2022-11-19) inputs file rtl_tcp RTL-SDR **with TLS**

Neither one of these match the current version string in the 0.4.1 release of the addon, which is:

> rtl_433 version 22.11 branch  at 202211191645 inputs file rtl_tcp RTL-SDR

I'm assuming that's from testing outside the addon, which is great for troubleshooting, but **if not** please comment because that would perhaps give us a next step on figuring things out.

I'll share a personal anecdote just to show how tricky debugging this sort of situation is. I have a soil sensor that wasn't reaching normal rtl_433 setup, so I purchased a Raspberry Pi 0 to relay the signal over wifi. I had all sorts of trouble, ranging from USB resets to long periods of time where no data was sent. I purchased an RTL-SDR blog v4 radio to try... and then everything that was broken started working again. It's been solid for 3-4 months now! If this had happened around the same time as a new addon release, I'm sure I would have thought it was somehow involved.

sciurius, 2023-11-04:
The SoapySDR is the one running on my Fedora workstation, rtl-433-22.11-1.20230714git37b804c.fc37.x86_64.

ap20132022, 2023-11-05:
has anyone idea, how I can install the old version (downgrade) to 0.4.0? Thank you very much.

pbabilas, 2023-11-05:
> has anyone idea, how I can install the old version (downgrade) to 0.4.0? Thank you very much.

You can uae restore from backup or local install, just download zip from github to /addons dir unzip it and in addos store refresh repositories.

raffaelevall, 2023-11-05:
> Qualcuno ha idea di come posso installare la vecchia versione (downgrade) alla 0.4.0? Grazie mille.

I had restored the backup and it hadn't worked. Restored again, unplugged the dongle and ran it without the dongle. After connecting the dongle and restarting the add-on it started working again. Malfunction of the dongle must be excluded

Siwus90, 2023-11-05:
> I had restored the backup and it hadn't worked. Restored again, unplugged the dongle and ran it without the dongle. After connecting the dongle and restarting the add-on it started working again. [...]

Thank you, it worked!

raffaelevall, 2023-11-05:
> Thank you, it worked!

A crazy workaround 😂😂😂

ap20132022, 2023-11-05:
> I had restored the backup and it hadn't worked. Restored again, unplugged the dongle and ran it without the dongle. After connecting the dongle and restarting the add-on it started working again. [...]

similar to me. After restoring to 0.4.0 it didn't work. After reconnecting USB device, it works. I'm happy now. After 4 day sleepness, it works again.

pbabilas, 2023-11-06:
for me not working. after unplug usb drive and plug it again it is not recognized until I reboot vm... ;(

found more logs:
```[00] {280} 2a aa aa aa aa 8b 75 39 61 df f8 5f f6 3b f6 7f a5 de fa ff c6 9e 20 07 a0 09 c4 09 80 5a 21 05 00 00 00 
lacrosse_breezepro_decode: Sync word not found
lacrosse_wr1_decode: Packet too long: 280 bits
lacrosse_th_decode: packet length: 280
lacrosse_th_decode: Sync word not found
lacrosse_breezepro_decode: Wrong packet length: 155
lacrosse_wr1_decode: packet length: 155
lacrosse_wr1_decode: Sync word not found
lacrosse_th_decode: Packet too short: 155 bits
Tuned to 433.920MHz.
Allocating 15 zero-copy buffers
pulse_slicer_pwm(): LaCrosse TX141-Bv2, TX141TH-Bv2, TX141-Bv3, TX141W, TX145wsdth, (TFA, ORIA) sensor
bitbuffer:: Number of rows: 4 
[00] {33} bd 12 64 8a 80 : 10111101 00010010 01100100 10001010 1
[01] {33} bd 12 64 8a 80 : 10111101 00010010 01100100 10001010 1
[02] {33} bd 12 64 8a 80 : 10111101 00010010 01100100 10001010 1
[03] {33} bd 12 64 8a 80 : 10111101 00010010 01100100 10001010 1
pulse_slicer_pwm(): LaCrosse TX141-Bv2, TX141TH-Bv2, TX141-Bv3, TX141W, TX145wsdth, (TFA, ORIA) sensor
bitbuffer:: Number of rows: 4 
[00] {33} bd 12 64 8a 80 : 10111101 00010010 01100100 10001010 1
[01] {33} bd 12 64 8a 80 : 10111101 00010010 01100100 10001010 1
[02] {33} bd 12 64 8a 80 : 10111101 00010010 01100100 10001010 1
[03] {33} bd 12 64 8a 80 : 10111101 00010010 01100100 10001010 1
Tuned to 868.300MHz.
Allocating 15 zero-copy buffers
lacrosse_breezepro_decode: Wrong packet length: 156
lacrosse_wr1_decode: packet length: 156
lacrosse_wr1_decode: Sync word not found
lacrosse_th_decode: packet length: 156
lacrosse_th_decode: Sync word not found
pulse_slicer_pcm(): Bresser Weather Center 5-in-1
bitbuffer:: Number of rows: 1 
[00] {279} 55 55 55 55 55 16 ea 74 43 bf de a7 e7 f7 ed ff 4b bd f5 ff 8b bc 40 21 58 18 08 12 00 b4 42 0a 00 00 00 
lacrosse_breezepro_decode: Sync word not found
lacrosse_wr1_decode: Packet too long: 279 bits
lacrosse_th_decode: packet length: 279
lacrosse_th_decode: Sync word not found
pulse_slicer_pcm(): Bresser Weather Center 5-in-1
bitbuffer:: Number of rows: 1 
[00] {281} 15 55 55 55 55 45 ba 9c d0 ef f9 b3 f9 9d fb 7f d2 ef 7d 7f e3 2f 10 06 4c 06 62 04 80 2d 10 82 80 00 00 00 
lacrosse_breezepro_decode: Sync word not found
lacrosse_wr1_decode: Packet too long: 281 bits
lacrosse_th_decode: packet length: 281
lacrosse_th_decode: Sync word not found
Tuned to 433.920MHz.
Allocating 15 zero-copy buffers
pulse_slicer_pwm(): LaCrosse TX141-Bv2, TX141TH-Bv2, TX141-Bv3, TX141W, TX145wsdth, (TFA, ORIA) sensor
bitbuffer:: Number of rows: 4 
[00] {33} bd 12 64 8a 80 : 10111101 00010010 01100100 10001010 1
[01] {33} bd 12 64 8a 80 : 10111101 00010010 01100100 10001010 1
[02] {33} bd 12 64 8a 80 : 10111101 00010010 01100100 10001010 1
[03] {33} bd 12 64 8a 80 : 10111101 00010010 01100100 10001010 1
pulse_slicer_pwm(): LaCrosse TX141-Bv2, TX141TH-Bv2, TX141-Bv3, TX141W, TX145wsdth, (TFA, ORIA) sensor
bitbuffer:: Number of rows: 4 
[00] {33} bd 12 64 8a 80 : 10111101 00010010 01100100 10001010 1
[01] {33} bd 12 64 8a 80 : 10111101 00010010 01100100 10001010 1
[02] {33} bd 12 64 8a 80 : 10111101 00010010 01100100 10001010 1
[03] {33} bd 12 64 8a 80 : 10111101 00010010 01100100 10001010 1
Tuned to 868.300MHz.
Allocating 15 zero-copy buffers
pulse_slicer_pcm(): Bresser Weather Center 5-in-1
bitbuffer:: Number of rows: 1 
[00] {280} 2a aa aa aa aa 8b 75 3a 21 df fd db f7 bb f7 3f a5 9e fa ff c5 de 20 02 24 08 44 08 c0 5a 61 05 00 00 00 
lacrosse_breezepro_decode: Sync word not found
lacrosse_wr1_decode: Packet too long: 280 bits
lacrosse_th_decode: packet length: 280
lacrosse_th_decode: Sync word not found
lacrosse_breezepro_decode: Wrong packet length: 156
lacrosse_wr1_decode: packet length: 156
lacrosse_wr1_decode: Sync word not found
lacrosse_th_decode: packet length: 156
lacrosse_th_decode: Sync word not found
pulse_slicer_pcm(): Bresser Weather Center 5-in-1
bitbuffer:: Number of rows: 1 
[00] {282} 0a aa aa aa aa a2 dd 4e 88 77 ff 76 fd ee fd cf e9 67 be bf f1 77 88 00 89 02 11 02 30 16 98 41 40 00 00 00 
lacrosse_breezepro_decode: Sync word not found
lacrosse_wr1_decode: Packet too long: 282 bits
lacrosse_th_decode: packet length: 282
lacrosse_th_decode: Sync word not found
Tuned to 433.920MHz.
Allocating 15 zero-copy buffers
```

It means that get signal but can't decode?

deviantintegral, 2023-11-06:
Is there anyone where some combination of rebooting or re-plugging the radio hasn't fixed this?

I also just came across https://github.com/merbanan/rtl_433/issues/2426 (which is blocking the next stable tag of rtl_433) and wonder if this is at all related.

deviantintegral, 2024-02-17:
I think this issue is fixed with the release I just cut for those not using the next addon. Feel free to open a new issue if you are still having hangs.

closed 2024-02-17 by deviantintegral

Chartreusito, 2024-02-17:
Just switched to stable release instead of next to find things working again, thanks !

pbabilas, 2024-02-18:
Damn, after upgrade , failed to work again :(  need to try replug again :(

Chartreusito, 2024-02-18:
> Damn, after upgrade , failed to work again :( need to try replug again :(

Are you using a mains powered usb hub ? If not, i definitely advice you to try with that, it draws a lot of power and a powered hub seems to help.

pbabilas, 2024-02-22:
> > Damn, after upgrade , failed to work again :( need to try replug again :(
> 
> Are you using a mains powered usb hub ? If not, i definitely advice you to try with that, it draws a lot of power and a powered hub seems to help.

Wow, it's really fixed the issue. I've replug it to slot with more power and now it works. Thanks for advice!

pbabilas, 2024-02-26:
Was happy to early ;( Replug it worked only on v 0.4, after upgrade failed again. Need to check eboot and replug radio way again on new and old version.

Chartreusito, 2024-02-26:
> Was happy to early ;( Replug it worked only on v 0.4, after upgrade failed again. Need to check eboot and replug radio way again on new and old version.

I'd recommend you to switch to one of these, i got one of second hand for 7€ : 
https://www.amazon.com/Sabrent-Individual-Switches-Included-HB-UMP3/dp/B00TPMEOYM/

It survived every reboot since.

pbabilas, 2024-02-26:
my usb hub: https://www.tp-link.com/pl/home-networking/computer-accessory/uh700/
1.5A is not enought?

[3 more replies]