pbkhrv/rtl_433-hass-addons #168: Is it possible to Filter devices and add fields to data to InfluxDB ?

mitch55500, 2023-11-01:
### The problem

Hi all,

I have 2 questions/requests : 

- Is it possible to have the ouput (to influx) just some devices? Not all devices found by RTL_433 dongle?

- Always with data sent to my Influx DB, is it possible to add field like "years" or "months" (maybe in config file) to make easiest use in Grafana?

Thanks

### What addon are you reporting the bug for?

rtl_443

### What is the addon version?

The lastest

### What type of MQTT Broker are you using?

Other (details in the bug description)

deviantintegral, 2023-11-02:
> Is it possible to have the ouput (to influx) just some devices? Not all devices found by RTL_433 dongle?

Sure. You can configure Home Assistant's influx addon to only send some device data. But assuming you don't care about the data in home assistant either, the best thing to do is a combination of limiting protocols in the rtl_433 configuration, plus only running the discovery addon when actively adding new devices.
> Always with data sent to my Influx DB, is it possible to add field like "years" or "months" (maybe in config file) to make easiest use in Grafana?

I'm not sure what you're getting at here, but generally we emit ISO timestamps. That's what influx expects too - a single time field. However, I would expect you can convert the dates for display as needed with Grafana.

mitch55500, 2023-11-02:
Thanks for your answer.

`Sure. You can configure Home Assistant's influx addon to only send some device data. But assuming you don't care about the data in home assistant either, the best thing to do is a combination of limiting protocols in the rtl_433 configuration, plus only running the discovery addon when actively adding new devices.`

I hadn't think it, I'm going to do this (configure Home Assistant's influx addon).

`I'm not sure what you're getting at here, but generally we emit ISO timestamps. That's what influx expects too - a single time field. However, I would expect you can convert the dates for display as needed with Grafana.`

For example, I'd like to sort data by years or by months like this : 
[image]

I use influxDB v2. In Grafana, I can't do this (It might be possible but I don't know how to do it.).

In the config file, I have indicate : "report_meta time:iso:tz:local".

deviantintegral, 2023-11-04:
Oh! You can use something like `GROUP BY time(1w)` I think. I do this by day for electricity costs.

[image]

Here's the query as text. I'm still on influx 1.x, but I would be shocked if there wasn't equivalent functionality in the 2.x queries:

```
SELECT sum(mean) / 1000 / 3600 * $electrical_rate / 100  FROM (SELECT mean(value) FROM "W" WHERE $timeFilter GROUP BY friendly_name, time(1s) fill(previous)) GROUP BY friendly_name, time(1d);
```

closed 2024-02-17 by deviantintegral