pbkhrv/rtl_433-hass-addons #169: Auto Discovery publishes SCMPlus time value, skips the rest such as gas meter consumption. Case sensitive?

w2xg, 2023-11-02:
### The problem

Log from rtl_433

> time      : 2023-11-02T11:23:23.199319-0400
model     : SCMplus      id        : 78333333
Protocol_ID: 0x1E        Endpoint_Type: 0x9C       Endpoint_ID: 78333333     Consumption: 6474         Tamper    : 0x0A08        crc       : 0xCB82        Meter_Type: Gas           Integrity : CRC
>

(gas meter id changed to protect the innocent...)

Log from rtl_433 MQTT Auto Discovery:

> INFO:root:Published SCMplus/78333333: time
> INFO:root:Skipped SCMplus/78333333: ProtocolID, EndpointType, EndpointID, Consumption, Tamper, PacketCRC, MeterType

Device appearing as expected in hass but only the time, all the other entities are ignored. Consumption, and probably MeterType, are the desired data. Best guess is that it is skipping Consumption because of case?  

### What addon are you reporting the bug for?

rtl_443

### What is the addon version?

0.7.0

### What type of MQTT Broker are you using?

Home Assistant Mosquitto MQTT Broker

deviantintegral, 2023-11-04:
It looks like there's an issue upstream on this one: https://github.com/merbanan/rtl_433/issues/2207

catduckgnaf, 2024-02-19:
Hey! My auto discovery I don't push issues to upstream, 
You can easily fix the issue submit a PR, then use my add-on! easy
https://github.com/catduckgnaf/rtl_433_ha

IamTheCarl, 2024-04-12:
For anyone else who runs into this problem, you can work around it by using the rtl_433 addon without the auto-discovery addon and doing the discovery yourself by hand.

Just add a custom MQTT sensor:
```
mqtt:
  sensor:
    - name: "Mains Power"
      state_topic: "rtl_433/9b13b3f4-rtl433/devices/SCMplus/xxxxxxxx/Consumption"
      unit_of_measurement: 'kWh'
      force_update: true
      unique_id: "killowatt_hours_sensor"
      device_class: energy
      state_class: total
```