pbkhrv/rtl_433-hass-addons #170: Device added by Auto Discovery doesn't persist across reboots. "This entity is no longer being provided by the mqtt integration"

Jsalas424, 2023-11-04:
### The problem

Using the "next" version of both auto discovery rtl_433 I was able to add the [Govee water sensors to HA](https://github.com/merbanan/rtl_433/pull/2605). After reboot, the entity shows up as no longer provided and all downstream automatons fail even though messages were received. 

[image]

If I run rtl_433 MQTT Auto Discovery (next) again and trigger the wet sensor, it reappears and persists up until I reboot again. 

[image]

[image]

### What addon are you reporting the bug for?

rtl_433_mqtt_autodiscover

### What is the addon version?

next

### What type of MQTT Broker are you using?

Home Assistant Mosquitto MQTT Broker

deviantintegral, 2023-11-05:
Did you have the retain options set when running the auto discovery addon?

Jsalas424, 2023-11-05:
> Did you have the retain options set when running the auto discovery addon?

No. Problem solved, thanks for the help.

closed 2023-11-05 by Jsalas424