pbkhrv/rtl_433-hass-addons #175: New repository

catduckgnaf, 2024-01-12:
### The problem

Since this project is abandoned I have improved my completely rebuilt add-on

https://github.com/catduckgnaf/rtl_433_haos_addon

Right now the discovery filters don't work, but everything is extremely transparent and simple with a clear config file and easy to edit. The script is also automatically downloaded as part of this add-on until I can combine the two programs in a better way.

### What addon are you reporting the bug for?

rtl_443

### What is the addon version?

6

### What type of MQTT Broker are you using?

Home Assistant Mosquitto MQTT Broker

### Addon log messages

```txt
Since this project is abandoned I have improved my completely rebuilt add-on

https://github.com/catduckgnaf/rtl_433_haos_addon

Right now the discovery filters don't work, but everything is extremely transparent and simple with a clear config file and easy to edit. The script is also automatically downloaded as part of this add-on until I can combine the two programs in a better way.
```

deviantintegral, 2024-02-17:
Thanks for the update.

I had thought about forking the HASS bridge from rtl_433 at one point, but in truth I've found the upstream maintainers to be pretty open to PRs and updates. I think if we were to see lots of candidate PRs upstream that we're patching in here, it would be a good indication that upstream work isn't working.

I don't want to discourage your work - if anything, work that eventually leads to more modular code on the python side would be a huge win. If anyone ever takes the next step to make a proper integration (which I'm sure would be easier for users to set up), getting a python library spun up would be a key part of that. As you work, you may find ways you can start to share your work upstream in a way that even more users can benefit.

One other ask: I'd appreciate it if you stopped calling this repository abandoned here and in the discussion thread on the HA community. There's been commits, releases, and other activity all in the last year. Thank you.

closed 2024-02-17 by deviantintegral

catduckgnaf, 2024-02-17:
@deviantintegral 

I disagree with your first point. rtl_433 has the script under "examples" and they state the mission of their project is to be broad. You don't see PRs here, because this project is a mess, and the owner of it has left. There is no point, so your logic is flawed.

If you take a look at my work, you will see I opted to use mqtt, much like Zigbee2Mqtt. I have discovery built in with the add-on all in one. I did make extremely clear and simple and good sample configs, as ell as solving some major issues with discovery. I looked to an integration, but MQTT route seems like the way to go, and could always take it further later.

I will not do that, it is abandoned by its maintainer. He posted here many months ago saying he has no devices, and he essentially wanted someone to take over, and there has been nothing since.

I have a request for you. Can you please stop working on a project that the maintainer abandoned, is confusing, outdated, and complicated? Can we work together on the new version of the add-on. I could use your expertise and I would appreciate having you help. If you could post in the forums, that this is the better integration and direct people to mine, THAT WOULD be the best thing for the community.