pbkhrv/rtl_433-hass-addons #178: New version 0.8.0 not working

Eisbaer2, 2024-02-17:
### The problem

I just updated rtl_433 and rtl_433 MQTT Auto Discovery (new: 0.8.0). I use both to get data from my Bresser5in1 external weather sensor and show the data in HA.

With 0.8.0. the addon stops after a fews seconds. 

### What addon are you reporting the bug for?

rtl_433_mqtt_autodiscover

### What is the addon version?

0.8.0

### What type of MQTT Broker are you using?

Home Assistant Mosquitto MQTT Broker

### Addon log messages

```txt
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
Using an external mqtt broker.
Starting rtl_433_mqtt_hass.py...
[2024-02-17T10:53:24+0100] INFO:root:Enabling debug logging
[2024-02-17T10:53:24+0100] INFO:root:Discovering all devices
Exception ignored in: <function Client.__del__ at 0x7fb05d0900>
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/paho/mqtt/client.py", line 874, in __del__
    self._reset_sockets()
  File "/usr/local/lib/python3.12/site-packages/paho/mqtt/client.py", line 1133, in _reset_sockets
    self._sock_close()
  File "/usr/local/lib/python3.12/site-packages/paho/mqtt/client.py", line 1119, in _sock_close
    if not self._sock:
           ^^^^^^^^^^
AttributeError: 'Client' object has no attribute '_sock'
Traceback (most recent call last):
  File "/rtl_433_mqtt_hass.py", line 990, in <module>
    run()
  File "/rtl_433_mqtt_hass.py", line 917, in run
    rtl_433_bridge()
  File "/rtl_433_mqtt_hass.py", line 888, in rtl_433_bridge
    mqttc = mqtt.Client()
            ^^^^^^^^^^^^^
TypeError: Client.__init__() missing 1 required positional argument: 'callback_api_version'
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
```

### Additional information

I can see the data from the external sensor in MQTT explorer but the former HA entities are unavailable. Is there also a possibility to write the right config info for Bresser 5in1 somewhere in HA so that I don't need  rtl_433 MQTT Auto Discovery any longer?

Magicpulse, 2024-02-17:
Hi,
Same error here...

Skotniki17, 2024-02-17:
Same here after updating the addon

deviantintegral, 2024-02-17:
Thanks for the reports, looks like paho-mqtt released a new major version last week. PR is up at https://github.com/pbkhrv/rtl_433-hass-addons/pull/180.

deviantintegral, 2024-02-17:
0.8.1 is out fixing this, and I reported this upstream at https://github.com/merbanan/rtl_433/issues/2840.

catduckgnaf, 2024-02-19:
If you want an add-on that doesn't break because of issues with "upstream" use the new add-on that is better!

https://github.com/catduckgnaf/rtl_433_haos_addon

closed 2025-03-22 by deviantintegral