pbkhrv/rtl_433-hass-addons #181: Support RTL-SDR Blog V4 Dongle

jochakovsky, 2024-02-18:
### The problem

The RTL-SDR Blog V4 requires an updated version of librtlsdr (see the Debian section in https://www.rtl-sdr.com/v4/)

FYI, I was able to get the rtl_433 add-on working with my V4 hardware locally by changing the base image from 3.18 to 3.19, which upgrades librtlsdr from 0.8.0 to 2.0.1. Happy to submit a PR, but I wasn't sure if you'd want the base image updated in both add-ons or just rtl_433, or if you'd prefer a narrower method of upgrading librtlsdr.

### What addon are you reporting the bug for?

rtl_443

### What is the addon version?

master

### What type of MQTT Broker are you using?

Home Assistant Mosquitto MQTT Broker

deviantintegral, 2024-02-18:
I think it's fine to upgrade the base image, and to do it for the autodiscovery addon too. We'd have to upgrade alpine eventually anyways.

catduckgnaf, 2024-02-19:
My actively maintained add-on already supports V4. I encourage you and @deviantintegral  to continue development over there. For then we can truly work on a better add-on.

https://github.com/catduckgnaf/rtl_433_haos_addon

I have a better default rtl433_conf as well as editable discovery script built in!

https://github.com/catduckgnaf/rtl_433_haos_addon

deviantintegral, 2024-02-23:
Merged and fixed in next! Please feel free to reopen if it's not working.

PR #182 (2024-02-23): Update base Home Assistant Alpine image to 3.19
Fixes #181

## Summary

Adds support for RTL-SDR Blog V4 Dongle by upgrading base image, which includes an up to date version (2.0.1) of librtlsdr.

## Alternatives Considered

It would have been possible to fix this more narrowly by upgrading the base image only in rtl_433, or even by just upgrading librtlsdr. However, the base image would need to be upgraded eventually in any case.

## Testing Steps

Applied changes locally by copying updated version of addon to my Home Assistant Green running latest OS via Samba, with image field deleted from  [... cut]
commit 6bf5f2a8 (2024-02-23): Update base Home Assistant Alpine image to 3.19

Fixes #181
closed 2024-02-23 by deviantintegral