pbkhrv/rtl_433-hass-addons #187: Getting started

jay3702, 2024-04-02:
### The problem

I cannot get the add in to run. It's reading the config file because if I inject a nonsense configuration, it complains. I have an RTL SDR dongle on a USB port of a Raspberry Pi 4. 

It seems to start up with no errors, then immediately shuts down with no clue as to why. I'm sure this is an RTFM issue, but I'm stumped.

### What addon are you reporting the bug for?

rtl_443

### What is the addon version?

0.5.1

### What type of MQTT Broker are you using?

Home Assistant Mosquitto MQTT Broker

### Addon log messages

```txt
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
Starting rtl_433 with rtl_433.conf...
[rtl_433] rtl_433 version 23.11 branch  at 202311281352 inputs file rtl_tcp RTL-SDR
[rtl_433] MQTT: Publishing MQTT data to core-mosquitto port 1883
[rtl_433] MQTT: Publishing device info to MQTT topic "rtl_433/9b13b3f4-rtl433/devices[/type][/model][/subtype][/channel][/id]".
[rtl_433] MQTT: Publishing events info to MQTT topic "rtl_433/9b13b3f4-rtl433/events".
[rtl_433] MQTT: Publishing states info to MQTT topic "rtl_433/9b13b3f4-rtl433/states".
[rtl_433] Use "-F log" if you want any messages, warnings, and errors in the console.
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
```

jay3702, 2024-04-03:
Turns out to be a problem with the rtl-sdr dongle. I noticed some messages on the console about not being able to enumerate a USB device, so I tried another one, a Nooelectric E4000. The add in stars normally with that one.

deviantintegral, 2024-04-03:
Thanks for the update! If you still want to use the other one, a good test would be to put it on a powered USB hub. I had to do that for one I have on a Pi Zero.

closed 2024-04-03 by deviantintegral

jay3702, 2024-04-09:
Thanks, I picked up a powered USB hub, and that did the trick.

jay3702, 2024-04-10:
Also, the Nooelectric E4000 works fine without a powered hub once the configuration is correct.