pbkhrv/rtl_433-hass-addons #188: Config for device trigger conflicts with existing device trigger

invisible999, 2024-04-09:
### The problem

I am getting erros for every single ThremoPro temp/humidity sensors:

`Config for device trigger Thermopro-TX2-9-3-74 Thermopro-TX2-9-3-74-BTN conflicts with existing device trigger, cannot set up trigger, got: {'automation_type': 'trigger', 'type': 'button_short_release', 'subtype': 'button_1', 'topic': 'rtl_433/9b13b3f4-rtl433/devices/Thermopro-TX2/9/3/74/button', 'device': {'identifiers': ['Thermopro-TX2-9-3-74'], 'name': 'Thermopro-TX2-9-3-74', 'model': 'Thermopro-TX2', 'manufacturer': 'rtl_433', 'connections': []}, 'encoding': 'utf-8', 'qos': 0, 'payload': None, 'value_template': None}

Config for device trigger Thermopro-TX2-9-1-129 Thermopro-TX2-9-1-129-BTN conflicts with existing device trigger, cannot set up trigger, got: {'automation_type': 'trigger', 'type': 'button_short_release', 'subtype': 'button_1', 'topic': 'rtl_433/9b13b3f4-rtl433/devices/Thermopro-TX2/9/1/129/button', 'device': {'identifiers': ['Thermopro-TX2-9-1-129'], 'name': 'Thermopro-TX2-9-1-129', 'model': 'Thermopro-TX2', 'manufacturer': 'rtl_433', 'connections': []}, 'encoding': 'utf-8', 'qos': 0, 'payload': None, 'value_template': None}

Config for device trigger Thermopro-TX2C-1-8 Thermopro-TX2C-1-8-BTN conflicts with existing device trigger, cannot set up trigger, got: {'automation_type': 'trigger', 'type': 'button_short_release', 'subtype': 'button_1', 'topic': 'rtl_433/9b13b3f4-rtl433/devices/Thermopro-TX2C/1/8/button', 'device': {'identifiers': ['Thermopro-TX2C-1-8'], 'name': 'Thermopro-TX2C-1-8', 'model': 'Thermopro-TX2C', 'manufacturer': 'rtl_433', 'connections': []}, 'encoding': 'utf-8', 'qos': 0, 'payload': None, 'value_template': None}`

What I understood from the github issue page that this issue is fixed (or going to be fixed) at 'standard' RTL_433. When is going to be fixed in add-on?

https://github.com/merbanan/rtl_433/pull/2829

### What addon are you reporting the bug for?

rtl_443

### What is the addon version?

0.5.1

### What type of MQTT Broker are you using?

Home Assistant Mosquitto MQTT Broker

deviantintegral, 2024-04-11:
This is already available in the `next` version of the addon - that would be at most 24 hours from the upstream merge. Give it a try and make sure it works for you?

Otherwise, it will hit the stable addon once a new tag of rtl_433 is created. We could potentially patch the discovery script ahead of that release if it's affecting many people.

invisible999, 2024-04-12:
> This is already available in the `next` version of the addon - that would be at most 24 hours from the upstream merge. Give it a try and make sure it works for you?
>
> Otherwise, it will hit the [...]

Thanks. What is the timeframe for main (non-next) release incorporating the fix?

deviantintegral, 2024-04-16:
It mostly depends on when rtl_433 has a new stable release - we aim to track that unless there's a very good reason to not to.

deviantintegral, 2025-03-22:
rtl_433 25.02 is now in the stable addon. Thanks!