pbkhrv/rtl_433-hass-addons #189: rtl_433 does not start anymore

osmeest, 2024-04-14:
### The problem

I'm using a HomeAssistant Blue running under HA OS 11.1, with Supervisor 2024.04.0 and Core 2024.1.5.
rtl_433 0.5.1 add-on is installed, together with the Autodiscovery add-on and Mosquitto 6.4.0.
Mosquitto broker is also used with Zigbee2Mqtt and both work nicely together.

I just discovered that rtl_433 add-on does not remain running anymore. It starts but soon after, it stops.
The log gives:
```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
[12:40:11] INFO: The mqtt addon is not available.
[12:40:11] INFO: This is not a problem if you are using an external MQTT broker.
[12:40:11] INFO: If you are using the Home Assistant Mosquitto Broker addon, try restarting it, and then restart the rtl_433 addon.
[12:40:11] INFO: For an external broker, manually update the output line in the configuration file with mqtt connection settings, and restart the addon.
/tmp/rtl_433_heredoc: line 1: host: unbound variable
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
```

Just noticed the "/tmp/rtl_433_heredoc: line 1: host: unbound variable" error...
Unexpected, since I'm using the most basic configuration.

In the add-on config directory, there is a config template (which I don't remember having put by myself).
When I set the config file option with the name of the config template, I get:
```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
[12:26:58] INFO: The mqtt addon is not available.
[12:26:58] INFO: This is not a problem if you are using an external MQTT broker.
[12:26:58] INFO: If you are using the Home Assistant Mosquitto Broker addon, try restarting it, and then restart the rtl_433 addon.
[12:26:58] INFO: For an external broker, manually update the output line in the configuration file with mqtt connection settings, and restart the addon.
[12:26:58] WARNING: rtl_433 now supports automatic configuration and multiple radios. The rtl_433_conf_file option is deprecated. See the documentation for migration instructions.
Starting rtl_433 -c /config/rtl_433.conf.template
rtl_433 version 23.11 branch  at 202311281352 inputs file rtl_tcp RTL-SDR
Failed to stat "/config/rtl_433.conf.template"
Found Elonics E4000 tuner
Exact sample rate is: 250000.000414 Hz
[SDR] Using device 0: Realtek, RTL2838UHIDIR, SN: , "Generic RTL2832U OEM"
Allocating 15 zero-copy buffers
```

And in that case, the add-on remains active.
However, when a message is received on the SDR, the update doesn't reach the Mosquitto broker.

Since the path given is not correct, I also tried with rtl_433/rtl_433.conf.template as the rtl433_conf_file in the configuration but that works as badly as without any filename.

The conf.template has:
- device 0
- output mqtt://${host}:${port},user=${username},pass=${password},retain=${retain}
- report_meta time:iso:usec:tz
and a list of suppressed protocols (mostly TPMS).

When the .conf.template is removed, the add-on logs:
cat: can't open '/config/rtl_433/*.conf.template': No such file or directory

### What addon are you reporting the bug for?

rtl_443

### What is the addon version?

0.5.1

### What type of MQTT Broker are you using?

Home Assistant Mosquitto MQTT Broker

### Addon log messages

```txt
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
[12:40:11] INFO: The mqtt addon is not available.
[12:40:11] INFO: This is not a problem if you are using an external MQTT broker.
[12:40:11] INFO: If you are using the Home Assistant Mosquitto Broker addon, try restarting it, and then restart the rtl_433 addon.
[12:40:11] INFO: For an external broker, manually update the output line in the configuration file with mqtt connection settings, and restart the addon.
/tmp/rtl_433_heredoc: line 1: host: unbound variable
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
```

deviantintegral, 2024-04-19:
Can you post the `rtl_433.conf.template` file?

osmeest, 2024-04-19:
Sure! Here it is (gzipped to please the attach file function):
[rtl_433.conf.template.gz](https://github.com/pbkhrv/rtl_433-hass-addons/files/15045728/rtl_433.conf.template.gz)

deviantintegral, 2024-04-20:
It looks good. And, I realized that since your error is about mqtt, that check happens before even trying to start rtl_433.

Do you have https://github.com/hassio-addons/addon-ssh installed? If not, can you add it, connect to the terminal, and run this, which will create and run `test.sh`:

```console
cat > test.sh <<EOD
#!/usr/bin/with-contenv bashio

set -x
bashio::services.available "mqtt" && echo "MQTT is available"
bashio::services "mqtt" "host"
EOD

chmod +x test.sh
./test.sh
```

You should get a bunch of output. If you don't see "MQTT is available" and all of the connection info, then something is wrong with your MQTT addon not showing up in other addons.

If you do see info, please share it (but also redact any passwords!).

osmeest, 2024-04-20:
Thanks for the test script. 

In the output, I found an error instead of usable information:
`++++ curl --silent --show-error --write-out '\n%{http_code}' --request GET -H 'Authorization: Bearer <some_hex_hash>' -H 'Content-Type: application/json' -d '{}' http://supervisor/services/mqtt
+++ response='{"result":"error","message":"Service not enabled"}
400'`

Googling this error, I saw some messages that seems to imply that there could a problem when mosquito is started on boot. I restarted Mosquitto add-on and then the script reported "MQTT available".

I did a full system reboot and after that, the script still says "MQTT is available". I guess something must have gone wild during one of the component upgrades. Anyway, it is all working fine now. Thanks for the support !

closed 2024-04-20 by osmeest