pbkhrv/rtl_433-hass-addons #190: No MQTT topics created - No error in logs

maxperron, 2024-05-04:
### The problem

Hello,

I'm currently using rtlamr2mqtt (not maintained anymore) addon and I want to replace it with the rtl_433 addon.

I have installed the rtl_433 Home Assistant Add-on and using the default config file.

Here's the log once started:

> s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
Starting rtl_433 with rtl_433.conf...
[rtl_433] rtl_433 version 23.11 branch  at 202311281352 inputs file rtl_tcp RTL-SDR
[rtl_433] MQTT: Publishing MQTT data to core-mosquitto port 1883
[rtl_433] MQTT: Publishing device info to MQTT topic "rtl_433/9b13b3f4-rtl433/devices[/type][/model][/subtype][/channel][/id]".
[rtl_433] MQTT: Publishing events info to MQTT topic "rtl_433/9b13b3f4-rtl433/events".
[rtl_433] MQTT: Publishing states info to MQTT topic "rtl_433/9b13b3f4-rtl433/states".
[rtl_433] Use "-F log" if you want any messages, warnings, and errors in the console.
[rtl_433] Found Rafael Micro R828D tuner
[rtl_433] Exact sample rate is: 250000.000414 Hz
[rtl_433] Allocating 15 zero-copy buffers

Everything looks fine in the log, but when I look with MQTT explorer I see no topic:
[image]

I have seen this issue #185 and reinstalled the addon, but still no topic. I tried to reboot a couple of time, either, Home Assistant, Mosquitto and rtl_433 but still no topic.

Any idea what could cause that?

### What addon are you reporting the bug for?

rtl_443

### What is the addon version?

0.5.1

### What type of MQTT Broker are you using?

Home Assistant Mosquitto MQTT Broker

### Addon log messages

```txt
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
Starting rtl_433 with rtl_433.conf...
[rtl_433] rtl_433 version 23.11 branch  at 202311281352 inputs file rtl_tcp RTL-SDR
[rtl_433] MQTT: Publishing MQTT data to core-mosquitto port 1883
[rtl_433] MQTT: Publishing device info to MQTT topic "rtl_433/9b13b3f4-rtl433/devices[/type][/model][/subtype][/channel][/id]".
[rtl_433] MQTT: Publishing events info to MQTT topic "rtl_433/9b13b3f4-rtl433/events".
[rtl_433] MQTT: Publishing states info to MQTT topic "rtl_433/9b13b3f4-rtl433/states".
[rtl_433] Use "-F log" if you want any messages, warnings, and errors in the console.
[rtl_433] Found Rafael Micro R828D tuner
[rtl_433] Exact sample rate is: 250000.000414 Hz
[rtl_433] Allocating 15 zero-copy buffers
```

maxperron, 2024-05-24:
It finally works, I don't know what changed.

closed 2024-05-24 by maxperron

deviantintegral, 2024-05-26:
I wonder if you ran into https://github.com/home-assistant/core/pull/117267 which was recently fixed in Home Assistant itself.