pbkhrv/rtl_433-hass-addons #192: No data from Bresser-6in1 weather station

ATS65, 2024-05-14:
### The problem

I used Johan’s excellent arcticle [Adding a Bresser 5 in 1 sensor to Home Assistant 3](https://www.vromans.org/johan/articles/hass_bresser51/index.html) to integrate a Bresser 5in1 weather station.
Acquired supported DVB-stick, installed rtl_433 and configured to listen to 868.3MHz and send data to local mosquitto-broker.
According logs everything works fine until listening begins. Unfortunately that is also where logs terminate, nothing gets received at all.

### What addon are you reporting the bug for?

rtl_443

### What is the addon version?

23.11

### What type of MQTT Broker are you using?

Home Assistant Mosquitto MQTT Broker

### Addon log messages

```txt
Log from rtl_433 on HASS running Generic x86-64 installation on HP thin client:

Starting rtl_433 -c /config/rtl_433.ats.conf
[10:55:18] WARNING: rtl_433 now supports automatic configuration and multiple radios. The rtl_433_conf_file option is deprecated. See the documentation for migration instructions.
rtl_433 version 23.11 branch at 202311281352 inputs file rtl_tcp RTL-SDR
New defaults active, use “-Y classic -s 250k” if you need the old defaults
Registering protocol [172] “Bresser Weather Center 6-in-1, 7-in-1 indoor, soil, new 5-in-1, 3-in-1 wind gauge, Froggit WH6000, Ventus C8488A”
MQTT: Publishing MQTT data to 192.168.69.69 port 1883
MQTT: Publishing device info to MQTT topic “rtl_433/bresser51”.
[Protocols] Registered 1 out of 250 device decoding protocols
[Input] The internals of input handling changed, read about and report problems on PR #1978
[SDR] Found 1 device(s)
[SDR] trying device 0: Realtek, RTL2838UHIDIR, SN: 00000001
Found Fitipower FC0012 tuner
[SDR] Using device 0: Realtek, RTL2838UHIDIR, SN: 00000001, “Generic RTL2832U OEM”
[SDR] Sample rate set to 1024000 S/s.
[Input] Bit detection level set to 0.0 (Auto).
[SDR] Tuner gain set to Auto.
[Input] Reading samples in async mode…
[SDR] rtlsdr_set_center_freq 868300000 = 0
[SDR] Tuned to 868.300MHz.
Allocating 15 zero-copy buffers
[MQTT] MQTT Connected…
[acquire_thread] acquire_thread enter…
[MQTT] MQTT Connection established.
Failed to allocate zero-copy buffer for transfer 10
Falling back to buffers in userspace
Allocating 15 (non-zero-copy) user-space buffers
[Baseband] low pass filter for 1024000 Hz at cutoff 204800 Hz, 4.9 us
***END OF LOG***

Log from rtl_433 on Windows10 using same stick:

rtl_433 version 21.12 branch at 202112141644 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at “C:\Users.…\Desktop\HASS\rtl433\rtl_433.conf”…
Reading conf from “C:\Users.…\Desktop\HASS\rtl433\rtl_433.conf”.
New defaults active, use “-Y classic -s 250k” for the old defaults!
Registering protocol [172] “Bresser Weather Center 6-in-1, 7-in-1 indoor, new 5-in-1, 3-in-1 wind gauge, Froggit WH6000, Ventus C8488A”
Publishing MQTT data to 192.168.69.69 port 1883
Publishing device info to MQTT topic “rtl_433/bresser51”.
Registered 1 out of 207 device decoding protocols
Found 1 device(s)
trying device 0: Realtek, RTL2838UHIDIR, SN: 00000001
Found Fitipower FC0012 tuner
Using device 0: Generic RTL2832U OEM
Sample rate set to 1024000 S/s.
Bit detection level set to 0.0 (Auto).
Tuner gain set to Auto.
Reading samples in async mode…
Tuned to 868.300MHz.
MQTT Connected…
baseband_demod_FM: low pass filter for 1024000 Hz at cutoff 204800 Hz, 4.9 us
MQTT Connection established.
Exact bit width (in us) is 121.74 vs 123.05, 38 bit preamble
bresser_6in1_decode: {144} 90 9b 14 30 4c a6 18 ff ff ff 11 28 24 62 51 ff f0 b5
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time : 2024-05-03 12:36:30
model : Bresser-6in1 id : 14304ca6
channel : 0 Battery : 1 Temperature: 24.6 C Humidity : 51 Sensor type: 1
Wind Gust : 0.0 m/s Wind Speed: 0.0 m/s Direction : 112 Integrity : CRC
pulse_demod_pcm(): Bresser Weather Center 6-in-1, 7-in-1 indoor, new 5-in-1, 3-in-1 wind gauge, Froggit WH6000, Ventus C8488A
bitbuffer:: Number of rows: 1
[00] {275} 55 55 55 55 55 16 ea 48 4d 8a 18 26 53 0c 7f ff ff 88 94 12 31 28 ff f8 5a 80 00 00 00 00 00 00 00 00 00
Pulse data: 154 pulses
[ 0] Pulse: 0, Gap: 4, Period: 4
…
```

### Additional information

Using same DVB-stick on Windows10 I have proven it works and is supported as well as weather station is sending data.

markusm66, 2024-05-14:
Hi ATS65,

I have the same SDR Hardware Stick and the same result as you. 

My HA Hardware is a Raspberry Pi 5:

```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
[21:30:08] WARNING: rtl_433 now supports automatic configuration and multiple radios. The rtl_433_conf_file option is deprecated. See the documentation for migration instructions.
Starting rtl_433 -c /config/rtl_433/bresser.conf
rtl_433 version 23.11-117-gc4781e42 branch master at 202405090020 inputs file rtl_tcp RTL-SDR

New defaults active, use "-Y classic -s 250k" if you need the old defaults

MQTT: Publishing MQTT data to 192.168.178.135 port 1883
MQTT: Publishing device info to MQTT topic "rtl_433/bresser61".
MQTT: Publishing events info to MQTT topic "rtl_433/events".
MQTT: Publishing states info to MQTT topic "rtl_433/states".
[Protocols] Registered 1 out of 257 device decoding protocols
Registering protocol [172] "Bresser Weather Center 6-in-1, 7-in-1 indoor, soil, new 5-in-1, 3-in-1 wind gauge, Froggit WH6000, Ventus C8488A"
[Input] The internals of input handling changed, read about and report problems on PR #1978
[SDR] Found 1 device(s)
[SDR] trying device 0: Realtek, RTL2838UHIDIR, SN: 00000001
Detached kernel driver
Found Fitipower FC0012 tuner
[SDR] Using device 0: Realtek, RTL2838UHIDIR, SN: 00000001, "Generic RTL2832U OEM"
Exact sample rate is: 1000000.026491 Hz
[SDR] Sample rate set to 1000000 S/s.
[Input] Bit detection level set to 0.0 (Auto).
[SDR] Tuner gain set to Auto.
[Input] Reading samples in async mode...
[SDR] rtlsdr_set_center_freq 868300000 = 0
[SDR] Tuned to 868.300MHz.
[MQTT] MQTT Connected...
[acquire_thread] acquire_thread enter...
[MQTT] MQTT Connection established.
[Auto Level] Estimated noise level is -18.5 dB, adjusting minimum detection level to -15.5 dB
[Baseband] low pass filter for 1000000 Hz at cutoff 200000 Hz, 5.0 us
[Auto Level] Estimated noise level is -21.5 dB, adjusting minimum detection level to -18.5 dB
[Auto Level] Estimated noise level is -24.1 dB, adjusting minimum detection level to -21.1 dB
[Auto Level] Estimated noise level is -26.3 dB, adjusting minimum detection level to -23.3 dB
[Auto Level] Estimated noise level is -28.3 dB, adjusting minimum detection level to -25.3 dB
[Auto Level] Estimated noise level is -30.0 dB, adjusting minimum detection level to -27.0 dB
[Auto Level] Estimated noise level is -31.5 dB, adjusting minimum detection level to -28.5 dB
[Auto Level] Estimated noise level is -32.9 dB, adjusting minimum detection level to -29.9 dB
[Auto Level] Estimated noise level is -34.0 dB, adjusting minimum detection level to -31.0 dB
[Auto Level] Estimated noise level is -35.0 dB, adjusting minimum detection level to -32.0 dB
[Auto Level] Estimated noise level is -36.7 dB, adjusting minimum detection level to -33.7 dB
[Auto Level] Estimated noise level is -38.0 dB, adjusting minimum detection level to -35.0 dB
[Auto Level] Estimated noise level is -39.4 dB, adjusting minimum detection level to -36.4 dB
[Auto Level] Estimated noise level is -40.5 dB, adjusting minimum detection level to -37.5 dB
[Auto Level] Estimated noise level is -41.6 dB, adjusting minimum detection level to -38.6 dB
```

Same config on my MacMini works stable for weeks.

markusm66, 2024-05-25:
Since I am using a usb extension cable the issue is gone. I have no Idee why, but now it works for me.

ATS65, 2024-05-26:
Tried the same and used USB extension cable.
Unfortunately behavour keeps 100% same as it was :( No matter how far stick is positioned from ha-host.
Tested with both add-ons: rtl_433 and rtl_433 (next)

deviantintegral, 2024-05-26:
@ATS65 since you’re running on a thin client, I’m guessing you’re comfortable and familiar with using Docker directly? If so, can you try logging in with the terminal and running rtl_433 from https://github.com/hertzg/rtl_433_docker and see if it works? If it doesn’t, then we can completely rule out the addon itself.

ATS65, 2024-05-27:
Thx for the response! I appreciate the proposal to eliminate potential addon-problem.
Unfortunately I do not really have experience in handling hass docker containers.
I am using HA Core 2023.11.1, OS 12.1, Supervisor 2024.05.1
Have you probably got a link for dummies on how to continue?

deviantintegral, 2024-05-28:
The above is perfect - it shows something hanging. Can you remove all flags except `-Fkv` and `-v`? And then try `-f`…? You’ll want to let each run for an hour or so, or at least until you get a sense as to what data (if any) they pick up.

ATS65, 2024-05-29:
Was unfortunately not able to follow your above recommendation (did not find valid combination of params to start).
What I actually tried instead was creating the following stack:
```
version: "3"

services:
  rtl433:
    image: hertzg/rtl_433:latest
    devices:
      - "/dev/bus/usb/002/008"
    command:
      - "-c/rtl_433/rtl_433.conf"
    volumes:
      - /mnt/data/supervisor/homeassistant/rtl_433:/rtl_433
```
defining config/rtl_433/rtl_433.conf with 
```
verbose       8
gain          0
frequency     868.30M
sample_rate   1024k
ppm_error     0
report_meta   time:tz
protocol      172 # Bresser Weather Center 6-in-1 (newer 5-in-1)
convert       si
output        mqtt://192.168.69.69:1883,retain=1,devices=rtl_433/bresser51
output        kv
```
Result of this run is (no more change no matter how long I let it run)
```
rtl_433 version 23.11 (2023-11-28) inputs file rtl_tcp RTL-SDR with TLS
New defaults active, use "-Y classic -s 250k" if you need the old defaults
Registering protocol [172] "Bresser Weather Center 6-in-1, 7-in-1 indoor, soil, new 5-in-1, 3-in-1 wind gauge, Froggit WH6000, Ventus C8488A"
MQTT: Publishing MQTT data to 192.168.69.69 port 1883
MQTT: Publishing device info to MQTT topic "rtl_433/bresser51".
[Protocols] Registered 1 out of 250 device decoding protocols
[Input] The internals of input handling changed, read about and report problems on PR #1978
[SDR] Found 1 device(s)
[SDR] trying device 0: Realtek, RTL2838UHIDIR, SN: 00000001
Found Fitipower FC0012 tuner
[SDR] Using device 0: Realtek, RTL2838UHIDIR, SN: 00000001, "Generic RTL2832U OEM"
[SDR] Sample rate set to 1024000 S/s.
[Input] Bit detection level set to 0.0 (Auto).
[SDR] Tuner gain set to Auto.
[Input] Reading samples in async mode...
[SDR] rtlsdr_set_center_freq 868300000 = 0
[SDR] Tuned to 868.300MHz.
[acquire_thread] acquire_thread enter...
[MQTT] MQTT Connected...
[Baseband] low pass filter for 1024000 Hz at cutoff 204800 Hz, 4.9 us
[MQTT] MQTT Connection established.
```

-> seems we have proven that behaviour is identical to addon

ATS65, 2024-05-31:
good news: have replaced USB-extension-cable.
Since that time rtl_433-container peridically received data.
Switched back to add-on and it is also working properly now :)
Many thx to all of you!

closed 2024-05-31 by ATS65