pbkhrv/rtl_433-hass-addons #193: After Restart ned to disconnect the USB transmitter for it to be detected by the addon

9812ga98sd, 2024-05-17:
### The problem

After I restart my PC I have to physically disconnect and reconnect the tuner for it to be detected. This issue only arose after the latest upgrade.

The add on  normally shuts itself down as it doesnt find the tuner.

Once physically reconnected it finds the tuner:

Found Rafael Micro R820T/2 tuner
Exact sample rate is: 250000.000414 Hz
Allocating 15 zero-copy buffers

### What addon are you reporting the bug for?

rtl_443

### What is the addon version?

0.5.1

### What type of MQTT Broker are you using?

Home Assistant Mosquitto MQTT Broker

### Addon log messages

```txt
[13:59:51] WARNING: rtl_433 now supports automatic configuration and multiple radios. The rtl_433_conf_file option is deprecated. See the documentation for migration instructions.
Starting rtl_433 -c /config/rtl_433.conf
rtl_433 version 23.11 branch  at 202311281352 inputs file rtl_tcp RTL-SDR
MQTT: Publishing MQTT data to 192.168.1.216 port 1883
MQTT: Publishing device info to MQTT topic "rtl_433[/id]".
Use "-F log" if you want any messages, warnings, and errors in the console.
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
[14:15:29] WARNING: rtl_433 now supports automatic configuration and multiple radios. The rtl_433_conf_file option is deprecated. See the documentation for migration instructions.
Starting rtl_433 -c /config/rtl_433.conf
rtl_433 version 23.11 branch  at 202311281352 inputs file rtl_tcp RTL-SDR
MQTT: Publishing MQTT data to 192.168.1.216 port 1883
MQTT: Publishing device info to MQTT topic "rtl_433[/id]".
Use "-F log" if you want any messages, warnings, and errors in the console.
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
Starting rtl_433 -c /config/rtl_433.conf
[15:31:32] WARNING: rtl_433 now supports automatic configuration and multiple radios. The rtl_433_conf_file option is deprecated. See the documentation for migration instructions.
rtl_433 version 23.11 branch  at 202311281352 inputs file rtl_tcp RTL-SDR
MQTT: Publishing MQTT data to 192.168.1.216 port 1883
MQTT: Publishing device info to MQTT topic "rtl_433[/id]".
Use "-F log" if you want any messages, warnings, and errors in the console.
Found Rafael Micro R820T/2 tuner
Exact sample rate is: 250000.000414 Hz
Allocating 15 zero-copy buffers
```

deviantintegral, 2024-05-26:
This feels like an issue with the motherboard or radio. You could check to see if the device shows up to the Linux kernel at all after a reboot. First restart, then log in to the HASS console and run `dmesg  | grep -i realtek` (assuming that’s your tuner). You can also run `dmesg` directly and look for what the normal message are to search for.

dscowan, 2024-10-11:
I'm having the same issue.  If I reboot the hardware "Home Assistant Blue / Hardkernel ODROID-N2/N2+" the add on starts and then dies with no device found.  If I unplug and re-plug the device and start the addon it will find it.  I've had this hardware setup for over 2 years and just started having this issue with the latest upgrade.

dscowan, 2024-10-14:
@deviantintegral I do not remember upgrading RTL_433 addon but that doesn't mean I didn't.  I believe this started with upgrade to home assistant supervisor 2024.10 but not exactly sure.

deviantintegral, 2024-10-14:
Ok, thanks for checking. The Supervisor upgrade shouldn't have changed anything with this addon, but I suppose it's possible. I just upgraded supervisor myself so if that's related hopefully it shows up soon. I've also held back my Pi install on HAOS 12 because I've seen some USB issues reported with 13 / 13.1 that would be a pain for me to debug. But, this is all just guessing!

deviantintegral, 2025-03-22:
I'm pretty confident this has to be an issue outside of this addon, since it's the linux kernel and HAOS that's responsible for exposing the devices in the first place. I've not seen this myself on a fully updated Pi install and in a VM. Feel free to reopen if you have more information. Thanks!

closed 2025-03-22 by deviantintegral