pbkhrv/rtl_433-hass-addons #198: RTL_433 Stops as soon as it starts

maranhaoBruno, 2024-09-17:
### The problem

As soon as I start the add on it stops itself. I thought this was the result of not "finding" the device (I'm using a RSP1A bought directly from SDRPlay), so I created a configuration file where I include "device 3" since the device appears as "Bus 003 Device 003: ID 1df7:3000" when I type "lsusb" into the terminal. Do I have to install any SoapySDR drivers?

### What addon are you reporting the bug for?

rtl_443

### What is the addon version?

0.5.2

### What type of MQTT Broker are you using?

Home Assistant Mosquitto MQTT Broker

### Addon log messages

```txt
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
[12:09:35] WARNING: rtl_433 now supports automatic configuration and multiple radios. The rtl_433_conf_file option is deprecated. See the documentation for migration instructions.
Starting rtl_433 -c /config/rtl_433/rtl_433.conf
rtl_433 version 23.11 branch  at 202311281352 inputs file rtl_tcp RTL-SDR
MQTT: Publishing MQTT data to 10.0.0.78 port 1883
MQTT: Publishing device info to MQTT topic "rtl_433/9b13b3f4-rtl433/devices[/type][/model][/subtype][/channel][/id]".
MQTT: Publishing events info to MQTT topic "rtl_433/9b13b3f4-rtl433/events".
MQTT: Publishing states info to MQTT topic "rtl_433/9b13b3f4-rtl433/states".
Use "-F log" if you want any messages, warnings, and errors in the console.
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
```

maranhaoBruno, 2024-09-20:
@deviantintegral Thanks for the response and the info on the indexing.

Thanks again for helping me troubleshoot this; would you mind telling me how to turn on verbose logging? Is this a system's level setting in my config file, because the system logger (https://www.home-assistant.io/integrations/logger/) doesn't appear to have a "verbose" setting.

I only have one radio, and I know it is not an issue with the hardware itself because I was able to listen to local FM stations when attached to my laptop.

deviantintegral, 2024-09-20:
Ah, I meant in rtl_433 itself: https://github.com/merbanan/rtl_433/blob/master/conf/rtl_433.example.conf#L21-L25

Also, to answer on SoapySDR, it should just work out of the box. Something else to double check is to run rtl_433 directly on your laptop and see if it detects the radio.

maranhaoBruno, 2024-09-20:
Your suspicion was correct, I'm getting "[SDR] No supported devices found." in the log. I wonder why that is. When I connected the RSP1A to my laptop I utilized it through the SDRconnect application, but I also believe I had to install some driver on my MacBook. I'll try to install the driver on my HASS RPi5.

maranhaoBruno, 2024-09-20:
So I played around with the configuration file because an example one said that the default was to utilize RTL-SDR, but the Github page for this add-on states that the RSP1A is supported through SoapySDR. Adding `device ""` to the configuration file results in "[sdr_open] No input driver for SoapySDR compiled in." appearing in the log. I think the issue is a lack of SoapySDR on the platform.

maranhaoBruno, 2024-09-20:
@deviantintegral this is a little bit over my head, but what I'm understanding is that the rtl_433-hass-addons needs to be updated to support SoapySDR and the devices that rtl_433 supports through that driver, so I should sit tight until the next release, is that correct?

deviantintegral, 2024-09-21:
@maranhaoBruno I've added support for this in https://github.com/pbkhrv/rtl_433-hass-addons/pull/199. I've left testing steps, but if it's beyond your comfort that's OK.

Actually... I have a HackRF! I don't use it for rtl_433, but apparently it also requires soapysdr. So, I can test too.

Joseangel5651, 2024-10-13:
I had it configured and working correctly in HA, for other reasons I removed the USB device. 
After months when I connect it to HA I also get the same error.

Starting rtl_433 -c /config/rtl_433/rtl_433.conf
[16:08:58] WARNING: rtl_433 now supports automatic configuration and multiple radios. The rtl_433_conf_file option is deprecated. See the documentation for migration instructions.
rtl_433 version 23.11 branch  at 202311281352 inputs file rtl_tcp RTL-SDR
MQTT: Publishing MQTT data to 192.168.31.139 port 1883
MQTT: Publishing device info to MQTT topic "rtl_433/9b13b3f4-rtl433/devices[/type][/model][/subtype][/channel][/id]".
MQTT: Publishing events info to MQTT topic "rtl_433/9b13b3f4-rtl433/events".
MQTT: Publishing states info to MQTT topic "rtl_433/9b13b3f4-rtl433/states".
Use "-F log" if you want any messages, warnings, and errors in the console.
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

deviantintegral, 2024-10-20:
I've merged https://github.com/pbkhrv/rtl_433-hass-addons/pull/199 into the next branch.

closed 2024-10-20 by deviantintegral