pbkhrv/rtl_433-hass-addons #200: rt_433 (next) - is there an additional step to grab the latest master besides uninstall and install?

brycedjohnson, 2024-10-01:
### The problem

Hi,
I recently got a change merged in rtl_433 that I wanted to check with HA.

I saw I should be able to uninstall and reinstall next to get the latest?  But it doesn't appear to grab my commit (from 8 hours ago). (43758a383672409c0e2a620f498f6abddd90a7f3)

Do I need to wait a bit for that to get pulled in?  Or does it automaticly build it when you do the uninstall/install?

Thanks!
Bryce

### What addon are you reporting the bug for?

rtl_443

### What is the addon version?

rtl_433 (next)

### What type of MQTT Broker are you using?

Home Assistant Mosquitto MQTT Broker

### Addon log messages

```txt
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
Starting rtl_433 with rtl_433.conf...
[rtl_433] rtl_433 version 23.11-163-ge25cfc0f branch master at 202409200913 inputs file rtl_tcp RTL-SDR
[rtl_433] MQTT: Publishing MQTT data to core-mosquitto port 1883
[rtl_433] MQTT: Publishing device info to MQTT topic "rtl_433/9b13b3f4-rtl433-next/devices[/type][/model][/subtype][/channel][/id]".
[rtl_433] MQTT: Publishing events info to MQTT topic "rtl_433/9b13b3f4-rtl433-next/events".
[rtl_433] MQTT: Publishing states info to MQTT topic "rtl_433/9b13b3f4-rtl433-next/states".
[rtl_433] 
[rtl_433] New defaults active, use "-Y classic -s 250k" if you need the old defaults
[rtl_433] 
[rtl_433] Found Rafael Micro R828D tuner
[rtl_433] RTL-SDR Blog V4 Detected
[rtl_433] Exact sample rate is: 1000000.026491 Hz
```

brycedjohnson, 2024-10-02:
I guess I just had to be patient!  It uninstalled and reinstalled this morning again and it worked!  This for the great work on this!

closed 2024-10-02 by brycedjohnson

deviantintegral, 2024-10-02:
Yep, it runs daily so it can be up to 24 hours of lag. Glad it worked for you!