pbkhrv/rtl_433-hass-addons #202: Hanging on [rtl_433] Allocating 15 zero-copy buffers

LavaTiger99, 2024-10-10:
### The problem

Setup:
HomeAssistant Green device
Mosquitto broker installed & working
rtl_433_next never worked (didn't auto-connect to mosquitto)
Successfully installed rtl_433, and device pings started showing up in the logs, and it was automatically pushing to MQTT (seen in mqtt explorer)
After a couple hours, it stopped publishing/working. After trying to restart it multiple times, the startup of rtl_433 keeps hanging on [rtl_433] Allocating 15 zero-copy buffers
Strange, since it worked fine a few hours ago.
Tried uninstalling it, reinstalling it.
Tried updating librtlsdr:
```
[core-ssh ~]$ apk add librtlsdr
(1/2) Installing libusb (1.0.26-r3)
(2/2) Installing librtlsdr (2.0.1-r0)
OK: 107 MiB in 103 packages
```

### What addon are you reporting the bug for?

rtl_443

### What is the addon version?

0.5.2

### What type of MQTT Broker are you using?

Home Assistant Mosquitto MQTT Broker

### Addon log messages

```txt
Starting rtl_433 with rtl_433.conf...
[rtl_433] rtl_433 version 23.11 branch  at 202311281352 inputs file rtl_tcp RTL-SDR
[rtl_433] Found Rafael Micro R820T/2 tuner
[rtl_433] Exact sample rate is: 250000.000414 Hz
[rtl_433] [SDR] Using device 0: Realtek, RTL2838UHIDIR, SN: 00000001, "Generic RTL2832U OEM"
[rtl_433] Allocating 15 zero-copy buffers
```

deviantintegral, 2024-10-11:
Installing librtlsdr won’t do anything there because that’s not inside the addon container. But, we already include otherwise your radio wouldn’t show up at all.

I would start by enabling verbose mode, enabling key / value output, and disabling any protocol filters. See if you’re getting any data at all, even if it’s not from devices you care about.

I’ll also say some of the radios out there are not very robust. In one of my installs I had to both move the radio to a separate powered USB hub, and I had to disable frequency hopping and use two radios because they would lock up after a while.

uffevedenbrant, 2024-10-14:
My first SDR radio i bought last week Beginning oct 2024 worked without any issues at all. Took the complete setup to my summer house as its there i have the need for 433.
But it was a little fun to have it in town as well so i ordered a second one.
Failed completely. I could see rtl_433 (next) seeing 433 devices but it really did not look like the first one. 
rtl_433 MQTT Auto Discovery (next) actually never added the devices..

However as i had one working instance and one failing i did some comparision.
The one that worked had a file ../config/rtl_433/rtl_433.conf.template with a lot of config in it.
The one that failed hada empty file.

I copied the content from the working HA instans to the failing and then the failing started to wrok.
No clue if the installation in some way created the 0-byte file or something else happened.
However while testing i removed/installed the rtl- addons multiple times so i guess it should have been fixed with reinstallations.

Config in working HA instance. ( note that i never configured this file in the working HA instance.. )

# This is an empty template for configuring rtl_433. mqtt information will be
# automatically added. Create multiple files ending in '.conf.template' to
# manage multiple rtl_433 radios, being sure to set the 'device' setting. The
# device must be set before mqtt output lines.
# https://github.com/merbanan/rtl_433/blob/master/conf/rtl_433.example.conf

output mqtt://${host}:${port},user=${username},pass=${password},retain=${retain}
report_meta time:iso:usec:tz

# To keep the same topics when switching between the normal and edge versions,
# use this output line instead.
# output mqtt://${host}:${port},user=${username},pass=${password},retain=${retain},devices=rtl_433/9b13b3f4-rtl433/devices[/type][/model][/subtype][/channel][/id],events=rtl_433/9b13b3f4-rtl433/events,states=rtl_433/9b13b3f4-rtl433/states

# Uncomment the following line to also enable the default "table" output to the
# addon logs.
# output kv

# Disable TPMS sensors by default. These can cause an overwhelming number of
# devices and entities to show up in Home Assistant.
# This list is generated by running:
# rtl_433 -R help 2>&1 | grep -i tpms | sd '.*\[(\d+)\].*' 'protocol -/run.sh'
#    [59]  Steelmate TPMS
#    [60]  Schrader TPMS
#    [82]  Citroen TPMS
#    [88]  Toyota TPMS
#    [89]  Ford TPMS
#    [90]  Renault TPMS
#    [95]  Schrader TPMS EG53MA4, PA66GF35
#    [110]  PMV-107J (Toyota) TPMS
#    [123]* Jansite TPMS Model TY02S
#    [140]  Elantra2012 TPMS
#    [156]  Abarth 124 Spider TPMS
#    [168]  Schrader TPMS SMD3MA4 (Subaru)
#    [180]  Jansite TPMS Model Solar
#    [186]  Hyundai TPMS (VDO)
#    [201]  Unbranded SolarTPMS for trucks
#    [203]  Porsche Boxster/Cayman TPMS
protocol -59
protocol -60
protocol -82
protocol -88
protocol -89
protocol -90
protocol -95
protocol -110
protocol -123
protocol -140
protocol -156
protocol -168
######################## END

uffevedenbrant, 2024-10-14:
`# This is an empty template for configuring rtl_433. mqtt information will be
# automatically added. Create multiple files ending in '.conf.template' to
# manage multiple rtl_433 radios, being sure to set the 'device' setting. The
# device must be set before mqtt output lines.
# https://github.com/merbanan/rtl_433/blob/master/conf/rtl_433.example.conf

output mqtt://${host}:${port},user=${username},pass=${password},retain=${retain}
report_meta time:iso:usec:tz

# To keep the same topics when switching between the normal and edge versions,
# use this output line instead.
# output mqtt://${host}:${port},user=${username},pass=${password},retain=${retain},devices=rtl_433/9b13b3f4-rtl433/devices[/type][/model][/subtype][/channel][/id],events=rtl_433/9b13b3f4-rtl433/events,states=rtl_433/9b13b3f4-rtl433/states

# Uncomment the following line to also enable the default "table" output to the
# addon logs.
# output kv

# Disable TPMS sensors by default. These can cause an overwhelming number of
# devices and entities to show up in Home Assistant.
# This list is generated by running:
# rtl_433 -R help 2>&1 | grep -i tpms | sd '.*\[(\d+)\].*' 'protocol -/run.sh'
#    [59]  Steelmate TPMS
#    [60]  Schrader TPMS
#    [82]  Citroen TPMS
#    [88]  Toyota TPMS
#    [89]  Ford TPMS
#    [90]  Renault TPMS
#    [95]  Schrader TPMS EG53MA4, PA66GF35
#    [110]  PMV-107J (Toyota) TPMS
#    [123]* Jansite TPMS Model TY02S
#    [140]  Elantra2012 TPMS
#    [156]  Abarth 124 Spider TPMS
#    [168]  Schrader TPMS SMD3MA4 (Subaru)
#    [180]  Jansite TPMS Model Solar
#    [186]  Hyundai TPMS (VDO)
#    [201]  Unbranded SolarTPMS for trucks
#    [203]  Porsche Boxster/Cayman TPMS
protocol -59
protocol -60
protocol -82
protocol -88
protocol -89
protocol -90
protocol -95
protocol -110
protocol -123
protocol -140
protocol -156
protocol -168`

uffevedenbrant, 2024-10-14:
Only diff that i now can think of is that on the first install i went for the standard rtl_433 and then later decided to install the "next" version.

On the second i installed the "next" version immediately.
If there is a diff in the installation and the file is zerobyte in "next" and has some reasonable content in the standard version..
But if the file exists.. It will not be touched/replaced..

deviantintegral, 2024-10-20:
> I removed the config/rtl_433 folder after uninstalling, but is there anything else I need to remove to completely fresh install everything?

That should be it. It should create the rtl_433 folder and default configuration file on first start. Does your addon config have the legacy conf_file directive set? If so, remove that:

https://github.com/pbkhrv/rtl_433-hass-addons/blob/main/rtl_433/run.sh#L27

Robbi-Wan-Kenobi, 2024-11-03:
Where you able to solve this issue? I'm having the same Allocating 15 zero-copy buffers problem. 
Is there a way to debug this issue, or get more information on ways to solve it?

MichalDuban, 2024-12-19:
Hi, I have very similar issue. I installed the addon, but there is _Allocating 15 zero-copy buffers_ problem. I was able to get some detailed log (please check below). I think this coule be problematic line? **[sdr_set_center_freq] Failed to set center freq.**

```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
[16:37:22] WARNING: rtl_433 now supports automatic configuration and multiple radios. The rtl_433_conf_file option is deprecated. See the documentation for migration instructions.
Starting rtl_433 -c /config/-F log
rtl_433 version 23.11 branch  at 202311281352 inputs file rtl_tcp RTL-SDR
Failed to stat "/config/-F log"
Found Rafael Micro R828D tuner
[SDR] Using device 0: RTLSDRBlog, Blog V4, SN: 00000001, "Generic RTL2832U OEM"
Exact sample rate is: 250000.000414 Hz
[sdr_set_center_freq] Failed to set center freq.
Allocating 15 zero-copy buffers
```

Hope this will help to solve the issue. Thanks!

MichalDuban, 2024-12-19:
Could it be cause by driver incopatibility? I found interesting post [here](https://www.rtl-sdr.com/v4/). They are using different drivers, which should be backward-compatible with older devices..

kostalmichal, 2025-02-12:
I have the same problem with RTL-SDR Blog v4. I tried it on windows, where I had similar problem, but updating drivers solved the problem. Should it be possible to update driver in this addon?

deviantintegral, 2025-03-22:
This should now be working in the latest addon release. Thanks!

closed 2025-03-22 by deviantintegral