pbkhrv/rtl_433-hass-addons #203: TypeError: argument of type 'int' is not iterable

KomelT, 2024-11-17:
### The problem

I am running the plugin in HA and then getting this error. Is this the plugin's fault, or can it be something else?

Kind regards

### What addon are you reporting the bug for?

rtl_433_mqtt_autodiscover

### What is the addon version?

0.8.2

### What type of MQTT Broker are you using?

Home Assistant Mosquitto MQTT Broker

### Addon log messages

```txt
This is the output

Starting rtl_433_mqtt_hass.py...
[2024-11-17T16:00:39+0100] INFO:root:Discovering all devices
[2024-11-17T16:00:39+0100] INFO:root:MQTT connected: Connection Accepted.
[2024-11-17T16:00:39+0100] INFO:root:Subscribing to: rtl_433/#
[2024-11-17T16:00:42+0100] INFO:root:Published Nexus-TH-3-208: time, channel, battery_ok, temperature_C, humidity
[2024-11-17T16:00:42+0100] ERROR:root:JSON decode error: 2024-11-17T16:00:41.540689+0100
Exception in thread Thread-1 (_thread_main):
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/threading.py", line 1073, in _bootstrap_inner
    self.run()
  File "/usr/local/lib/python3.12/threading.py", line 1010, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/local/lib/python3.12/site-packages/paho/mqtt/client.py", line 3591, in _thread_main
    self.loop_forever(retry_first_connection=True)
  File "/usr/local/lib/python3.12/site-packages/paho/mqtt/client.py", line 1756, in loop_forever
    rc = self._loop(timeout)
         ^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/paho/mqtt/client.py", line 1164, in _loop
    rc = self.loop_read()
         ^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/paho/mqtt/client.py", line 1556, in loop_read
    rc = self._packet_read()
         ^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/paho/mqtt/client.py", line 2439, in _packet_read
    rc = self._packet_handle()
         ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/paho/mqtt/client.py", line 3033, in _packet_handle
    return self._handle_publish()
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/paho/mqtt/client.py", line 3327, in _handle_publish
    self._handle_on_message(message)
  File "/usr/local/lib/python3.12/site-packages/paho/mqtt/client.py", line 3570, in _handle_on_message
    on_message(self, self._userdata, message)
  File "/rtl_433_mqtt_hass.py", line 752, in mqtt_message
    bridge_event_to_hass(client, topicprefix, data)
  File "/rtl_433_mqtt_hass.py", line 840, in bridge_event_to_hass
    if "model" not in data:
       ^^^^^^^^^^^^^^^^^^^
TypeError: argument of type 'int' is not iterable
```
```

deviantintegral, 2024-11-23:
It reads like somehow `data` (which is supposed to be a dictionary) is coming back as a single number. Can you try again but with the log level set to debug?

deviantintegral, 2025-03-22:
Feel free to reopen if you have more information. Thanks!

closed 2025-03-22 by deviantintegral