pbkhrv/rtl_433-hass-addons #206: Stops working, does not automatically restart.

AndyW999, 2024-12-30:
### The problem

The rtl433 next add on stops, it does not automatically restart.

MQTT Explorer is still seeing new data.

In logs - 
rtlsdr_demod_write_reg failed with -9
r82xx_write: i2c wr failed=-9 reg=06 len=1

Pressing Start and it all works again.

Using a cheap RTL820T2 dongle.

### What addon are you reporting the bug for?

rtl_443

### What is the addon version?

rtl_433 version 24.10-12-g568b3abd branch master at 202412030005

### What type of MQTT Broker are you using?

Home Assistant Mosquitto MQTT Broker

### Addon log messages

```txt
rtlsdr_demod_write_reg failed with -9
r82xx_write: i2c wr failed=-9 reg=06 len=1
```

deviantintegral, 2025-01-01:
That error means that the USB connection broke to the adapter. If you're using an SBC like a Raspberry Pi, try connecting the RTL dongle through a powered USB hub. Also, you can check `dmesg` to confirm that the USB device is disconnecting and reconnecting.

Do you have the watchdog option enabled? I would have thought that would catch this. Note it's got a backoff where it delays restarting for longer and longer periods until giving up. So if it takes a while for the USB connection to settle, HAOS may have timed out and stopped trying to restart the addon. You can see info about it in the supervisor logs.

AndyW999, 2025-01-02:
It is an RPi5 and the dongle is connected to an unpowered USB hub.

Watchdog is enabled but appears not to have managed to restart it.

dmesg - 

➜  ~ dmesg | grep rtl2832
[774903.885233] rtl2832 15-0010: Realtek RTL2832 successfully attached
[774903.895596] rtl2832_sdr rtl2832_sdr.3.auto: Registered as swradio0
[774903.895603] rtl2832_sdr rtl2832_sdr.3.auto: Realtek RTL2832 SDR attached
[774903.895605] rtl2832_sdr rtl2832_sdr.3.auto: SDR API is still slightly experimental and functionality changes may follow
[774912.968161] rtl2832 15-0010: Realtek RTL2832 successfully attached
[774912.977615] rtl2832_sdr rtl2832_sdr.3.auto: Registered as swradio0
[774912.977623] rtl2832_sdr rtl2832_sdr.3.auto: Realtek RTL2832 SDR attached
[774912.977625] rtl2832_sdr rtl2832_sdr.3.auto: SDR API is still slightly experimental and functionality changes may follow
[774921.622005] rtl2832 15-0010: Realtek RTL2832 successfully attached
[774921.632868] rtl2832_sdr rtl2832_sdr.3.auto: Registered as swradio0
[774921.632876] rtl2832_sdr rtl2832_sdr.3.auto: Realtek RTL2832 SDR attached
[774921.632879] rtl2832_sdr rtl2832_sdr.3.auto: SDR API is still slightly experimental and functionality changes may follow
[774930.605405] rtl2832 15-0010: Realtek RTL2832 successfully attached
[774930.616390] rtl2832_sdr rtl2832_sdr.3.auto: Registered as swradio0
[774930.616397] rtl2832_sdr rtl2832_sdr.3.auto: Realtek RTL2832 SDR attached
[774930.616399] rtl2832_sdr rtl2832_sdr.3.auto: SDR API is still slightly experimental and functionality changes may follow
[774939.342270] rtl2832 15-0010: Realtek RTL2832 successfully attached
[774939.354735] rtl2832_sdr rtl2832_sdr.3.auto: Registered as swradio0
[774939.354740] rtl2832_sdr rtl2832_sdr.3.auto: Realtek RTL2832 SDR attached
[774939.354743] rtl2832_sdr rtl2832_sdr.3.auto: SDR API is still slightly experimental and functionality changes may follow
[774948.013934] rtl2832 15-0010: Realtek RTL2832 successfully attached
[774948.024188] rtl2832_sdr rtl2832_sdr.3.auto: Registered as swradio0
[774948.024193] rtl2832_sdr rtl2832_sdr.3.auto: Realtek RTL2832 SDR attached
[774948.024196] rtl2832_sdr rtl2832_sdr.3.auto: SDR API is still slightly experimental and functionality changes may follow
[774957.320560] rtl2832 15-0010: Realtek RTL2832 successfully attached
[774957.331022] rtl2832_sdr rtl2832_sdr.3.auto: Registered as swradio0
[774957.331028] rtl2832_sdr rtl2832_sdr.3.auto: Realtek RTL2832 SDR attached
[774957.331031] rtl2832_sdr rtl2832_sdr.3.auto: SDR API is still slightly experimental and functionality changes may follow
[774966.343514] rtl2832 15-0010: Realtek RTL2832 successfully attached
[774966.353502] rtl2832_sdr rtl2832_sdr.3.auto: Registered as swradio0
[774966.353507] rtl2832_sdr rtl2832_sdr.3.auto: Realtek RTL2832 SDR attached
[774966.353510] rtl2832_sdr rtl2832_sdr.3.auto: SDR API is still slightly experimental and functionality changes may follow
[774975.596605] rtl2832 15-0010: Realtek RTL2832 successfully attached
[774975.606892] rtl2832_sdr rtl2832_sdr.3.auto: Registered as swradio0
[774975.606897] rtl2832_sdr rtl2832_sdr.3.auto: Realtek RTL2832 SDR attached
[774975.606900] rtl2832_sdr rtl2832_sdr.3.auto: SDR API is still slightly experimental and functionality changes may follow
[774984.335244] rtl2832 15-0010: Realtek RTL2832 successfully attached
[774984.345438] rtl2832_sdr rtl2832_sdr.3.auto: Registered as swradio0
[774984.345443] rtl2832_sdr rtl2832_sdr.3.auto: Realtek RTL2832 SDR attached
[774984.345446] rtl2832_sdr rtl2832_sdr.3.auto: SDR API is still slightly experimental and functionality changes may follow
[774992.954499] rtl2832 15-0010: Realtek RTL2832 successfully attached
[774992.964816] rtl2832_sdr rtl2832_sdr.3.auto: Registered as swradio0
[774992.964822] rtl2832_sdr rtl2832_sdr.3.auto: Realtek RTL2832 SDR attached
[774992.964824] rtl2832_sdr rtl2832_sdr.3.auto: SDR API is still slightly experimental and functionality changes may follow

deviantintegral, 2025-01-02:
Can you check for usb related messages with `grep -i usb`?

AndyW999, 2025-01-02:
This is all I can get - 

[774957.344214] rc rc2: Realtek RTL2832U reference design as /devices/platform/axi/1000120000.pcie/1f00200000.usb/xhci-hcd.0/usb1/1-1/1-1.2/rc/rc2
[774957.344278] rc rc2: lirc_dev: driver dvb_usb_rtl28xxu registered at minor = 0, raw IR receiver, no transmitter
[774957.344332] input: Realtek RTL2832U reference design as /devices/platform/axi/1000120000.pcie/1f00200000.usb/xhci-hcd.0/usb1/1-1/1-1.2/rc/rc2/input40
[774957.344451] usb 1-1.2: dvb_usb_v2: schedule remote query interval to 200 msecs
[774957.353760] usb 1-1.2: dvb_usb_v2: 'Realtek RTL2832U reference design' successfully initialized and connected
[774961.142590] dvb_usb_v2: 'Realtek RTL2832U reference design:1-1.2' successfully deinitialized and disconnected
[774966.280392] usb 1-1.2: dvb_usb_v2: found a 'Realtek RTL2832U reference design' in warm state
[774966.339989] usb 1-1.2: dvb_usb_v2: will pass the complete MPEG2 transport stream to the software demuxer
[774966.340010] usb 1-1.2: media controller created
[774966.343819] usb 1-1.2: DVB: registering adapter 0 frontend 0 (Realtek RTL2832 (DVB-T))...
[774966.361757] rc rc2: Realtek RTL2832U reference design as /devices/platform/axi/1000120000.pcie/1f00200000.usb/xhci-hcd.0/usb1/1-1/1-1.2/rc/rc2
[774966.361868] rc rc2: lirc_dev: driver dvb_usb_rtl28xxu registered at minor = 0, raw IR receiver, no transmitter
[774966.361928] input: Realtek RTL2832U reference design as /devices/platform/axi/1000120000.pcie/1f00200000.usb/xhci-hcd.0/usb1/1-1/1-1.2/rc/rc2/input41
[774966.362037] usb 1-1.2: dvb_usb_v2: schedule remote query interval to 200 msecs
[774966.375895] usb 1-1.2: dvb_usb_v2: 'Realtek RTL2832U reference design' successfully initialized and connected
[774970.168858] dvb_usb_v2: 'Realtek RTL2832U reference design:1-1.2' successfully deinitialized and disconnected
[774975.533546] usb 1-1.2: dvb_usb_v2: found a 'Realtek RTL2832U reference design' in warm state
[774975.593151] usb 1-1.2: dvb_usb_v2: will pass the complete MPEG2 transport stream to the software demuxer
[774975.593170] usb 1-1.2: media controller created
[774975.596656] usb 1-1.2: DVB: registering adapter 0 frontend 0 (Realtek RTL2832 (DVB-T))...
[774975.620085] rc rc2: Realtek RTL2832U reference design as /devices/platform/axi/1000120000.pcie/1f00200000.usb/xhci-hcd.0/usb1/1-1/1-1.2/rc/rc2
[774975.620170] rc rc2: lirc_dev: driver dvb_usb_rtl28xxu registered at minor = 0, raw IR receiver, no transmitter
[774975.620282] input: Realtek RTL2832U reference design as /devices/platform/axi/1000120000.pcie/1f00200000.usb/xhci-hcd.0/usb1/1-1/1-1.2/rc/rc2/input42
[774975.620437] usb 1-1.2: dvb_usb_v2: schedule remote query interval to 200 msecs
[774975.629676] usb 1-1.2: dvb_usb_v2: 'Realtek RTL2832U reference design' successfully initialized and connected
[774979.425042] dvb_usb_v2: 'Realtek RTL2832U reference design:1-1.2' successfully deinitialized and disconnected
[774984.272175] usb 1-1.2: dvb_usb_v2: found a 'Realtek RTL2832U reference design' in warm state
[774984.331765] usb 1-1.2: dvb_usb_v2: will pass the complete MPEG2 transport stream to the software demuxer
[774984.331784] usb 1-1.2: media controller created
[774984.335278] usb 1-1.2: DVB: registering adapter 0 frontend 0 (Realtek RTL2832 (DVB-T))...
[774984.353889] rc rc2: Realtek RTL2832U reference design as /devices/platform/axi/1000120000.pcie/1f00200000.usb/xhci-hcd.0/usb1/1-1/1-1.2/rc/rc2
[774984.353957] rc rc2: lirc_dev: driver dvb_usb_rtl28xxu registered at minor = 0, raw IR receiver, no transmitter
[774984.354008] input: Realtek RTL2832U reference design as /devices/platform/axi/1000120000.pcie/1f00200000.usb/xhci-hcd.0/usb1/1-1/1-1.2/rc/rc2/input43
[774984.354353] usb 1-1.2: dvb_usb_v2: schedule remote query interval to 200 msecs
[774984.367797] usb 1-1.2: dvb_usb_v2: 'Realtek RTL2832U reference design' successfully initialized and connected
[774988.137008] dvb_usb_v2: 'Realtek RTL2832U reference design:1-1.2' successfully deinitialized and disconnected
[774992.891410] usb 1-1.2: dvb_usb_v2: found a 'Realtek RTL2832U reference design' in warm state
[774992.951009] usb 1-1.2: dvb_usb_v2: will pass the complete MPEG2 transport stream to the software demuxer
[774992.951027] usb 1-1.2: media controller created
[774992.954544] usb 1-1.2: DVB: registering adapter 0 frontend 0 (Realtek RTL2832 (DVB-T))...
[774992.977749] rc rc2: Realtek RTL2832U reference design as /devices/platform/axi/1000120000.pcie/1f00200000.usb/xhci-hcd.0/usb1/1-1/1-1.2/rc/rc2
[774992.977927] rc rc2: lirc_dev: driver dvb_usb_rtl28xxu registered at minor = 0, raw IR receiver, no transmitter
[774992.977994] input: Realtek RTL2832U reference design as /devices/platform/axi/1000120000.pcie/1f00200000.usb/xhci-hcd.0/usb1/1-1/1-1.2/rc/rc2/input44
[774992.978199] usb 1-1.2: dvb_usb_v2: schedule remote query interval to 200 msecs
[774992.987418] usb 1-1.2: dvb_usb_v2: 'Realtek RTL2832U reference design' successfully initialized and connected
[777319.832750] dvb_usb_v2: 'Realtek RTL2832U reference design:1-1.2' successfully deinitialized and disconnected
[782539.944145] usb 1-1.4: USB disconnect, device number 31
[782539.944352] cp210x ttyUSB1: cp210x converter now disconnected from ttyUSB1
[783147.434480] usb 1-1.4: new full-speed USB device number 32 using xhci-hcd
[783147.568178] usb 1-1.4: New USB device found, idVendor=10c4, idProduct=ea60, bcdDevice= 1.00
[783147.568185] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[783147.568189] usb 1-1.4: Product: CP2102 USB to UART Bridge Controller
[783147.568192] usb 1-1.4: Manufacturer: Silicon Labs
[783147.568195] usb 1-1.4: SerialNumber: 0001
[783147.580716] usb 1-1.4: cp210x converter now attached to ttyUSB1

deviantintegral, 2025-01-02:
I was expecting to see the dongle resetting in those messages, but it's not there.

Since you're using an unpowered USB hub, along with other suggestions like https://github.com/bemasher/rtlamr/discussions/306 and https://github.com/merbanan/rtl_433/discussions/2968#discussioncomment-9780942, I would start with swapping that out for a powered one. I have a [SABRENT USB hub 3.2x1 with power adapter](https://amazon.ca/gp/product/B00TPMEOYM/) which has handy power buttons for each USB port.

AndyW999, 2025-01-03:
I agree, cheap nasty USB hub!

I will get a powered one...

Thanks

Andy.

closed 2025-01-03 by AndyW999