pbkhrv/rtl_433-hass-addons #207: AutoDiscovery paho exception

thtech182, 2025-01-02:
### The problem

When starting the Auto Discovery (next) add-on, it throws an exception that appears to be related to the paho MQTT client. The rtl_433 (next) add-on is running and shows the temperatures from my sensors. But they are not updating in HA.
My rtl_433.conf.template file is attached.
[rtl_433.txt](https://github.com/user-attachments/files/18294592/rtl_433.txt)

### What addon are you reporting the bug for?

rtl_433_mqtt_autodiscover

### What is the addon version?

next

### What type of MQTT Broker are you using?

Home Assistant Mosquitto MQTT Broker

### Addon log messages

```txt
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
mqtt found in this Home Assistance instance.
Starting rtl_433_mqtt_hass.py...
[2025-01-02T15:03:20-0600] INFO:root:Enabling debug logging
[2025-01-02T15:03:20-0600] INFO:root:Discovering all devices
[2025-01-02T15:03:20-0600] DEBUG:root:MQTT Client: Starting Loop
[2025-01-02T15:03:20-0600] DEBUG:paho.mqtt.client:Sending CONNECT (u1, p1, wr0, wq0, wf0, c1, k60) client_id=b''
[2025-01-02T15:03:20-0600] DEBUG:paho.mqtt.client:Received CONNACK (0, 0)
[2025-01-02T15:03:20-0600] INFO:root:MQTT connected: Connection Accepted.
[2025-01-02T15:03:20-0600] INFO:root:Subscribing to: rtl_433/+/events
[2025-01-02T15:03:20-0600] DEBUG:paho.mqtt.client:Sending SUBSCRIBE (d0, m1) [(b'rtl_433/+/events', 0)]
[2025-01-02T15:03:20-0600] DEBUG:paho.mqtt.client:Received SUBACK
[2025-01-02T15:03:20-0600] DEBUG:paho.mqtt.client:Received PUBLISH (d0, q0, r1, m0), 'rtl_433/9b13b3f4-rtl433/events', ...  (165 bytes)
[2025-01-02T15:03:20-0600] DEBUG:root:MQTT message: "{\"time\":\"2025-01-01T04:35:05.720379-0600\",\"model\":\"ERT-SCM\",\"id\":44746536,\"physical_tamper\":2,\"ert_type\":12,\"encoder_tamper\":0,\"consumption_data\":519046,\"mic\":\"CRC\"}"
[2025-01-02T15:03:20-0600] DEBUG:root:homeassistant/sensor/ERT-SCM-44746536/ERT-SCM-44746536-UTC/config:{"device_class": "timestamp", "name": "Timestamp", "entity_category": "diagnostic", "enabled_by_default": false, "icon": "mdi:clock-in", "state_topic": "rtl_433/9b13b3f4-rtl433/devices/ERT-SCM/44746536/time", "unique_id": "ERT-SCM-44746536-UTC", "device": {"identifiers": ["ERT-SCM-44746536"], "name": "ERT-SCM-44746536", "model": "ERT-SCM", "manufacturer": "rtl_433"}}
[2025-01-02T15:03:20-0600] DEBUG:paho.mqtt.client:Sending PUBLISH (d0, q0, r1, m2), 'b'homeassistant/sensor/ERT-SCM-44746536/ERT-SCM-44746536-UTC/config'', ... (369 bytes)
[2025-01-02T15:03:20-0600] DEBUG:root:homeassistant/sensor/ERT-SCM-44746536/ERT-SCM-44746536-consumption/config:{"name": "SCM Consumption Value", "value_template": "{{ value|int }}", "state_class": "total_increasing", "state_topic": "rtl_433/9b13b3f4-rtl433/devices/ERT-SCM/44746536/consumption_data", "unique_id": "ERT-SCM-44746536-consumption", "device": {"identifiers": ["ERT-SCM-44746536"], "name": "ERT-SCM-44746536", "model": "ERT-SCM", "manufacturer": "rtl_433"}}
[2025-01-02T15:03:20-0600] DEBUG:paho.mqtt.client:Sending PUBLISH (d0, q0, r1, m3), 'b'homeassistant/sensor/ERT-SCM-44746536/ERT-SCM-44746536-consumption/config'', ... (358 bytes)
[2025-01-02T15:03:20-0600] INFO:root:Published ERT-SCM-44746536: time, consumption_data
[2025-01-02T15:03:20-0600] INFO:root:Skipped ERT-SCM-44746536: physical_tamper, ert_type, encoder_tamper
[2025-01-02T15:03:20-0600] DEBUG:paho.mqtt.client:Received PUBLISH (d0, q0, r1, m0), 'rtl_433/9b13b3f4-rtl433-next/events', ...  (104 bytes)
[2025-01-02T15:03:20-0600] ERROR:paho.mqtt.client:Caught exception in on_message: 'utf-8' codec can't decode byte 0xe0 in position 87: invalid continuation byte
Exception in thread Thread-1 (_thread_main):
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/threading.py", line 1075, in _bootstrap_inner
    self.run()
  File "/usr/local/lib/python3.12/threading.py", line 1012, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/local/lib/python3.12/site-packages/paho/mqtt/client.py", line 3591, in _thread_main
    self.loop_forever(retry_first_connection=True)
  File "/usr/local/lib/python3.12/site-packages/paho/mqtt/client.py", line 1756, in loop_forever
    rc = self._loop(timeout)
         ^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/paho/mqtt/client.py", line 1164, in _loop
    rc = self.loop_read()
         ^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/paho/mqtt/client.py", line 1556, in loop_read
    rc = self._packet_read()
         ^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/paho/mqtt/client.py", line 2439, in _packet_read
    rc = self._packet_handle()
         ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/paho/mqtt/client.py", line 3033, in _packet_handle
    return self._handle_publish()
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/paho/mqtt/client.py", line 3327, in _handle_publish
    self._handle_on_message(message)
  File "/usr/local/lib/python3.12/site-packages/paho/mqtt/client.py", line 3570, in _handle_on_message
    on_message(self, self._userdata, message)
  File "/rtl_433_mqtt_hass.py", line 875, in mqtt_message
    logging.debug("MQTT message: " + json.dumps(msg.payload.decode()))
                                                ^^^^^^^^^^^^^^^^^^^^
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xe0 in position 87: invalid continuation byte
```

deviantintegral, 2025-01-02:
This sounds like rtl_433 is sending data that's not in the utf-8 character set. Can you use an mqtt client like MQTT Explorer (or the "Listen to a topic" in the MQTT integration configuration) to watch `rtl_433/9b13b3f4-rtl433-next/events` and share the message it's choking on? My guess is this needs to be reported as a bug upstream to the rtl_433 project.

thtech182, 2025-01-02:
This is the only message that shows under that topic right now, I will keep it listening and see if anything else pops up but doubtful.

Message 0 received on rtl_433/9b13b3f4-rtl433-next/events at 4:07 PM:
b'{"time":"2025-01-01T15:46:09.820231-0600","model":"AradMsMeter-Dialog3G","id":"00000000\xe0","volume_m3":0}'
QoS: 0 - Retain: true

deviantintegral, 2025-01-02:
> "id":"00000000\xe0"

That's the problem! `\xe0` escapes to a specific byte, in this case `e0` which isn't valid. It looks like there's an open issue to fully support this water meter over at https://github.com/merbanan/rtl_433/issues/1992. Given volume is 0, I'm assuming it's not valid data.

You should be able to disable the decoder to work around this.

thtech182, 2025-01-02:
Oh sweet! I'm still learning how to configure all of this.. I assume I'd disable one of the protocols? I found protocol 260 # Arad/Master Meter Dialog3G water utility meter and commented it out. But on restart it still throws the error. Is there something more I need to do?

Thanks for the quick help on this by the way!

deviantintegral, 2025-01-02:
You may need to disable it with a minus sign (protocol -260).

thtech182, 2025-01-02:
Up and running again. I had to publish an empty payload to that topic to clear out the retained bad message.

Thank you!

closed 2025-01-02 by deviantintegral