pbkhrv/rtl_433-hass-addons #210: Can't enable protocol 200 trough the config file

uvalkonovic, 2025-01-21:
### The problem

I created config file rtl_433.conf then I put it to te /homeassistant/ directory and added rtl_433.conf to the config line of rtl_433/
I enabled there protocol 200 by adding line to the config. When the plugin starts, it does not enable the protocol 200.
On the differnt machine, when I start rtl_433 with -R 200 option. it receives the data.

It seems, that adding the line protocol 200 to config, it does not enable by default disabled protocol 200.

### What addon are you reporting the bug for?

rtl_443

### What is the addon version?

0.5.2

### What type of MQTT Broker are you using?

Home Assistant Mosquitto MQTT Broker

### Addon log messages

```txt
Reattached kernel driver
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
[17:24:29] WARNING: rtl_433 now supports automatic configuration and multiple radios. The rtl_433_conf_file option is deprecated. See the documentation for migration instructions.
Starting rtl_433 -c /config/rtl_433.conf
rtl_433 version 23.11 branch  at 202311281352 inputs file rtl_tcp RTL-SDR
Detached kernel driver
Found Rafael Micro R820T/2 tuner
[SDR] Using device 0: Realtek, RTL2838UHIDIR, SN: 00000001, "Generic RTL2832U OEM"
Exact sample rate is: 250000.000414 Hz
Allocating 15 zero-copy buffers
```

### Additional information

I Googled a lot, but did not find the solution.
Thnks for any help.

tuttutZOOM, 2025-01-26:
I had the same issue for protocol 231. It was resolved in my case by switching to “rtl_433 (next)”. I assume that the version of rtl_433 in the current hass package is not current enough…

uvalkonovic, 2025-01-28:
Can you post here content of your config file?
Thanks.

tuttutZOOM, 2025-01-28:
Here is my current config file that's working for me in rtl_433 (next):
`device 1        # Second USB device
convert si
verbose 2

protocol 40     # [40]   Acurite 592TXR Temp/Humidity Sensor
protocol 55     # [55]   Acurite 606TX Temperature Sensor
protocol 231    # [231]  Govee Water Leak Detector H5054

output kv
output mqtt://_yourhostname_:1883,user=_yourusername_,pass=_yourpassword_,devices=rtl_433/9b13b3f4-rtl433/devices[/type][/model][/subtype][/channel][/id],events=rtl_433/9b13b3f4-rtl433/events,states=rtl_433/9b13b3f4-rtl433/states

# Disable TPMS sensors by default
protocol -59
protocol -60
protocol -82
protocol -88
protocol -89
protocol -90
protocol -95
protocol -110
protocol -123
protocol -140
protocol -156
protocol -168
protocol -180
protocol -186
protocol -201
protocol -203`

uvalkonovic, 2025-01-28:
Protocol 231 is not by default disabled, so you dont have same issue as me.
Thanks anyway.

deviantintegral, 2025-02-08:
Can you turn on verbose mode? It should show and confirm it activating protocol 200.

uvalkonovic, 2025-02-11:
When enabling protocols via config file, I put there

protocol 200
protocol 3
protocol 146

s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
[11:07:35] WARNING: rtl_433 now supports automatic configuration and multiple radios. The rtl_433_conf_file option is deprecated. See the documentation for migration instructions.
Starting rtl_433 -c /config/rtl_433.conf
rtl_433 version 23.11 branch  at 202311281352 inputs file rtl_tcp RTL-SDR
Registering protocol [200] "Auriol 4-LD5661/4-LD5972/4-LD6313 temperature/rain sensors"
Registering protocol [3] "Prologue, FreeTec NC-7104, NC-7159-675 temperature sensor"
Registering protocol [146] "Auriol AFW2A1 temperature/humidity sensor"
[Protocols] Registered 3 out of 250 device decoding protocols
[Input] The internals of input handling changed, read about and report problems on PR #1978
[SDR] Found 1 device(s)
[SDR] trying device 0: Realtek, RTL2838UHIDIR, SN: 00000001
Found Rafael Micro R820T/2 tuner
Exact sample rate is: 250000.000414 Hz
[SDR] Using device 0: Realtek, RTL2838UHIDIR, SN: 00000001, "Generic RTL2832U OEM"
[SDR] Sample rate set to 250000 S/s.
[Input] Bit detection level set to 0.0 (Auto).
[SDR] Tuner gain set to Auto.
[Input] Reading samples in async mode...
[SDR] Tuned to 433.920MHz.
Allocating 15 zero-copy buffers

I receiving only Prologue protocol 3 sensors

time      : 2025-02-11 11:08:13
model     : Prologue-TH
subtype   : 5
id        : 254
Channel   : 3
Battery   : 1
Temperature: 4.70 C
Humidity  : 12 %
Button    : 0
[pulse_slicer_ppm] Prologue, FreeTec NC-7104, NC-7159-675 temperature sensor
codes     : {38}17fa80bc30, {36}5fea02f0c, {36}5fea02f0c, {36}5fea02f0c, {36}5fea02f0c, {36}5fea02f0c, {0}0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2025-02-11 11:08:33
model     : Prologue-TH
subtype   : 5
id        : 204
Channel   : 1
Battery   : 1
Temperature: 7.60 C
Humidity  : 10 %
Button    : 0
[pulse_slicer_ppm] Prologue, FreeTec NC-7104, NC-7159-675 temperature sensor
codes     : {38}1732013028, {36}5cc804c0a, {36}5cc804c0a, {36}5cc804c0a, {36}5cc804c0a, {36}5cc804c0a, {36}5cc804c0a, {0}0

On another PC with the same radio and antenna I get all normally:

time      : 2025-02-11 11:07:42
Model     : Auriol-4LD5661                         ID        : 5a
Battery OK: 1            Temperature: 7.9 C        Rain      : 7.0 mm
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2025-02-11 11:07:58
model     : Prologue-TH  subtype   : 5             id        : 204
Channel   : 1            Battery   : 1             Temperature: 7.60 C       Humidity  : 10 %          Button    : 0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2025-02-11 11:08:13
model     : Prologue-TH  subtype   : 5             id        : 254
Channel   : 3            Battery   : 1             Temperature: 4.70 C       Humidity  : 12 %          Button    : 0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2025-02-11 11:11:01
model     : Auriol-AFW2A1 id        : 57
Channel   : 2            Battery   : 1             Button    : 0             Temperature: 4.5 C        Humidity  : 53 %

deviantintegral, 2025-03-22:
Isn't that protocol 146, not 200? I see

```
[146]  Auriol AFW2A1 temperature/humidity sensor
```

which matches the model you're getting.