pbkhrv/rtl_433-hass-addons #211: s6-overlay-suexec fatal error

nsd2386, 2025-01-27:
### The problem

Following a restart of the system running HA, the addon failed to load due to an S6 fatal error. 
Manually starting the addon yielded the same result.
Uninstalling and re-installing yielded the same result.

Installed "rtl_433 (next)" and that was able to start normally and HA is able to see data from the sensors again.

### What addon are you reporting the bug for?

rtl_443

### What is the addon version?

0.5.2

### What type of MQTT Broker are you using?

Home Assistant Mosquitto MQTT Broker

### Addon log messages

```txt
s6-overlay-suexec: fatal: unable to spawn /package/admin/s6-overlay-3.1.5.0/libexec/preinit: No such file or directory
```

deviantintegral, 2025-02-08:
That's odd! What architecture are you running? When I check locally I see that file exists:

```
$ docker run --rm ghcr.io/pbkhrv/rtl_433-hass-addons-rtl_433-aarch64:0.5.2 stat /package/admin/s6-overlay-3.1.5.0/libexec/preinit
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
  File: /package/admin/s6-overlay-3.1.5.0/libexec/preinit
  Size: 2502      	Blocks: 8          IO Block: 4096   regular file
Device: 2eh/46d	Inode: 1638013     Links: 1
Access: (0755/-rwxr-xr-x)  Uid: (    0/    root)   Gid: (    0/    root)
Access: 2023-05-04 23:55:50.000000000 +0000
Modify: 2023-05-04 23:55:50.000000000 +0000
Change: 2025-02-08 16:11:49.428992228 +0000
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
```

nsd2386, 2025-02-08:
I am currently running HA Core 2025.1.4 and Supervisor 2025.02.0 on a rpi3-64. The error seemed odd as the add-on was running smoothly prior to a system restart but then would throw the same error every time I tried to start it after. As I said, uninstalling and re-installing yielded the same result as well. However, _rtl_433 Home Assistant Add-on (next)_ has been working correctly since the installing, including through multiple system restarts

deviantintegral, 2025-02-08:
This feels like an issue with the sd card. You could try removing the image in the terminal and manually pulling it with docker to see if there’s any errors.

deviantintegral, 2025-03-22:
Feel free to reopen if you have more information.

closed 2025-03-22 by deviantintegral