pbkhrv/rtl_433-hass-addons #212: conf.template doesn't need `\` for output

p8me, 2025-02-03:
I may be wrong, sorry if that's the case.
I had to remove `\` in the `output` for conf for things to work.

```
- `output mqtt://\${host}:\${port},user=\${username},pass=\${password},retain=\${retain}`
+ `output mqtt://${host}:${port},user=${username},pass=${password},retain=${retain}`
```

otherwise I would get this error:
```
[rtl_433] MQTT: MQTT connect (${host}:${port}) failed: cannot parse address
```

deviantintegral, 2025-02-08:
That's surprising. Are you using the automatic configuration? Does your configuration file end in `.template`?

p8me, 2025-02-10:
yep it's `rtl_433.conf.template`

could it have anything to do with my setup? I am using the home assistant on a single rpi.

Ziron, 2025-05-28:
Thank you! Removing the `\` in the `output` line finally fixed the errors and I got the program to connect to mqtt!

andrebrei, 2025-06-05:
Just curious.
Did you @p8me add a template with a c&p from the run.sh script?
 
On first start (no config dir) I get:
```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
/run.sh: line 42: $1: unbound variable
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
```

Note the `/run.sh: line 42: $1: unbound variable` which I believe is the result of the missing escape for `-$1` https://github.com/pbkhrv/rtl_433-hass-addons/blob/e334ffe7a7874544170acf74fd4b41a72953ef17/rtl_433/run.sh#L63
(a HERE part gets shell expanded still so `$something`without the escape the shell will try to expand)

If that wouldn't fail to expand then the auto-created config template would have the output config line without escapes which works.
The comment would get in the way still because the template file gets expanded on start again so the `-$1` needs to be a `-\$1` after the first execution still. That means the run.sh line needs to have `-\\\$1` for it to work.
e.g. with a single escape in run.sh I get on 2nd+ runs currently:
`/tmp/rtl_433_heredoc: line 1: $1: unbound variable`
With a double escape in the run.sh this error goes away too and rtl_433 comes up.