pbkhrv/rtl_433-hass-addons #214: Protocols over #250 not supported

Al-Dek, 2025-02-19:
### The problem

It`s impossible to use protocols over #250.
Add-on  shows error "**[rtl_433] Protocol number specified (251) is larger than number of protocols**" and completely stops.

In rtl_433 add-on docs noted:
The full list of supported protocols and devices can be found under "Supported device protocols" section of the [rtl_433's README](https://github.com/merbanan/rtl_433/blob/master/README.md).

In the mentioned README.md listed protocols up to #275.
Is this because rtl_433 has now release [24.10](https://github.com/merbanan/rtl_433/releases/tag/24.10) but rtl_433 add-on has only 23.11?

### What addon are you reporting the bug for?

rtl_443

### What is the addon version?

0.5.2

### What type of MQTT Broker are you using?

Home Assistant Mosquitto MQTT Broker

### Addon log messages

```txt
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
Starting rtl_433 with rtl_433.conf...
[rtl_433] rtl_433 version 23.11 branch  at 202311281352 inputs file rtl_tcp RTL-SDR
[rtl_433] MQTT: Publishing MQTT data to core-mosquitto port 1883
[rtl_433] MQTT: Publishing device info to MQTT topic "rtl_433/9b13b3f4-rtl433/devices[/type][/model][/subtype][/channel][/id]".
[rtl_433] MQTT: Publishing events info to MQTT topic "rtl_433/9b13b3f4-rtl433/events".
[rtl_433] MQTT: Publishing states info to MQTT topic "rtl_433/9b13b3f4-rtl433/states".
[rtl_433] Protocol number specified (251) is larger than number of protocols
[rtl_433] 
[rtl_433] 		= Supported device protocols =
[rtl_433]     [01]  Silvercrest Remote Control
[rtl_433]     [02]  Rubicson, TFA 30.3197 or InFactory PT-310 Temperature Sensor
[rtl_433]     [03]  Prologue, FreeTec NC-7104, NC-7159-675 temperature sensor
[rtl_433]     [04]  Waveman Switch Transmitter
[rtl_433]     [06]* ELV EM 1000
[rtl_433]     [07]* ELV WS 2000
[rtl_433]     [08]  LaCrosse TX Temperature / Humidity Sensor
[rtl_433]     [10]  Acurite 896 Rain Gauge
[rtl_433]     [11]  Acurite 609TXC Temperature and Humidity Sensor
[rtl_433]     [12]  Oregon Scientific Weather Sensor
[rtl_433]     [13]* Mebus 433
[rtl_433]     [14]* Intertechno 433
[rtl_433]     [15]  KlikAanKlikUit Wireless Switch
[rtl_433]     [16]  AlectoV1 Weather Sensor (Alecto WS3500 WS4500 Ventus W155/W044 Oregon)
[rtl_433]     [17]  Cardin S466-TX2
[rtl_433]     [18]  Fine Offset Electronics, WH2, WH5, Telldus Temperature/Humidity/Rain Sensor
[rtl_433]     [19]  Nexus, FreeTec NC-7345, NX-3980, Solight TE82S, TFA 30.3209 temperature/humidity sensor
[rtl_433]     [20]  Ambient Weather F007TH, TFA 30.3208.02, SwitchDocLabs F016TH temperature sensor
[rtl_433]     [21]  Calibeur RF-104 Sensor
[rtl_433]     [22]  X10 RF
[rtl_433]     [23]  DSC Security Contact
[rtl_433]     [24]* Brennenstuhl RCS 2044
[rtl_433]     [25]  Globaltronics GT-WT-02 Sensor
[rtl_433]     [26]  Danfoss CFR Thermostat
[rtl_433]     [29]  Chuango Security Technology
[rtl_433]     [30]  Generic Remote SC226x EV1527
[rtl_433]     [31]  TFA-Twin-Plus-30.3049, Conrad KW9010, Ea2 BL999
[... 203 lines cut ...]
[rtl_433]     [234]  Watchman Sonic Advanced / Plus, Tekelek
[rtl_433]     [235]  Oil Ultrasonic SMART FSK
[rtl_433]     [236]  Gasmate BA1008 meat thermometer
[rtl_433]     [237]  Flowis flow meters
[rtl_433]     [238]  Wireless M-Bus, Mode T, 32.768kbps (-f 868.3M -s 1000k)
[rtl_433]     [239]  Revolt NC-5642 Energy Meter
[rtl_433]     [240]  LaCrosse TX31U-IT, The Weather Channel WS-1910TWC-IT
[rtl_433]     [241]  EezTire E618, Carchet TPMS
[rtl_433]     [242]* Baldr / RainPoint rain gauge.
[rtl_433]     [243]  Celsia CZC1 Thermostat
[rtl_433]     [244]  Fine Offset Electronics WS90 weather station
[rtl_433]     [245]* ThermoPro TX-2C Thermometer and Humidity sensor
[rtl_433]     [246]  TFA 30.3151 Weather Station
[rtl_433]     [247]  Bresser water leakage
[rtl_433]     [248]* Nissan TPMS
[rtl_433]     [249]  Bresser lightning
[rtl_433]     [250]  Schou 72543 Day Rain Gauge
[rtl_433] 
[rtl_433] * Disabled by default, use -R n or a conf file to enable
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
```

deviantintegral, 2025-02-20:
It looks like there's even been a release since then: https://github.com/merbanan/rtl_433/releases/tag/25.02

As an immediate workaround, you can use the `next` version of the addon which is built from the master branch of rtl_433.

Al-Dek, 2025-02-20:
@deviantintegral 
> It looks like there's even been a release since then: https://github.com/merbanan/rtl_433/releases/tag/25.02
> 
> As an immediate workaround, you can use the `next` version of the addon which is built from the master branch of rtl_433.

Great!
rtl_433 next works perfect. 
I have some errors with the logging but it not affected data obtaining from devices.
Thank you!

```
Logger: homeassistant.components.hassio.http
Source: components/hassio/http.py:235
integration: Home Assistant Supervisor (documentation, issues)
First occurred: 11:31:29 (47 occurrences)
Last logged: 11:31:47

Client error on api addons/9b13b3f4_rtl433-next/logs/follow request Response payload is not completed: <TransferEncodingError: 400, message='Not enough data for satisfy transfer length header.'>
```

deviantintegral, 2025-03-01:
Interesting, I think that error is in home assistant or supervisor itself. Addons just send log data on standard output, and then home assistant is entirely responsible for wrapping it up into http for the web app.

Al-Dek, 2025-03-01:
These errors gone, as I think with one of the HA core updates. 
However, it is just curious, why some sensors are "formatted" differently...

```
...
[rtl_433] _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
[rtl_433] 
[rtl_433] time      : 2025-03-01T06:22:21+0300
[rtl_433] model     : Oregon-THGR810
[rtl_433] House Code: 119
[rtl_433] Channel   : 2
[rtl_433] Battery   : 1
[rtl_433] Celsius   : 10.00 C
[rtl_433] Humidity  : 38 %
[rtl_433] _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
[rtl_433] 
[rtl_433] time      : 2025-03-01T06:22:21+0300
[rtl_433] model     : Fineoffset-WS90
[rtl_433] ID        : 003948
[rtl_433] Battery   : 1.000
[rtl_433] Battery Voltage: 3040 mV
[rtl_433] Temperature: -3.1 C
[rtl_433] Humidity  : 78 %
[rtl_433] Wind direction: 190
[rtl_433] Wind speed: 0.9 m/s
[rtl_433] Gust speed: 1.0 m/s
[rtl_433] UVI       : 0.0
[rtl_433] Light     : 0.0 lux
[rtl_433] Flags     : 81
[rtl_433] Total Rain: 0.0 mm
[rtl_433] Rain Start: 0
[rtl_433] Supercap Voltage: 2.4 V
[rtl_433] Firmware Version: 150
[rtl_433] Extra Data: 3fff000000------0000ffbffb0000
[rtl_433] Integrity : CRC
[rtl_433] _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
[rtl_433] 
[rtl_433] time      : 2025-03-01T06:22:26+0300
[rtl_433] model     : Oregon-PCR800
[rtl_433] House Code: 5
[rtl_433] Channel   : 0
[rtl_433] Battery   : 0
[rtl_433] Rain Rate :   0.0 mm/h
[rtl_433] Total Rain: 1227.988 mm
[rtl_433] _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
[rtl_433] time      : 2025-03-01T06:22:28+0300
[rtl_433] model     : LaCrosse-TX141THBv2                    Sensor ID : 45
[rtl_433] Channel   : 0            Battery   : 0             Temperature: -3.60 C      Humidity  : 73 %          Test?     : No            Integrity : CRC
[rtl_433] _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
[rtl_433] 
...
[rtl_433] _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
[rtl_433] time      : 2025-03-01T06:22:32+0300
[rtl_433] model     : Oregon-WGR800 House Code: 10
[rtl_433] Channel   : 0            Battery   : 1             Gust      : 1.9 m/s       Average   : 0.0 m/s       Direction : 247.5 degrees
[rtl_433] _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
[rtl_433] 
...
```

deviantintegral, 2025-03-02:
It probably has to do with what it sees as the width of the terminal, and if it goes too long it's going to switch to a single column layout. That's why `kv` output is best for humans, and there's json and other formats for machine reading.

atx32, 2025-03-04:
> It looks like there's even been a release since then: https://github.com/merbanan/rtl_433/releases/tag/25.02
> 
> As an immediate workaround, you can use the `next` version of the addon which is built from the master branch of rtl_433.

Is there a plan to update the release version of the add-on to use the most current rtl_433?

deviantintegral, 2025-03-22:
I just pushed an update to the stable branch which includes rtl_433 25.02. Enjoy!

closed 2025-03-22 by deviantintegral