pbkhrv/rtl_433-hass-addons #216: autodiscovery: WARNING:root:No suitable identifier found for model: Bresser-5in1

jaroslawp, 2025-03-02:
### The problem

My rtl433 autodiscovery home assistant addon stopped working recently with this message:
```
 WARNING:root:No suitable identifier found for model: Bresser-5in1
```
and  corresponnding device/entities are not created in home assistant ..
My config:
```
mqtt_host: ""
mqtt_port: 1883
mqtt_user: ""
mqtt_password: ""
mqtt_retain: true
rtl_topic: rtl_433/+/events
device_topic_suffix: devices=rtl_433/+/devices/Bresser-5in1/+
discovery_prefix: homeassistant
discovery_interval: 600
force_update: false
log_level: debug
```
any idea what the problem could be?

Thanks!
Jarek

### What addon are you reporting the bug for?

rtl_433_mqtt_autodiscover

### What is the addon version?

0.8.2

### What type of MQTT Broker are you using?

Home Assistant Mosquitto MQTT Broker

### Addon log messages

```txt
[2025-03-02T15:56:18+0100] DEBUG:paho.mqtt.client:Sending PINGREQ
[2025-03-02T15:56:18+0100] DEBUG:paho.mqtt.client:Received PINGRESP
[2025-03-02T15:56:28+0100] DEBUG:paho.mqtt.client:Received PUBLISH (d0, q0, r0, m0), 'rtl_433/9b13b3f4-rtl433/events', ...  (312 bytes)
[2025-03-02T15:56:28+0100] DEBUG:root:MQTT message: "{\"time\":\"2025-03-02 15:56:27+0100\",\"protocol\":119,\"model\":\"Bresser-5in1\",\"id\":125,\"battery_ok\":1,\"temperature_C\":10.4,\"humidity\":56,\"wind_max_m_s\":0.8,\"wind_avg_m_s\":0.8,\"wind_dir_deg\":0,\"rain_mm\":0,\"mic\":\"CHECKSUM\",\"mod\":\"FSK\",\"freq1\":868.35712,\"freq2\":868.19706,\"rssi\":-8.05163,\"snr\":17.96473,\"noise\":-26.0164}"
[2025-03-02T15:56:28+0100] WARNING:root:No suitable identifier found for model: Bresser-5in1
```

deviantintegral, 2025-03-02:
Aside, you don't have to run the autodiscovery addon all the time. You can set "retain" and then just run it when you're adding new devices. Of course, that won't work in the short term.

> device_topic_suffix: devices=rtl_433/+/devices/Bresser-5in1/+

That seems wrong to include a specific device. Can you try removing it entirely? Personally, I use `devices[/type][/model][/subtype]/C[channel:0]` as my weather stations change IDs when changing batteries.

jaroslawp, 2025-03-03:
Yes, it usually does not run all the time : I just chnaged batteries in my device so was trying to get it back (yes it also does change id on battery change).

Indeed, removing the ```device_topic_suffix``` alltogether fixes the problem,
Thank you!

Jarek

closed 2025-03-03 by jaroslawp