pbkhrv/rtl_433-hass-addons #218: It has been working for months...

AndyW999, 2025-03-18:
### The problem

I don't know if it was because of an update but the log details below.

Maybe because of a HA update?

Was very reliable after I plugged it into a powered hub rather than directly into RPi5.

Thanks

Andy.

### What addon are you reporting the bug for?

rtl_443

### What is the addon version?

25.02-4-gda289d37

### What type of MQTT Broker are you using?

Home Assistant Mosquitto MQTT Broker

### Addon log messages

```txt
Starting rtl_433 with rtl_433.conf...
[rtl_433] rtl_433 version 25.02-4-gda289d37 branch master at 202503111533 inputs file rtl_tcp RTL-SDR SoapySDR with TLS
[rtl_433] Detached kernel driver
[rtl_433] rtlsdr_demod_write_reg failed with -7
[rtl_433] rtlsdr_demod_write_reg failed with -7
[rtl_433] Found Rafael Micro R820T tuner
[rtl_433] r82xx_read: i2c rd failed=-1 reg=00 len=5
[rtl_433] r82xx_init: failed=-1
[rtl_433] Exact sample rate is: 250000.000414 Hz
[rtl_433] [SDR] Using device 0: Realtek, RTL2838UHIDIR, SN: 00000001, "Generic RTL2832U OEM"
[rtl_433] r82xx_read: i2c wr failed=-1 reg=00 len=1
[rtl_433] r82xx_set_freq: failed=-1
[rtl_433] rtlsdr_demod_read_reg failed with -7
[rtl_433] rtlsdr_demod_write_reg failed with -9
[rtl_433] [sdr_set_sample_rate] Failed to set sample rate.
[rtl_433] cb transfer status: 1, canceling...
[rtl_433] [rtlsdr_read_loop] LIBUSB_ERROR_NOT_FOUND: Entity not found! Check your RTL-SDR dongle, USB cables, and power supply.
[rtl_433] [SDR] async read failed (-5).
[rtl_433] [Input] Input device start failed, exiting!
[rtl_433] Reattaching kernel driver failed!
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
```

AndyW999, 2025-03-21:
I did all the usual things, restart, power cycle SDR, try a different one, no change.

And a few days alter it magically started working again...

But it now won't send anything to MQTT but it does show my weather station in the logs - which it never used to.

deviantintegral, 2025-03-22:
Do you see any reference to mqtt in the logs? For example, here's mine (note I've customized things so the topics are very likely to be different for you:

```
[doorbell] MQTT: Publishing MQTT data to core-mosquitto port 1883
[doorbell] MQTT: Publishing device info to MQTT topic "rtl_433/local-rtl433/devices[/type][/model][/subtype][/channel][/id]".
[doorbell] MQTT: Publishing events info to MQTT topic "rtl_433/local-rtl433/events".
[doorbell] MQTT: Publishing states info to MQTT topic "rtl_433/local-rtl433/states".
```

AndyW999, 2025-03-22:
OK - fixed!

closed 2025-03-22 by AndyW999