pbkhrv/rtl_433-hass-addons #222: Nexus-TH temp/humidity sensor stopped working on 0.6.0

myromeo, 2025-03-23:
### The problem

As title, basic Nexus-TH temperature / humidity station stopped being received after update to 0.6.0. Working fine on 0.5.2. 

### What addon are you reporting the bug for?

rtl_443

### What is the addon version?

0.6.0

### What type of MQTT Broker are you using?

Home Assistant Mosquitto MQTT Broker

### Addon log messages

```txt
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
Starting rtl_433 with rtl_433mhz.conf...
[rtl_433mhz] rtl_433 version 25.02 branch  at 202502191252 inputs file rtl_tcp RTL-SDR SoapySDR with TLS
[rtl_433mhz] MQTT: Publishing MQTT data to 192.168.0.16 port 1883
[rtl_433mhz] MQTT: Publishing device info to MQTT topic "rtl_433/9b13b3f4-rtl433/devices[/type][/model][/subtype][/channel][/id]".
[rtl_433mhz] MQTT: Publishing events info to MQTT topic "rtl_433/9b13b3f4-rtl433/events".
[rtl_433mhz] MQTT: Publishing states info to MQTT topic "rtl_433/9b13b3f4-rtl433/states".
[rtl_433mhz] Found Rafael Micro R820T tuner
[rtl_433mhz] Exact sample rate is: 250000.000414 Hz
[rtl_433mhz] [SDR] Using device 1: Realtek, RTL2838UHIDIR, SN: 00000001, "Generic RTL2832U OEM"
[rtl_433mhz] [R82XX] PLL not locked!
```

### Additional information

`# rtl_433 Configuration Template
# MQTT information will be automatically added.
# Create multiple files ending in '.conf.template' to manage multiple rtl_433 radios.
# The 'device' setting must be set before MQTT output lines.
# For more details, visit: https://github.com/merbanan/rtl_433/blob/master/conf/rtl_433.example.conf

# Device Configuration
device 1  # Realtek dongle (loft antenna)

# MQTT Output Configuration
output mqtt://192.168.0.16:1883,user=mqtt,pass=mqtt,retain=1,devices=rtl_433/9b13b3f4-rtl433/devices[/type][/model][/subtype][/channel][/id],events=rtl_433/9b13b3f4-rtl433/events,states=rtl_433/9b13b3f4-rtl433/states

# Uncomment to enable the default "table" output to the addon logs
output kv

report_meta level

# Logging Configuration
verbose 3  # 1=fatal, 2=critical, 3=error, 4=warning (default), 5=notice, 6=info, 7=debug, 8=trace

# SDR Configuration
gain 48
frequency 433.92M
sample_rate 250k
pulse_detect minmax

# TPMS Protocols (Disabled by Default)
# Disable TPMS sensors by default. These can cause an overwhelming number of devices and entities to show up in Home Assistant.
# Generated by running: rtl_433 -R help 2>&1 | grep -i tpms | sd '.*\[(\d+)\].*' 'protocol -/run.sh'
protocol -59  # Steelmate TPMS
protocol -60  # Schrader TPMS
protocol -82  # Citroen TPMS
protocol -88  # Toyota TPMS
protocol -89  # Ford TPMS
protocol -90  # Renault TPMS
protocol -95  # Schrader TPMS EG53MA4, PA66GF35
protocol -110 # PMV-107J (Toyota) TPMS
protocol -123 # Jansite TPMS Model TY02S
protocol -140 # Elantra2012 TPMS
protocol -156 # Abarth 124 Spider TPMS
protocol -168 # Schrader TPMS SMD3MA4 (Subaru)
protocol -180 # Jansite TPMS Model Solar
protocol -186 # Hyundai TPMS (VDO)
protocol -201 # Unbranded SolarTPMS for trucks
protocol -203 # Porsche Boxster/Cayman TPMS
protocol -212 # Renault TPMS

# Custom Decoder
decoder n=gswds07,m=OOK_PWM,s=476,l=1364,r=15000,g=1600,bits>=24,bits<=25,invert,get=@0:{20}:id,get=@20:{4}:event:[10:open 14:closed 6:low_battery 7:tamper],unique
`

deviantintegral, 2025-03-23:
Do you get any other sensors at all?

I'm wondering what happens if you pare down your configuration to the basics, in case the SDR configuration or custom decoder is somehow causing a problem.

I don't see any bug reports upstream about this sensor, and nothing in the [commits since the last release](https://github.com/merbanan/rtl_433/compare/25.02...master) indicating it was fixed.

Can you pull your dongle out to another system and see if it picks up the temperature sensor? If it doesn't, and you can see it works with an older version of rtl_433, then at least we'd know it's a problem with rtl_433 itself and not the addon.

myromeo, 2025-03-23:
Hi,

I’m receiving several Yale HSA sensors without issue and nexus has worked stable on 0.5.2. Only on upgrading to 0.6.0 did the Nexus sensor stop. Yale HSA works fine on 0.6.0

I’ve rolled back to 0.5.2 and it’s working fine? Just stops on 0.6.0

deviantintegral, 2025-03-23:
The only changes made were to upgrade to the latest stable releases of the base OS and rtl_433.

If you can replicate the issue outside of Home Assistant OS (using https://github.com/hertzg/rtl_433_docker, `brew`, whatever) on a different computer then the next step would be to file a bug report at https://github.com/merbanan/rtl_433/issues.

ChrisSutton123, 2025-03-24:
I've used this addon for a number of years on the various iterations of software releases.
I use it to receive signals from both window/door sensors and SOS buttons.
After upgrading to 0.6.0 nothing is received by the dongle. Checked by setting "output kv" in "rtl_433.conf.template". If I downgrade back to 0.5.2 all works perfectly.
I have tested on two different system with same results.
I have never modified "rtl_433.conf.template" in the past do I need to for version 0.6.0

clb92, 2025-03-24:
@ChrisSutton123 I can't figure out how to downgrade. My last backup in HA was several months ago, for some reason (is it not supposed to back up when updating addons?), so I really don't want to do it that way.

ChrisSutton123, 2025-03-24:
I get the same error message. "PLL not locked"

I can email you my 0.5.2 backup tar file if it helps

clb92, 2025-03-24:
No need now, but thanks. I removed the addon from HA and I am now using an old version of the standalone rtl_433 docker container, and now it works again.

I tried the standalone docker with latest version too, and that fails, so something has changed with rtl_433 that means my receiver doesn't work with newer versions.

ChrisSutton123, 2025-03-24:
Fixed my problem:-

It seems in rtl_433 release 24.10 there was a breaking change:-
         "Changed Smoke-GS558 (Protocol 86) and Akhan-100F14 (Protocol 48) to default disabled"

My devices used "Smoke-G5558"
I needed to add "protocol 86" to "rtl_433/rtl_433.conf.template"
Now i can use version 0.6.0 of the addon

clb92, 2025-03-24:
Hmm, perhaps I just need to enable some protocols as well. That would explain why it could pick up some TPMS still, but not my buttons.

clb92, 2025-03-25:
That was it, thank you so much! Protocol 48 did the trick for me. Apparently every single button I have uses that protocol.

deviantintegral, 2025-03-27:
Nice! Glad you got this sorted. I'm closing this as fixed, but will leave it pinned in the issue list for a while.

closed 2025-03-27 by deviantintegral

PetrP2B, 2025-05-20:
Hi, I have the same problem, since the last update I keep getting "PLL not locked" messages in the log and RTL_433 doesn't work, sometimes it sporadically receives a value from the sensor, but it works poorly. I tried buying a new US dongle and it's the same.